espressif/arduino-esp32 #1152: WiFi.mode() breaks analogRead() for lower pins

Lakoja, 2018-02-26:
### Description:
I'm trying to read (input) voltage over a voltage divider. The simplified test code is fairly simple:

`
int v = analogRead(voltage_pin);
// output value
WiFi.mode(WIFI_AP);
v = analogRead(voltage_pin);
// output value
`

The second output is always the maximum range value (ie 4095).
The first is a reasonable expected value.

This happens only for lower pin numbers. In my design gpio 27 and lower. It does not happen for higher pins like 35.
This happens for several boards (different manufacturer) and is stable over time.

I'm a little surprised by this side-effect. Did I miss something?

beegee-tokyo, 2018-02-26:
Known problem.
[ADC2 channel cannot be used when WiFi is in use](http://desire.giesecke.tk/index.php/2018/01/30/adc2-channel-cannot-be-used-when-wifi-is-in-use/)
Solution might come from ESP-IDF sometime, but not sure.

Lakoja, 2018-02-26:
Thanks for the reply.
Me any my luck while choosing pins for tasks. :-)

I have now a fly-wire to pin 34...

copercini, 2018-02-26:
I will close this issue to concentrate it at https://github.com/espressif/arduino-esp32/issues/102

It looks like a fix will come soon =)

closed 2018-02-26 by copercini

Meliox, 2018-05-13:
Both issues are closed, but it has not been fixed as of today in arduino-esp32... Has it been fixed other places, or?