espressif/arduino-esp32 #1153: Problem with BLE - pCharacteristic->setValue();

Boladehenoz, 2018-02-26:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Hello, i have problems writing to a characteristic, when i try to use setValue i get the following by the ESP exception decoder. I paste my BLE configuration code here too, as you can see i define 4 characteristics, one of them is for reading and the other ones for reading. I think that there is where the problem is, but im not sure.

### Sketch:
#define SERVICE_UUID           "0000ffd0-0000-1000-8000-00805f9b34fb" // UART service UUID
#define CHARACTERISTIC_UUID_RX "0000ffe1-0000-1000-8000-00805f9b34fb"
#define CHARACTERISTIC_UUID_TX_pos "0000ffd1-0000-1000-8000-00805f9b34fb"
#define CHARACTERISTIC_UUID_TX_luz "0000ffd2-0000-1000-8000-00805f9b34fb"
#define CHARACTERISTIC_UUID_TX_cal "0000ffd3-0000-1000-8000-00805f9b34fb"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        for (int i = 0; i < rxValue.length(); i++)
        {
          //Serial.print(rxValue[i]);
          inputString_BT += rxValue[i];
        }
      }
      BTstring_Complete = 1;
    }
};

void setup() {
  Serial.begin(115200);
  BLEDevice::init("OpenSun");

  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  /////////////////////////////////////////////////////////////////////////////////
 BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID_RX,
                                         BLECharacteristic::PROPERTY_WRITE |
                                         BLECharacteristic::PROPERTY_NOTIFY
                                       );
pCharacteristic->addDescriptor(new BLE2902());

pCharacteristic->setCallbacks(new MyCallbacks());
  /////////////////////////////////////////////////////////////////////////////////
  BLECharacteristic *pCharacteristic2 = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID_TX_pos,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_NOTIFY
                                       );
pCharacteristic2->addDescriptor(new BLE2902());
 /////////////////////////////////////////////////////////////////////////////////
 BLECharacteristic *pCharacteristic3 = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID_TX_cal,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_NOTIFY
                                       );
pCharacteristic3->addDescriptor(new BLE2902());                                       
 /////////////////////////////////////////////////////////////////////////////////
 BLECharacteristic *pCharacteristic4 = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID_TX_luz,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_NOTIFY
                                       );
pCharacteristic4->addDescriptor(new BLE2902());
 /////////////////////////////////////////////////////////////////////////////////

  pCharacteristic2->setValue(&cc, 1);
  pCharacteristic3->setValue(&cc, 1);
  pCharacteristic4->setValue(&cc, 1);
  
  // Start the service
  pService->start();
  
  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}

To write i use: 

if (deviceConnected) {
      pCharacteristic2->setValue("1");
      pCharacteristic2->notify();
    }

for example. Thank you all^^.

### Debug Messages:

0x4013c6a5: std::__cxx11::basic_string  , std::allocator  >::_M_data() const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h line 2200
:  (inlined by) std::__cxx11::basic_string  , std::allocator  >::_M_is_local() const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h line 170
:  (inlined by) std::__cxx11::basic_string  , std::allocator  >::swap(std::__cxx11::basic_string  , std::allocator  >&) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.tcc line 70
0x400d7578: std::__cxx11::basic_string  , std::allocator  >::operator=(std::__cxx11::basic_string  , std::allocator  >&&) at C:\Users\Boladeheno\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEValue.cpp line 92
:  (inlined by) BLEValue::setValue(unsigned char*, unsigned int) at C:\Users\Boladeheno\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLEValue.cpp line 135
0x400d40fe: BLECharacteristic::setValue(unsigned char*, unsigned int) at C:\Users\Boladeheno\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src/BLECharacteristic.cpp line 683
0x400d30a6: loop() at C:\Users\Boladeheno\Google Drive\NTS\Proyecto Jorge\OpenSunESP32/OpenSunESP32.ino line 184
0x40152d4a: loopTask(void*) at C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 18 (discriminator 1)

beegee-tokyo, 2018-02-26:
Never used `.setValue(std::string value),` but did you try to write `pCharacteristic2->setValue((std::string)"1");` instead?

emj7493, 2018-02-26:
yep, i got the same result, even if i only notify, without using setValue i got the same output.

beegee-tokyo, 2018-02-27:
Just saw something in your code.
You are doing the `->setValue()` thing inside `setup()`. That might be the problem!
Try to move the `->setValue` stuff into your main loop.
Delete 
```cpp
pCharacteristic2->setValue(&cc, 1);
pCharacteristic3->setValue(&cc, 1);
pCharacteristic4->setValue(&cc, 1);

```
from your `setup()`    
then do the `->setValue()` only inside `loop()`. Something like
```cpp
void loop() {
  if (deviceConnected) {
    pCharacteristic2->setValue("1");
    pCharacteristic2->notify();
  }
}
```

everslick, 2018-04-19:
@Boladehenoz does the suggestion from beegee-tokey fix your issue? If so, can you please close this issue?

closed 2019-08-15 by stale[bot]