espressif/arduino-esp32 #1154: Trouble getting Blink to upload to ESP32 feather

lemons13579, 2018-02-27:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?Adafruit feather ESP32?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE?
Flash Frequency:					?80Mhz?
Upload Speed:						?921600?

### Description:
When I "open a git Bash session pointing to ARDUINO_SKETCHBOOK_DIR and execute git submodule update --init --recursive", I thik I "point it" to the right spot, but it just says "fata;: Not a git repository (or any of the parent directories): . git" and I don't know what that means. Any help would be great, thanks!

Also, ignoring that issue, when I try to upload to the board (just uploading blink test) I get the message "exec: "C:\\Users\\[username]\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++": file does not exist
Error compiling for board Adafruit ESP32 Feather."
If these are the same issue then that's great, thanks!

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

beegee-tokyo, 2018-02-27:
You need to install arduino-esp32 first. Follow the tutorial e.g. [for Windows](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md)

lemons13579, 2018-02-27:
I think I did, I've followed all of the steps up to and past that point, so I don't know whats going wrong, maybe I'm not having it "start" in the right place? Any advice?

beegee-tokyo, 2018-02-27:
In Arduino IDE open the preferences. On the top you see "Sketchbook location".
For me that is "B:\Arduino-Dev".
Open git bash and go to your Sketchbook location, create a folder **_hardware_** and inside hardware create a folder **_espressif_**. This is folder where you start to clone arduino-esp32. See a list of commands below.
For me the folder structure looks like:
```bash
B:\
    ->Arduino-Dev\
        ->hardware\
            ->espressif\
```
You need to replace B:\Arduino-Dev with your sketchbook location.
Then I used git bash instead of the git GUI.
```bash
cd <your_sketchbook_location>
mkdir hardware
cd hardware
mkdir espressif
cd espressif
git clone https://github.com/espressif/arduino-esp32.git ./esp32
```
after cloning is finished do
```
cd esp32
git submodule update --init --recursive
cd tools
get.exe
```
That should install the stuff in the correct place.
**_Important is to get the correct sketchbook folder from your Arduino-IDE preferences!_**

beegee-tokyo, 2018-02-27:
If it works for you, close the issue please.

closed 2018-02-27 by lemons13579