espressif/arduino-esp32 #1155: Running python get.py results in ssl error

renegadeandy, 2018-02-27:
### Hardware:
Board:	ESP32 OOOOUUUUU DEV BOARD . ?ESP32 Dev Module?

### Description:
When installing esp32 arduino core on mac osx I run python get.py but it fails with:

```
 python get.py
System: Darwin, Info: Darwin-16.7.0-x86_64-i386-64bit
Platform: x86_64-apple-darwin
Tool xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz already downloaded
Extracting xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
Tool esptool-4dab24e-macos.tar.gz already downloaded
Extracting esptool-4dab24e-macos.tar.gz
Downloading mkspiffs-0.2.1-osx.tar.gz
Traceback (most recent call last):
  File "get.py", line 139, in <module>
    get_tool(tool)
  File "get.py", line 94, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1049, in endheaders
    self._send_output(message_body)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 893, in _send_output
    self.send(msg)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 855, in send
    self.connect()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/httplib.py", line 1274, in connect
    server_hostname=server_hostname)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 352, in wrap_socket
    _context=self)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 579, in __init__
    self.do_handshake()
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/ssl.py", line 808, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
```

lbernstone, 2018-02-27:
Duplicate https://github.com/espressif/arduino-esp32/issues/1143

closed 2018-02-27 by copercini

MStylesMS, 2018-06-29:
I have this problem as well.  Was it ever resolved?