espressif/arduino-esp32 #1156: ESP32 PICO D4 arduino ide support

fedy0, 2018-02-27:
How do I program esp32 PICO d4 in arduino ide? (there's no board variant based on it included yet)

lbernstone, 2018-02-28:
PICO KIT looks like it is just a standard ESP32 dev module.

fedy0, 2018-02-28:
Actually, I'm using TTGOV2 board ( https://github.com/LilyGO/TTGO-LORA32-V2.0) which uses esp32 PICO d4.

I have tried to upload with all board variants of esp32 in arduino ide but failed.

OS: Windows 10
IDE: Arduino 1.6.9
arduino-esp32 (core) : latest version 
Board: TTGOV2
Error during upload:
"... Warning: Could not auto-detect flash size (FlashID=0xffffff, SizeID=0XffA, defaulting to 4m

 fatal error occurred : Unknown size 4m"

Any help would be greatly appreciated

stickbreaker, 2018-02-28:
Update Arduino to the current version 1.8.5

copercini, 2018-02-28:
it was committed few hours ago: https://github.com/espressif/arduino-esp32/commit/d29cfdb10407d5d6c01880f8b66de6677bcd621e

fedy0, 2018-02-28:
I saw it and I just updated and tested now. It uploaded completely (using "esp32 PICO kit" option) but it was continuously restarting the MCU

copercini, 2018-02-28:
any error or message in the serial?

fedy0, 2018-02-28:
None, just after the display of boot messages on the serial monitor it restarts

tolo69, 2018-03-03:
I'm able to upload the Sketch to my ESP32_PICO_kit_v4. Since in Arduino v1.8.3 there is no board variant for ESP32_PICO I have tried with board configuration of ESP32 Dev Module (ESP-WROOM-32).
Upload was successfull, but SW is not running. Now I try with Arduino v1.8.5 ....

beegee-tokyo, 2018-03-03:
@tolo69 what sketch did you try to run? Try something simple first, like sending a message over Serial every 10 seconds.

tolo69, 2018-03-03:
Thanks for the hint !  Now my simple Sketch for the PICO_kit_v4  runs - just toggling IO15 and cyclic sending a string to COM-port.
used configuration:    Arduino-IDE v1.8.5, board: ESP32 Dev Module.

fedy0, 2018-03-04:
@tolo69 @copercini @Ibernstone @beegee-tokyo @stickbreaker

Mine is not working. I am using TTGOV2 LoRa32 based on esp32 PICO d4 with arduino 1.8.5. I selected both "ESP32_PICO_kit" and "esp32 dev module" in Arduino board options but after upload it could not run... Always restarting and printing unrecognized characters in the serial monitor.

Please anybody successful using this board (https://github.com/LilyGO/TTGO-LORA32-V2.0) on arduino ide, help me?

tolo69, 2018-03-04:
which URL-path for ESP32 board configuration have you included in the IDE-Settings ?

fedy0, 2018-03-04:
I cloned the repo (https://github.com/espressif/arduino-esp32/) into my Arduino ide folders (C:/arduino-1.8.5/hardware/)

It's up to date with this repo. 

Are you using the same TTGOV2 board? @tolo69

Please comment @Ibernstone if any suggestion

beegee-tokyo, 2018-03-04:
@fedy0 I found this [article](https://www.cnx-software.com/2017/10/13/this-ttgo-board-combines-esp32-lora-radio-and-oled-display-for-just-10/) I guess it is the board you have. In the article they say download the [TTGO folder](https://eyun.baidu.com/s/3hsiTNgg#sharelink/path=%2F) and copy it into `<Sketchbook Folder>/hardware` or `Arduino/hardware` Unfortunately the link goes to a Chinese Baidu post and to download the zip you need an account on Baidu. But maybe you can find `TTGO WIFI-OLED.zip` somewhere else.

tolo69, 2018-03-04:
I'm using same URL for board config but HW ist ESP32_PICO_kit_v4.

choopk, 2018-03-07:
How do i read/write from SD Card? I'm using the TTGOV2 board, but the SD card library doesn't know support this board unfortunately

fedy0, 2018-03-07:
Please and please can I get help on how to use my TTGOV2 board based on esp32 PICO d4 in arduino IDE?

Its compiles and uploads a simple serial print program but it continuously restarts

Any clue about my challenge above @choopk

choopk, 2018-03-07:
fedy0 change your baudrate to 115200 or upload speed to something lower.

choopk, 2018-03-08:
@fedy0 i am using arduino 1.8.5. 

Board:ESP32 dev Module
Flash Mode:QIO
Flash frequency:80Mhz
Flash size:4Mb(32Mb)
Upload speed:921600 or 115200
coredebug:none
Programmer:Parallel programmer

HOpe it helps

fedy0, 2018-03-09:
Yeah thanks @choopk. Although I have the same setup with yours and have tried upload severally... Never knew that I MUST press the reset button after every upload.

The module (TTGOV2) doesn't actually perform "Hard resetting..." as indicated in the arduino compile/upload log window. One must press reset button. 

Thanks guys

closed 2018-03-09 by fedy0

Aarti2561, 2022-02-16:
ets Jun  8 2016 00:22:57
22:16:21.391 -> 
22:16:21.391 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:16:21.425 -> flash read err, 1000
22:16:21.425 -> ets_main.c 371 
22:16:22.227 -> ets Jun  8 2016 00:22:57
22:16:22.227 -> 
22:16:22.227 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:16:22.261 -> configsip: 188777542, SPIWP:0xee
22:16:22.261 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:16:22.261 -> mode:DIO, clock div:1
22:16:22.261 -> load:0x3fff0018,len:4
22:16:22.261 -> load:0x3fff001c,len:1044
22:16:22.261 -> load:0x40078000,len:10124
22:16:22.261 -> load:0x40080400,len:5856
22:16:22.261 -> entry 0x400806a8
22:16:22.366 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Hello!
I am using ESP32 Pico D4 module and trying to upload a sketch. But for some reason the following error message is displayed on the serial monitor, after uploading the code.
I am not able to understand this. Any help will be appreciated!
22:16:22.400 -> Memory dump at 0x400e5f20: 15000100 05501450 10041410
22:16:22.400 -> Core 1 register dump:
22:16:22.400 -> PC      : 0x400e5f25  PS      : 0x00060530  A0      : 0x800e6098  A1      : 0x3ffb1ee0  
22:16:22.400 -> A2      : 0x00000011  A3      : 0x3ffbddd4  A4      : 0x3ffc1220  A5      : 0x00020c20  
22:16:22.400 -> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000021  A9      : 0x00013ffc  
22:16:22.436 -> A10     : 0x00000000  A11     : 0x3ffbddd4  A12     : 0x00000101  A13     : 0x00000100  
22:16:22.436 -> A14     : 0x00020c02  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x00000000  
22:16:22.436 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
22:16:22.436 -> 
22:16:22.436 -> ELF file SHA256: 0000000000000000
22:16:22.436 -> 
22:16:22.436 -> Backtrace: 0x400e5f25:0x3ffb1ee0 0x400e6095:0x3ffb1f10 0x400e43d2:0x3ffb1f40 0x400d0ea5:0x3ffb1f80 0x400d21aa:0x3ffb1fb0 0x400862fd:0x3ffb1fd0

Aarti2561, 2022-02-16:
Hello!
I am trying to upload an Arduino sketch on ESP32 pico d4. But for some reason, the code is not working as expected. The following error message is displayed after uploading the code.
I am not able to understand this. Any help is highly appreciated!
22:16:21.391 -> 
22:16:21.391 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:16:21.425 -> flash read err, 1000
22:16:21.425 -> ets_main.c 371 
22:16:22.227 -> ets Jun  8 2016 00:22:57
22:16:22.227 -> 
22:16:22.227 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
22:16:22.261 -> configsip: 188777542, SPIWP:0xee
22:16:22.261 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
22:16:22.261 -> mode:DIO, clock div:1
22:16:22.261 -> load:0x3fff0018,len:4
22:16:22.261 -> load:0x3fff001c,len:1044
22:16:22.261 -> load:0x40078000,len:10124
22:16:22.261 -> load:0x40080400,len:5856
22:16:22.261 -> entry 0x400806a8
22:16:22.366 -> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
22:16:22.400 -> Memory dump at 0x400e5f20: 15000100 05501450 10041410
22:16:22.400 -> Core 1 register dump:
22:16:22.400 -> PC      : 0x400e5f25  PS      : 0x00060530  A0      : 0x800e6098  A1      : 0x3ffb1ee0  
22:16:22.400 -> A2      : 0x00000011  A3      : 0x3ffbddd4  A4      : 0x3ffc1220  A5      : 0x00020c20  
22:16:22.400 -> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000021  A9      : 0x00013ffc  
22:16:22.436 -> A10     : 0x00000000  A11     : 0x3ffbddd4  A12     : 0x00000101  A13     : 0x00000100  
22:16:22.436 -> A14     : 0x00020c02  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x00000000  
22:16:22.436 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
22:16:22.436 -> 
22:16:22.436 -> ELF file SHA256: 0000000000000000
22:16:22.436 -> 
22:16:22.436 -> Backtrace: 0x400e5f25:0x3ffb1ee0 0x400e6095:0x3ffb1f10 0x400e43d2:0x3ffb1f40 0x400d0ea5:0x3ffb1f80 0x400d21aa:0x3ffb1fb0 0x400862fd:0x3ffb1fd0

Aarti2561, 2022-02-16:
This message appears on the serial monitor