espressif/arduino-esp32 #1157: attempting wifi access causes HUZZAH32 to lock up serial ports

JohnGEyles, 2018-02-27:
### Hardware:
Board:	Adafruit HUZZAH32
Core Installation/update date:	8Jan2018
IDE name:	Arduino IDE
Flash Frequency:	80Mhz
Upload Speed:	921600

### Description:
If I try to connect to my wifi network, it jams up the serial port.   Any further attempts to access the HUZZAH32 result in the errors shown below.

If I then plug the HUZZAH32 into a different serial port (or even a different computer), I often get the same error if I try to upload a sketch.   The only way I can fix things is if I somehow (by trying various ports) manage to upload a different sketch (that doesn't try any WiFi stuff).   IOW, it's the execution my sketch (after the HUZZAH32 resets when I unplug it into another port) that causes the problem.  This is a little scary, because if I'm ever unable to upload a non-malignant sketch, my device might become permanently unusable (is there some other way to clear out the bad sketch ?)

If I comment out the WiFi.begin and the while loop that WiFi.status, it works just fine (the other code is fairly ordinary stuff that sends SPI commands).

```

### Sketch:
```cpp

//Change the code below by your sketch
#include <SPI.h>
#include <WiFi.h>

const char* ssid = "<MINE>";
const char* password =  "<MINE>";

int i, rc;

int enabs[] = {14, 15, 25, 27, 32, 33};

void setup() {

   for (i=0; i<6; i++) {
      pinMode (enabs[i], OUTPUT);
      digitalWrite (enabs[i], HIGH);
   }

   Serial.begin(9600);

   WiFi.begin(ssid, password);

   while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println("Connected to the WiFi network");

   SPI.begin();
}

void loop() {
    if(Serial.available()>0)
    {
        rc=Serial.read();
        wrtledlvl (2, rc, 1);
    }
        
    Serial.print ("Most recent byte: "); Serial.println (rc);
    
}

void wrtledlvl (int chn, int lev, int sticky) {
    SPI.beginTransaction (SPISettings(1000, MSBFIRST, SPI_MODE0));
    digitalWrite (enabs[chn], LOW);
    SPI.transfer (0); SPI.transfer (lev);
    if (sticky) SPI.transfer (2<<4); SPI.transfer (lev);
    digitalWrite (enabs[chn], HIGH);
    SPI.endTransaction ();

```

### Debug Messages:
```
Traceback (most recent call last):
  File "esptool.py", line 2524, in <module>
  File "esptool.py", line 2517, in _main
  File "esptool.py", line 2246, in main
  File "esptool.py", line 177, in __init__
  File "serial/__init__.py", line 88, in serial_for_url
  File "serial/serialposix.py", line 268, in open
serial.serialutil.SerialException: [Errno 16] could not open port /dev/cu.SLAB_USBtoUART: [Errno 16] Resource busy: '/dev/cu.SLAB_USBtoUART'
Failed to execute script esptool
An error occurred while uploading the sketch```

JohnGEyles, 2018-02-27:
Following up ... I tried this again, but with one of the built-in examples (File->Examples->WiFi->SimpleWiFiServer) in case there was something wrong with the code I put in my sketch (above in my OP).    And the same thing happened.   And the USB port now seems to be permanently allergic to my HUZZAH32 (even though I managed to get a benign non-wifi sketch uploaded to it from my other computer).   I've tried resetting the SMC and the PRAM, multiple times, to no avail.   

This is really extremely frustrating.   I got the HUZZAH32 in order to have Wifi access, and I'm pretty much afraid to try to upload any code that attempts to do so.

beegee-tokyo, 2018-02-28:
When WiFi is enabled the power consumption is much higher. Maybe you USB ports cannot supply enough.

Did you try different USB cables?
Did you try to disable power saving of you computers USB ports?
Did you try to supply your Huzzah ESP32 from a different power supply while uploading through USB?

JohnGEyles, 2018-02-28:
Hmm, maybe that's it.   Can't seem to find a figure for current consumption with wifi enabled, but I guess it's worth a try.    How do you suggest providing additional power during upload ?   I believe if I try to provide add'l power through the battery JST connector, it switches over to USB power when a USB cable is connected.    Are there USB charging-type ports that also have data ?    It'd be pretty weird if the thing won't work off a regular (non-charging ISB port).   I suppose I could insert a several-second delay (in setup) before it tries to turn-on wifi, and then switch the thing from the upload port to a charging port.

bill-orange, 2018-03-01:
My hunch is that power is your problem. 

I pretty much have a battery hooked up to both Esp32 and ESP8266 regardless of use of the USB port use. I have not experienced your specific problem but I have found USB power to result in unpredictable instability. 

If you don’t have a suitable battery handy do you have a machine with a USB 3.0 port? There’s a lot more current there.

JohnGEyles, 2018-03-01:
Thanks for responding.   I'm thinking power is the problem too.   I can't believe they don't bother informing about this (something like "if you plan to turn on wifi, USB 2.0 will not provide enough power").    There's still a lot going on that doesn't make sense though, like some of my USB ports now totally unable to recognize the thing, and one of them not even working with a thumb drive now (I'm gonna be pissed if this thing actually fried any of my USB ports).

I don't think a battery will  help though, because the HUZZAH32 page states "You can also connect a ... battery to the JST jack. This will let the Feather run on a rechargeable battery. When the USB power is powered, it will automatically switch over to USB for power, as well as start charging the battery (if attached)".

I don't have a machine with USB 3.0 either.    But what about something like this (a USB hub with AC power adapter):

https://www.amazon.com/AmazonBasics-Port-2-5A-power-adapter/dp/B00DQFGH80?th=1

bill-orange, 2018-03-01:
A battery might be better but that ought to work.

JohnGEyles, 2018-03-01:
I'm confused about the battery.    The quote (two posts up) from the HUZZAH32 web page:

https://learn.adafruit.com/adafruit-huzzah32-esp32-feather?view=all#power-management

... suggests that the battery will not help, because the thing switches to USB power and starts charging the battery, as soon as you connect USB.

bill-orange, 2018-03-01:
Once the battery is charged, I think it will draw off the battery.

JohnGEyles, 2018-03-01:
Ok, I ordered a battery, we'll see.   Any thoughts about why many of my USB ports seem no longer to recognize the ESP32, and at least one of them actually seems damaged (will not even recognize a thumb drive) ?

bill-orange, 2018-03-01:
That is odd.  USB ports can be damaged by a short.  Any reason to suspect your cable?

JohnGEyles, 2018-03-01:
You're not the first person to mention that.   I really should try a different cable.   Doh.

bill-orange, 2018-03-01:
A shorted cable could conceivably damage a USB port.

JohnGEyles, 2018-03-01:
The one USB port that seemed to have gone dead (on my laptop), thumb-drive not seen, was resurrected with SMC reset.   So I'm left with all the weirdness of some ports not seeing the ESP32, or going dead if they do see it and I try a sketch with wifi.   So I'll a battery and different USB cable and see what happens.

Man, I'm going to be pissed if all this is due to the USB 2.0 not supplying enough power for wifi.   

Thanks for the support, @bill-orange.

DO1AJG, 2018-03-17:
Any updates on the 'lost serial port issue"?
I kinda face the same situation after having played a couple of times with the arduino IDE OTA example. After enabling OTA my HUZZAH32 cannot be seen as a serial port anymore and the OTA feature doesn't any longer work due to some stupid mistake i made in the code ... 
So basically I cannot access the HUZZAH32 anymore ...
Best,
Andreas

JohnGEyles, 2018-03-17:
I'm not sure what OTA means.   I solved my problem through some combination of shorter/thicker USB cable, adding a battery, and realizing that it works with some of the USB ports on my computer (the ones connected to the hub in an Apple monitor) but not others.    It did semi-permanently hose some of my USB ports, but I re-surrected them by some combination of resetting the SMC and clearing the PRAM/NVRAM (this is on a Mac).   Good luck.

bill-orange, 2018-03-17:
@JohnGEyles All's well that ends well.  Yours did sound like a hardware problem. 

OTA is over the air programming.  I believe the library is a built-in so you can find examples in the example code pull-down.  It is much faster than serial.  You should try it.  I have an ESP8266 hooked up to my garage door.  With OTA I can tweak the code without dis-mounting it and hauling in the house.  The only OTA issue I have had involves its dependents on Bonjour (the Apple component available for PCs).  Somehow I can not detect my ESP8266 units for OTA unless I disconnect my computer from Ethernet and plug in a WiFi dongle.  this appears to be a Bonjour problem

JohnGEyles, 2018-03-17:
I'll give the OTA a try.   Since I'm using a Mac does that mean I don't need to mess with this Bonjour thing ?

bill-orange, 2018-03-17:
Yup. Bomjour should built in. It is an iTunes component.

stickbreaker, 2018-03-17:
@bill-orange The problem you describe (mDns clients not visible between WiFi and Wired) is a function of your Router.  It is isolating WiFi from your wired network. The broadcast packet from your ESP32 is being intercepted by your router and not forwarded to the Wired network.
 [wiki mDns](https://en.wikipedia.org/wiki/Multicast_DNS)
[image]

Chuck.

bill-orange, 2018-03-17:
That makes sense. I have discovered several oddities with my time warner router.

JohnGEyles, 2018-03-29:
Whenever Spectrum (aka. Time Warner) finally brings service to my house, I hope to just use a simple modem from them and use my Airport Extreme for router duties.

everslick, 2018-04-20:
Can this issue be closed?

closed 2018-04-20 by everslick