espressif/arduino-esp32 #1161: installation problem "AttributeError: function 'CancelIoEx' not found"

djwfly, 2018-03-01:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
For a few days I have been installing the arduino-esp32 development enviroment  again and again. But failed all the time. The esp32 module could been found in the arduino. But when compile and upload , it failed...  The error message is as below:

Archiving built core (caching) in: C:\DOCUME~1\dell\LOCALS~1\Temp\arduino_cache_197603\core\core_espressif_esp32_esp32_FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_2b738b201fc88a468b2acb947ad43f23.a
Traceback (most recent call last):

  File "esptool.py", line 25, in <module>

  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 389, in load_module

    # executes the module code. This is crucial because the module

  File "site-packages\serial\__init__.py", line 27, in <module>

  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 389, in load_module

    # executes the module code. This is crucial because the module

  File "site-packages\serial\serialwin32.py", line 15, in <module>

  File "c:\python27\Lib\site-packages\PyInstaller\loader\pyimod03_importers.py", line 389, in load_module

    # executes the module code. This is crucial because the module

  File "site-packages\serial\win32.py", line 182, in <module>

  File "ctypes\__init__.py", line 375, in __getattr__

  File "ctypes\__init__.py", line 380, in __getitem__

 AttributeError: function 'CancelIoEx' not found

Failed to execute script esptool

exit status -1

  My PC is windows xp sp3,  python  python 2.7.  I have tried the following methods:
 -----change pyserial from 2.7 to pyserial3.0.1;
 -----change arduino from version 1.8.2 to 1.8.5;
 -----change the directory  from" C:\Documents and Settings\dell\My Documents\Arduino\hardware \espressif\esp32" to "C:\Program Files\Arduino\hardware\espressif\esp32"
      
All the time it failed. This trouble me for a few days.  Anyone know how to resolve this problem, thanks very much!

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

beegee-tokyo, 2018-03-01:
Maybe you can try this [alternative innstallation for amArduino IDE](http://desire.giesecke.tk/index.php/2018/02/27/alternative-to-install-arduino-esp32-for-arduino-ide-windows/)

lbernstone, 2018-03-01:
I would also recommend to open a CMD prompt as administrator and try to run `python esptool.py chip_id` from the tools directory.

djwfly, 2018-03-02:
Hello beegee-tokyo , I reinstall according to your suggestion.  Step by step I follow the instruction. all is carried out,  but at the last step "run 'get.exe' in git bash " ,this could not be carried out with  the message "command not found". So I have to double click 'get.exe' in the directory so as to complete the install.
   Then I compile and upload a exemple, the same error appear.  Oh this make me crazy... 
  But when

beegee-tokyo, 2018-03-03:
Maybe it is because of Win XP? The above procedure worked for me. I made only one small error in my Wiki entry. It should be `./get.exe` not `get.exe`.
Did you try what lbernstone suggested and what is the result?

djwfly, 2018-03-03:
I try ./get.exe, not succeed,  get the follwing message. Maybe it is because of my windows xp 32Bit .

$ ./get.exe
System: Windows, Info: Windows-XP-5.1.2600-SP3
Platform: i686-mingw32
Downloading xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
Done
Extracting xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
Downloading esptool-4dab24e-windows.zip
Done
Extracting esptool-4dab24e-windows.zip
Downloading mkspiffs-0.2.1-windows.zip
Traceback (most recent call last):
  File "get.py", line 139, in <module>
    get_tool(tool)
  File "get.py", line 89, in get_tool
    r = requests.get(url)
  File "site-packages\requests\api.py", line 70, in get
  File "site-packages\requests\api.py", line 56, in request
  File "site-packages\requests\sessions.py", line 488, in request
  File "site-packages\requests\sessions.py", line 630, in send
  File "site-packages\requests\sessions.py", line 190, in resolve_redirects
  File "site-packages\requests\sessions.py", line 609, in send
  File "site-packages\requests\adapters.py", line 487, in send
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='github-production-release-asset-2e65be.s3.amazonaws.com', port=443): Max retries exceeded with url: /35944333/48db9c96-affb-11e7-93fc-2cf04816adee?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180303%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180303T033448Z&X-Amz-Expires=300&X-Amz-Signature=70dd3d03c64760d9a8117aa618ad4603779c398f7f640bd016ecc0305daa0380&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dmkspiffs-0.2.1-windows.zip&response-content-type=application%2Foctet-stream (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x01603390>: Failed to establish a new connection: [Errno 10060] ',))
Failed to execute script get

 I tried  lbernstone's suggestion,aslo failed.  My pc is very old and i have no budget to change another one.  Maybe have to use arduino esp32  later  or eclipse instead.

beegee-tokyo, 2018-03-03:
It seems that the toolchain and esptool are downloaded but then it fails downloading the mkspiffs tool.
But Arduino IDE should still work even without the mkspiffs tool (I guess).
If you still want to try, I can give you a link to download the complete folder with everything. You would just need to unzip it and copy it into your <Sketchbook Folder>. But the compressed file is still ~380MB.

Or you can try to work with another IDE like Atom or Visual Studio Basic (my preference). There the whole installation process is easier, but still no guarantee that it works on XP.

djwfly, 2018-03-03:
Yes, the arduino still work ,I try it on esp8266 chip.  I want to  have a try . Can you  give me the link of the complete folder. The visual studio have higher requiement not only PC but also knowledge. I am a mechanical engineer and embed MCU  is my hobby . I have limited  knowledge and arduino is more convenient.

beegee-tokyo, 2018-03-03:
[Link](desire.giesecke.tk/uploads/hardware.zip) The content should be in `<Sketchbook-folder>\hardware`.
If you need some tutorials check out [ESP32 weatherstation](http://desire.giesecke.tk/index.php/2018/01/29/esp32-weatherstation/). I have examples for I2C, SPI, Bluetooth, BLE, UDP, TCP and others in it.

JamesGKent, 2018-03-09:
the issue with esp32 on windows xp is get.exe, esptool.exe are basically a compiled version of the python scripts, they've been built with a new version of pyserial library that does not work on windows XP.
you could do what i did and modify the platform.txt file so that it uses python like the linux version does.
note you'll need to have a version of python and the pyserial library installed that support XP (python <=3.4, pyserial 2.7)
the mods i made:
```
line 9: tools.esptool.cmd.windows=esptool.py
line 81: recipe.objcopy.hex.pattern=py "{tools.esptool.path}/{tools.esptool.cmd}" --chip esp32 elf2image --flash_mode "{build.flash_mode}" --flash_freq "{build.flash_freq}" --flash_size "{build.flash_size}" -o "{build.path}/{build.project_name}.bin" "{build.path}/{build.project_name}.elf"
line 98: tools.esptool.upload.pattern=py "{path}/{cmd}" --chip esp32 --port "{serial.port}" --baud {upload.speed}  --before default_reset --after hard_reset write_flash -z --flash_mode {build.flash_mode} --flash_freq {build.flash_freq} --flash_size detect 0xe000 "{runtime.platform.path}/tools/partitions/boot_app0.bin" 0x1000 "{runtime.platform.path}/tools/sdk/bin/{build.boot}.bin" 0x10000 "{build.path}/{build.project_name}.bin" 0x8000 "{build.path}/{build.project_name}.partitions.bin"
```

note i used "py" at the start instead of python as i have 3 versions installed and none on the path.
however if you only have one version installed you can put it on the path and use "python"
or if you have several like me and want to force a particular version you could use "py -3" to force python 3, or "py -3.3" to force python 3.3 etc....

this should let you compile at lease, but doesn't fix the missing mkspiffs...

beegee-tokyo, 2018-03-10:
@JamesGKent Thanks for the solution for Windows XP. I added your solution to the [Wiki](http://desire.giesecke.tk/index.php/2018/03/10/esp32-framework-installation-on-windows-xp/)

djwfly, 2018-03-10:
Thanks very much.

everslick, 2018-04-20:
Can this issue be closed?

closed 2018-07-28 by copercini

SviMik, 2019-11-15:
You'll need an older version of esptool.py too, version from [Jan 23, 2018](https://github.com/espressif/esptool/blob/2ec0bc0d58b20f441cafb2a706ba6bf028176b86/esptool.py) worked for me.
The new version requires pyserial 3, and with 2.7 it will produce an error like
``AttributeError: 'Serial' object has no attribute 'dtr'``