espressif/arduino-esp32 #1164: btstack integration

fesch, 2018-03-02:
Hi

I wanted to try to integrate btstack (from bluekitchen) into arduino-esp32. I've come so far to compile the libbstack.a library and grep out nearly all h files, but I'm facing now the problem that I do not know where to put the file inside the hardware\espressif\esp32 so that my projet could find them

I've tried different places, but no chance so far.

Any hints?

fesch, 2018-03-02:
Adding "-I{compiler.sdk.path}/include/btstack" to the file platforms.txt made it ...

closed 2018-03-02 by fesch

copercini, 2018-03-02:
Are you planning to use it for RFCOMM/SPP ?

fesch, 2018-03-02:
Yes, that was the idea.

So far so good, but sub h files seams not to get included included automatically ... :(

copercini, 2018-03-02:
Take a look here: https://github.com/espressif/arduino-esp32/pull/1144
I can give you a help if you want to test it

fesch, 2018-03-02:
Cool! I did not find that one yet but will take a look at it!! Thanks :)

fesch, 2018-03-03:
Thanks, I've got your library up and running!
That is exactly what I was looking for ...

beegee-tokyo, 2018-03-03:
And if you want to play with Bluetooth and BLE running in parallel, I wrote a small [tutorial](http://desire.giesecke.tk/index.php/2018/02/27/initialize-bluetooth-serial-and-ble-server/)

fesch, 2018-03-03:
Thanks for the link.

BLE is not needed in my project, at least not yet. But who knows what the future brings ;-)

EdanPotter, 2019-07-18:
Any news about btstack integration?