espressif/arduino-esp32 #1171: Not able to use SoftwareSerial pins

beerni, 2018-03-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDEPlatform.io IDF component
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
I have a project running on an Arduino Uno and it works fine. The project use the SoftwareSerial pins 2 and 3 from Arduino Uno. 

When I migrate the project to the ESP32 Dev Module I am unable to use the serial ports. I connected the TD wire to TX0 pin and the RD wire to RX0. 

If I use the SoftwareSerial library a message error is thrown (see debug messages). How can I solve this problem? 

Thanks in advance 

### Sketch:
```cpp

//Change the code below by your sketch
#include <DYE_Fingerprint.h>
#include <SoftwareSerial.h>
SoftwareSerial mySerial(2, 3);

DYE_Fingerprint finger = DYE_Fingerprint(&mySerial);

void setup()  
{
  Serial.begin(9600);
  while (!Serial);
  delay(100);
  Serial.println("\n\nfinger detect test");

  // set the data rate for the sensor serial port
  finger.begin(57600);
  
  if (finger.verifyPassword()) {
    Serial.println("Found fingerprint sensor!");
  } else {
    Serial.println("Did not find fingerprint sensor :(");
    while (1) { delay(1); }
  }
  Serial.println("Waiting for valid finger...");
}

void loop()                     // run over and over again
{
  getFingerprintIDez();
  delay(50);            //don't ned to run this at full speed.
}

uint8_t getFingerprintID() {
  uint8_t p = finger.getImage();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image taken");
      break;
    case FINGERPRINT_NOFINGER:
      Serial.println("No finger detected");
      return p;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Communication error");
      return p;
    case FINGERPRINT_IMAGEFAIL:
      Serial.println("Imaging error");
      return p;
    default:
      Serial.println("Unknown error");
      return p;
  }

  // OK success!

  p = finger.image2Tz();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Image converted");
      break;
    case FINGERPRINT_IMAGEMESS:
      Serial.println("Image too messy");
      return p;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Communication error");
      return p;
    case FINGERPRINT_FEATUREFAIL:
      Serial.println("Could not find fingerprint features");
      return p;
    case FINGERPRINT_INVALIDIMAGE:
      Serial.println("Could not find fingerprint features");
      return p;
    default:
      Serial.println("Unknown error");
      return p;
  }
  
  // OK converted!
  p = finger.fingerFastSearch();
  if (p == FINGERPRINT_OK) {
    Serial.println("Found a print match!");
  } else if (p == FINGERPRINT_PACKETRECIEVEERR) {
    Serial.println("Communication error");
    return p;
  } else if (p == FINGERPRINT_NOTFOUND) {
    Serial.println("Did not find a match");
    return p;
  } else {
    Serial.println("Unknown error");
    return p;
  }   
  
  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID); 
  Serial.print(" with confidence of "); Serial.println(finger.confidence); 

  return finger.fingerID;
}

// returns -1 if failed, otherwise returns ID #
int getFingerprintIDez() {
  uint8_t p = finger.getImage();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.image2Tz();
  if (p != FINGERPRINT_OK)  return -1;

  p = finger.fingerFastSearch();
  if (p != FINGERPRINT_OK)  return -1;
  
  // found a match!
  Serial.print("Found ID #"); Serial.print(finger.fingerID); 
  Serial.print(" with confidence of "); Serial.println(finger.confidence);
  return finger.fingerID; 
}
```

### Debug Messages:
```
fingerprint.ino:2:28: fatal error: SoftwareSerial.h: No such file or directory
compilation terminated.
```

beegee-tokyo, 2018-03-04:
Why do you want to use SoftwareSerial? The ESP32 has 3 hardware serial ports. Use serial # 2 instead.

beerni, 2018-03-04:
I am a newbie so I don't have a lot of knowledge. Can you explain me why?

capedra, 2018-06-16:
@beegee-tokyo whenever you have some board with PSRAM, serial # 2 pins are used for it (CLK and CS of PSRAM), while serial # 1 is used internally for flash memory and according to Professor Andreas Spiess (@SensorsIot) we should avoid using this UART. Sooo, UART0 is our only choice. Maybe we should still consider using SoftwareSerial, I guess.

capedra, 2018-06-16:
Even following this tutorial: https://www.reddit.com/r/hackerboxes/comments/5xba6g/softwareserial_working_for_esp32arduino_ide/
it's still not working.

lbernstone, 2018-06-16:
@capedra : The default pins may be used, but the UART registers are not.  serial can be mapped to whatever pins you would like, and having real buffers is typically better than bit banging serial, especially on a multitasked system. 
@beerni : You did not give us any info on this library you are using, which would make it difficult to help.  [Lady Ada's library](https://github.com/adafruit/Adafruit-Fingerprint-Sensor-Library) defaults to using HarwareSerial on esp32.  That might be a better fit if compatible with your hardware.

closed 2018-09-16 by beerni