espressif/arduino-esp32 #1174: Implement std::function variants of WiFi.onEvent similar to ESP8266

PhracturedBlue, 2018-03-04:
### Description:
Could we get equivalent functions as the ESP8266 has for WiFi events:
```
WiFiEventHandler onStationModeConnected(std::function<void(const WiFiEventStationModeConnected&)>);
WiFiEventHandler onStationModeDisconnected(std::function<void(const WiFiEventStationModeDisconnected&)>);
WiFiEventHandler onStationModeAuthModeChanged(std::function<void(const WiFiEventStationModeAuthModeChanged&)>);
WiFiEventHandler onStationModeGotIP(std::function<void(const WiFiEventStationModeGotIP&)>);
WiFiEventHandler onStationModeDHCPTimeout(std::function<void(void)>);
WiFiEventHandler onSoftAPModeStationConnected(std::function<void(const WiFiEventSoftAPModeStationConnected&)>);
WiFiEventHandler onSoftAPModeStationDisconnected(std::function<void(const WiFiEventSoftAPModeStationDisconnected&)>);
```
These functions allow using std::function, which allows handling events in class functions cleanly.

sticilface, 2018-04-11:
I would second this. :)

timpur, 2018-04-26:
Any progress on this? Thinking of implementing.

tripflex, 2018-05-07:
Yes I third this, not only for functionality but also for easy porting of ESP8266 projects over to ESP32 👍

teprom, 2018-09-09:
Any update on this ?

timpur, 2018-09-09:
Think this now supported with #1366. See https://github.com/timpur/arduino-esp32/blob/d72e8137b3933470bfa8292e801389064fd8d001/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino#L78

ristomatti, 2018-10-12:
This probably would not solve the portability of ESP8266 projects @tripflex mentioned as there's not separate functions for each event but at least I find this new approach much more versatile and clean.

wfraga18, 2019-03-07:
I need too

closed 2019-08-15 by stale[bot]

alvarolb, 2021-08-10:
> Think this now supported with #1366.
> 
> See this for example: https://github.com/timpur/arduino-esp32/blob/d72e8137b3933470bfa8292e801389064fd8d001/libraries/WiFi/examples/WiFiClientEvents/WiFiClientEvents.ino#L78

Still not have std::function callbacks on events.

```cpp
   WiFiEventId_t eventID = WiFi.onEvent([](WiFiEvent_t event, WiFiEventInfo_t info){
        Serial.print("WiFi lost connection. Reason: ");
        Serial.println(info.disconnected.reason);
    }, WiFiEvent_t::SYSTEM_EVENT_STA_DISCONNECTED);
```

In this example there is a lambda without any capturing parameters, which C++ converts to a standard function pointer.

Setting the same lambda with capture by reference will not compile.

```cpp
   WiFiEventId_t eventID = WiFi.onEvent([&](WiFiEvent_t event, WiFiEventInfo_t info){
        Serial.print("WiFi lost connection. Reason: ");
        Serial.println(info.disconnected.reason);
    }, WiFiEvent_t::SYSTEM_EVENT_STA_DISCONNECTED);
```