espressif/arduino-esp32 #1177: missing files that are there

PeterAlfonsLoch, 2018-03-05:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Whatever I try to compile I get error messages of missing files, altho they all are there

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Arduino: 1.8.5 (Windows 10), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None"

In file included from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32/esp_assert.h:17:0,

                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/soc/soc/soc.h:20,

                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32/rom/ets_sys.h:21,

                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h:120,

                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos/freertos/FreeRTOS.h:99,

                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:32,

                 from sketch\Blink_sos.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/assert.h:23:25: fatal error: assert.h: No such file or directory

compilation terminated.

exit status 1
Error compiling for board Adafruit ESP32 Feather.
```

copercini, 2018-03-05:
> Core Installation/update date: ?11/jul/2017?

Your core is almost a year outdated, try to update it: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md#how-to-update-to-the-latest-code

PeterAlfonsLoch, 2018-03-05:
I did do that.

closed 2019-08-15 by stale[bot]