espressif/arduino-esp32 #1179: Remove a connected Wifi Station

PhilColbert, 2018-03-05:
Anyone know how to remove a wifi device thats connected to the esp32 ?

Struggling y to find a method to disconnect.

This is not disconnecting the esp32 as a station, its disconnecting another devices thats connected to the esp32.

Thanks

beegee-tokyo, 2018-03-05:
Give some more details. How is the esp32 connecting? HTTP, TCP or what else?

PhilColbert, 2018-03-05:
The esp32 isnt connecting.

Wifi devices are connecting to the esp32, just connecting to the access point on the esp32. ( http/tcp irrelevant at this point )

Problem is if a client connects to the access point on the esp32 while the wifi is coming up, it doesnt assign an IP, stays at 0.0.0.0

If this client can be ejected so it reconnects and gets a valid IP this would be good.

Thanks

beegee-tokyo, 2018-03-06:
Ah, never had AP active for a longer time (only for testing). And on Arduino level there is afaik no function to disconnect a station from the access point. Maybe there is something in the ESP-IDF functions?

@me-no-dev I think that this could be a bug in the ESP-IDF. Because the AP should not accept connections before it is fully up and running and able to assign IP addresses.

PhilColbert, 2018-03-06:
Have had a look, cant see much, cant see a way to stop connection before dhcp server comes up .

I think its the client connecting in between the wifi coming up and the dhcp server coming up.

I (3064) wifi: mode : softAP (30:ae:a4:6e:f2:01)
SYSTEM_EVENT_STA_GOT_IP!!!!!!!!!!!!
 I (3071) event: sta ip: 192.168.4.1, mask: 2 55.255.255.0, gw: 192.168.4.1
SYSTEM_EVENT_STA_GOT_IP

I (3840) wifi: n:3 0, o:3 0, ap:3 1, sta:255 255, prof:3
I (3841) wifi: station: 70:25:59:01:89:68 join, AID=1, g, 20

infoList.num :-1
Client connected :-0.0.0.0

infoList.num :-1
Client connected :-0.0.0.0

infoList.num :-1
Client connected :-0.0.0.0

If I then kick off the stations :- esp_wifi_deauth_sta(0);

The station comes back and .....gives it the same 0.0.0.0 IP ! :(

If I then wait and manually disconnect and reconnect the wifi  the station is assigned an IP.

However, this only seems to happen on a kobo ereader and not on an android device, perhaps its the way the ereader connects to the Access point, but it should never be assigned a 0.0.0.0 IP should it ?

Thanks

PhilColbert, 2018-03-06:
Bit more info, the device that connects to the esp32 access point, is actually assigned an IP, and then you can use that IP if you manually assign use it, however the station list still comes back as 0.0.0.0

Thanks

lbernstone, 2018-03-06:
As a workaround, perhaps you can set the lease time to seconds until the AP stabilizes, and then set it higher.  The affected machine(s) should request a new address then.

PhilColbert, 2018-03-06:
Thats an idea, however, when you kick the station off it comes back again and requests and gets the same 0 IP :( worth a try tho !
thanks

closed 2018-03-15 by PhilColbert