espressif/arduino-esp32 #1180: WEMOS LOLIN32 - "leaving hard resetting" Why am I getting this error when uploading.

twdodson67, 2018-03-05:
Sketch uses 551374 bytes (42%) of program storage space. Maximum is 1310720 bytes.
Global variables use 37472 bytes (12%) of dynamic memory, leaving 257440 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5461.3 kbit/s)...
Hash of data verified.
Compressed 12304 bytes to 8126...

Writing at 0x00001000... (100 %)
Wrote 12304 bytes (8126 compressed) at 0x00001000 in 0.7 seconds (effective 135.2 kbit/s)...
Hash of data verified.
Compressed 552528 bytes to 315957...

Writing at 0x00010000... (5 %)
Writing at 0x00014000... (10 %)
Writing at 0x00018000... (15 %)
Writing at 0x0001c000... (20 %)
Writing at 0x00020000... (25 %)
Writing at 0x00024000... (30 %)
Writing at 0x00028000... (35 %)
Writing at 0x0002c000... (40 %)
Writing at 0x00030000... (45 %)
Writing at 0x00034000... (50 %)
Writing at 0x00038000... (55 %)
Writing at 0x0003c000... (60 %)
Writing at 0x00040000... (65 %)
Writing at 0x00044000... (70 %)
Writing at 0x00048000... (75 %)
Writing at 0x0004c000... (80 %)
Writing at 0x00050000... (85 %)
Writing at 0x00054000... (90 %)
Writing at 0x00058000... (95 %)
Writing at 0x0005c000... (100 %)
Wrote 552528 bytes (315957 compressed) at 0x00010000 in 28.1 seconds (effective 157.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 1365.3 kbit/s)...
Hash of data verified.

**_Leaving...
Hard resetting..._**

dpharris, 2018-03-05:
A hard reset is normal and not an error, it indicates that the system is
rebooting the board to run your application.

David

twdodson67, 2018-03-05:
Thanks. Based on your response I can infer that the sketch made it to board.  The concern is why would it take soooooo long to reset. Like many minutes and nothing happens.  Never times out.  Nothing shows up on serial monitor.  I've run the script before.   Never taken this long.  It just started happening with my Heltec LoRa too.  

Tried multiple ports.

twdodson67, 2018-03-05:
I also tried manually resetting at the board while waiting for the "hard reset"

dpharris, 2018-03-05:
Try a basic, small app.  Also, try a manual reboot and see if that works.
Put in some debug print statements to see if the app has started.  Perhaps
you have used too many resources?

David

twdodson67, 2018-03-06:
Copy that.  Doubling down on trouble shooting.  Status to follow.

twdodson67, 2018-03-06:
Working now!!.  Had to ensure that the serial monitor was *open as the
sketch was uploaded*.  For my sketch, the serial message contained WEMOs IP
address.  This message posts *once and immediately after its uploaded*, if
you dont have your serial monitor open, you miss the message and you gotta
re-Upload the sketch,.

The statement   *Leaving... Hard resetting...* continued to display in the
message/ status box (red text at bottom of screen) but had *no* impact on
the boards' operation - just like David said it shouldn't.

Once I eventually figured out that I needed to leave the serial monitor
open while uploading sketch, the message appeared (aaahhh), I grabbed the
IP address, dropped it into Chrome  browser bar, and bada-bing!

Thanks David for stayin'  on it!.

Takaway:

-   Serial monitor will not store messages printed to it when not opened.
Seems sort of obvious now . . .
-  If you are like me and slow to open your monitor - code in a little
delay - so you can relax, take your time.
-  "Leaving . . . Hard resetting" message, not so bad after all.   Actually
a good thing. . .
-  The error message that I posted turns out to be an example of a
perfectly executed upload!  Ooopsy!

dpharris, 2018-03-06:
Good news ... you did it!

David

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

MeghaUkkali9, 2019-02-07:
Sketch uses 197112 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 12944 bytes (3%) of dynamic memory, leaving 314736 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6-beta1
Serial port COM3
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, Coding Scheme None
MAC: 24:0a:c4:08:56:6c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.1 kbit/s)...
Hash of data verified.
Compressed 17664 bytes to 11528...

Writing at 0x00001000... (100 %)
Wrote 17664 bytes (11528 compressed) at 0x00001000 in 1.0 seconds (effective 136.4 kbit/s)...
Hash of data verified.
Compressed 197280 bytes to 99081...

Writing at 0x00010000... (14 %)
Writing at 0x00014000... (28 %)
Writing at 0x00018000... (42 %)
Writing at 0x0001c000... (57 %)
Writing at 0x00020000... (71 %)
Writing at 0x00024000... (85 %)
Writing at 0x00028000... (100 %)
Wrote 197280 bytes (99081 compressed) at 0x00010000 in 8.9 seconds (effective 177.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 877.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

MeghaUkkali9, 2019-02-07:
Sketch uses 197112 bytes (15%) of program storage space. Maximum is 1310720 bytes.
Global variables use 12944 bytes (3%) of dynamic memory, leaving 314736 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6-beta1
Serial port COM3
Connecting........__
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.1 kbit/s)...
Hash of data verified.
Compressed 17664 bytes to 11528...

Writing at 0x00001000... (100 %)
Wrote 17664 bytes (11528 compressed) at 0x00001000 in 1.0 seconds (effective 136.4 kbit/s)...
Hash of data verified.
Compressed 197280 bytes to 99081...

Writing at 0x00010000... (14 %)
Writing at 0x00014000... (28 %)
Writing at 0x00018000... (42 %)
Writing at 0x0001c000... (57 %)
Writing at 0x00020000... (71 %)
Writing at 0x00024000... (85 %)
Writing at 0x00028000... (100 %)
Wrote 197280 bytes (99081 compressed) at 0x00010000 in 8.9 seconds (effective 177.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 877.7 kbit/s)...
Hash of data verified.

**Leaving...
Hard resetting via RTS pin...**

twdodson67, 2019-02-08:
Make sure your serial port is open when you upload.   See previous thread.

Thushararatnayake, 2019-02-09:
I am trying to upload "0V7670_BMP_MQTT" schetch to a camara bult with WEMOS LOlin32 with the serial monitor open, but still the picture does not appear in the OLED screen. (The camara is not working) The serial monitor just stops after displaying the text below. The same text will repeat when I reset the board. I have given my Wi Fi ssid and password but have commented the MQTT for the moment. Could you trace the problem?

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698
Connecting Wifi...
0.0.0.0
Reset display.
Server start

Dazza-W, 2019-06-11:
just getting back into some electronics/coding and this time around got one of those ESP01 UART "programmers" (which don't program.. without altering.. that's a whole rant in itself..LOL) 
anyway.. last night I was getting this "hard reset" as the last line on upload as well.. 
I assumed it was "normal" and the esp8266 was faulty as blink and wifiscan wouldn't seem to start although they looked to load fine, so tried 3 others.. same NULL result..  slept onit.. this morning I googled the hard reset and found this thread while trying upload again .. then even though I have a momentary button in the circuit to put the esp into flash mode I thought what if I power the esp/programmer down right on end so it doesn't reboot .. so flicked power to it off right as I saw "hard reset"  then repowered and the sketch is working
Not sure if this will help others.. but it seems to work for me... 2 diff esp boards .. 2 diff sketches  
:)

barath83, 2019-07-13:
I tried leaving Serial monitor open while uploading it isn't helping and sometimes shows a error
c:/users/barath/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open output file C:\Users\Barath\AppData\Local\Temp\arduino_build_125755/Ultrasonic_Sensor.ino.elf: Permission denied

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

tangguh102, 2019-08-13:
how to setting serial port to open ?
and where the status to see serial port is already open ?

lbernstone, 2019-08-13:
https://www.arduino.cc/en/guide/environment#toc12

tangguh102, 2019-08-13:
i still don't get it

lbernstone, 2019-08-13:
https://www.youtube.com/results?search_query=arduino+ide+serial+monitor

sakthi123-s, 2019-09-21:
I am using nodemcu i am also get same kind of error 
Hard resetting via Rts pins how to slove it any suggestions

tangguh102, 2019-09-23:
you can try older version library of node MCU.
i'm using version 2.0.0

stickbreaker, 2019-09-23:
@sakthi123-s that message "leaving hard resetting" is just the success message that the uploader has uploaded the "new" code to flash and is restarting the ESP32 to start running the new code.  Not an Error.

Chuck.

VeeruSubbuAmi, 2019-12-10:
This article might help you to to understand anf fix the issue
https://electronicsinnovation.com/hard-resetting-via-rts-pin-fixed-explained/

Dragondream7903, 2024-02-04:
Same error, I just go to serial monitor an change baud rate from 19200 to 9600, and it worked for me. Give it a try.