espressif/arduino-esp32 #1181: BluetoothSerial library restart sketch when remote client reconnect

reaper7, 2018-03-05:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			05/03/2018 Lates GIT version
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Sketch starts normally, 
when I connect android client (Serial Bluetooth), I got printed text, everything works right.
Next I disconect android client, still sketch works ok
but when I reconnect android client one more time I got this message:
```
assertion "!list_is_empty(list)" failed: file "C:/msys32/home/Note/esp/esp-idf/components/bt/bluedroid/osi/list.c", line 82, function: list_front
abort() was called at PC 0x401318f3 on core 0
```
and esp32 restart

BTW...is there any possibility to detect if remote device is connected?
and write/print to this device only when connected?

### Sketch:
```
//This example code is in the Public Domain (or CC0 licensed, at your option.)
//By Evandro Copercini - 2018
//
//This example creates a bridge between Serial and Classical Bluetooth (SPP)
//and also demonstrate that SerialBT have the same functionalities of a normal Serial

#include "BluetoothSerial.h"

#if !defined(CONFIG_BT_ENABLED) || !defined(CONFIG_BLUEDROID_ENABLED)
#error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
#endif

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin(); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
}

void loop() {
  Serial.println("Some text");
  SerialBT.println("Some text");
  delay(1000);
}
```

### Debug Messages:
ESP Exception Decoder:
```
Decoding 10 results
0x401314db: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x4008a7f0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x4008a8ef: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x401314db: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x400d83ee: list_front at C:/msys32/home/Note/esp/esp-idf/components/bt/bluedroid/osi/list.c line 214
0x400dd6c2: btc_spp_cb_handler at C:/msys32/home/Note/esp/esp-idf/components/bt/bluedroid/btc/profile/std/spp/btc_spp.c line 469
0x400d9006: btc_task at C:/msys32/home/Note/esp/esp-idf/components/bt/bluedroid/btc/core/btc_task.c line 100
```

UART Debug
```
The device started, now you can pair it with bluetooth!
Some text
[I][BluetoothSerial.cpp:46] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:61] esp_spp_cb(): ESP_SPP_START_EVT
Some text
Some text
Some text
Some text
[1B][0;31mE (9958) BT: btm_sec_conn_req
[1B][0m
[1B][0;31mE (9970) BT: btm_sec_connected
[1B][0m
[I][BluetoothSerial.cpp:86] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
[1B][0;31mE (18321) BT: port_rfc_closed RFCOMM connection in state 2 closed: Closed (res: 19)[1B][0m
[I][BluetoothSerial.cpp:58] esp_spp_cb(): ESP_SPP_CLOSE_EVT
[1B][0;31mE (18336) BT: rfc_find_lcid_mcb LCID reused LCID:0x41 current:0x0[1B][0m
[1B][0;31mE (18336) BT: RFCOMM_DisconnectInd LCID:0x41[1B][0m
Some text
Some text
Some text
Some text
Some text
Some text
Some text
[1B][0;31mE (24899) BT: btm_sec_conn_req
[1B][0m
[1B][0;31mE (24909) BT: btm_sec_connected
[1B][0m
[I][BluetoothSerial.cpp:86] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT
Some text
[V][BluetoothSerial.cpp:82] esp_spp_cb(): ESP_SPP_WRITE_EVT
assertion "!list_is_empty(list)" failed: file "C:/msys32/home/Note/esp/esp-idf/components/bt/bluedroid/osi/list.c", line 82, function: list_front
abort() was called at PC 0x401318f3 on core 0

Backtrace: 0x4008a7f0:0x3ffd8610 0x4008a8ef:0x3ffd8630 0x401318f3:0x3ffd8650 0x400d879e:0x3ffd8680 0x400dda72:0x3ffd86a0 0x400d93b6:0x3ffd86f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
The device started, now you can pair it with bluetooth!
Some text
[I][BluetoothSerial.cpp:46] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:61] esp_spp_cb(): ESP_SPP_START_EVT
Some text
Some text
Some text
Some text
Some text
```

copercini, 2018-03-05:
it's an IDF bug that was solved today: https://github.com/espressif/esp-idf/issues/1537 so it will work here in the next IDF compilation or using arduino as IDF component

> BTW...is there any possibility to detect if remote device is connected?
and write/print to this device only when connected?

I will implement it soon =)

copercini, 2018-03-05:
> BTW...is there any possibility to detect if remote device is connected?
and write/print to this device only when connected?

Here it is: https://github.com/espressif/arduino-esp32/pull/1183
now you can use `SerialBT.hasClient()` to check it

reaper7, 2018-03-06:
**hasClient()** works 👍 

I hoped that it will solve my problem but no :(
```
  if (SerialBT.hasClient() == true) {
    Serial.println("Client connected");
    //SerialBT.println("some text");
  } else {
    Serial.println("Client disconnected");
  }

  delay(1000)
```
detection itself works very nice, client reconnecting do not cause resets
but only when print/write to bt serial is commented.

so...we are waiting for a final solution in IDF...

I leave this issue open

reaper7, 2018-03-27:
fix finally merged: https://github.com/espressif/esp-idf/issues/1537#issuecomment-376374138

copercini, 2018-04-03:
now it depends of @me-no-dev to recompile the IDF for arduino

copercini, 2018-04-07:
Done!
Update your ESP32 core to the last version and check it now

reaper7, 2018-04-07:
👍 done :)

closed 2018-04-07 by reaper7

pir0c0pter0, 2018-10-22:
is the password function merged? If yes, how to use to BluetoothSerial?