espressif/arduino-esp32 #1182: ios hotpost not found ?

tablatronix, 2018-03-05:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

### Hardware:
Board:						ESP32 doit dev
Core Installation/update date:	today
IDE name:				        platformio

### Description:
esp8266 scan and connect both work with my ios hotspot.
Esp32 does not see it in either.

### Debug Messages:
```
*WM: Connecting to saved AP: shawn
*WM: connectTimeout not set, ESP waitForConnectResult... 
WM: EVENT: WIFI_REASON: 201
*WM: Event: SYSTEM_EVENT_STA_DISCONNECTED, reconnecting
*WM: Connection result: WL_NO_SSID_AVAIL
*WM: lastconxresult: WL_NO_SSID_AVAIL
```

tablatronix, 2018-03-05:
Why would this be? Is there any way to debug this, or get information about why esp32 hardware is not compatible with adhoc networks ?

JacoFourie, 2018-03-05:
I can see the WiFi in the scan but it wont connect to the hot spot.

stickbreaker, 2018-03-05:
@tablatronix I had the same error `WL_NO_SSID_AVAIL` it turned out to be a corruption in the nvs partition of my ESP32.  #1147 my issue was solved by re-initing the nvs with this sketch provided by @lbernstone :
```c++
#include <nvs.h>
#include <nvs_flash.h>
void setup() {
    int err;
    err=nvs_flash_init();
    Serial.println("nvs_flash_init: " + err);
    err=nvs_flash_erase();
    Serial.println("nvs_flash_erase: " + err);
 }
```

Chuck.

tablatronix, 2018-03-05:
What is NVS?

tablatronix, 2018-03-05:
I just ran it, no change. :(

tablatronix, 2018-03-05:
hmm I see no nvs

```
Partition nvs found, 20480 bytes
page[0].State=0xffffffff  means UNINITIALIZED  Page[0] not active, State = 0xffffffff Seqnr=4294967295 
page[1].State=0xffffffff  means UNINITIALIZED  Page[1] not active, State = 0xffffffff Seqnr=4294967295 
page[2].State=0xffffffff  means UNINITIALIZED  Page[2] not active, State = 0xffffffff Seqnr=4294967295 
page[3].State=0xffffffff  means UNINITIALIZED  Page[3] not active, State = 0xffffffff Seqnr=4294967295 
page[4].State=0xffffffff  means UNINITIALIZED  Page[4] not active, State = 0xffffffff Seqnr=4294967295 

Logical Page to Physical Sector
Page 0 -> 255
Page 1 -> 255
Page 2 -> 255
Page 3 -> 255
Page 4 -> 255

```

tablatronix, 2018-03-05:
ok i wiped, reloaded a sketch , reran show nvs and it showed stuff.

But still no ap :(

tablatronix, 2018-03-05:
it works!, not sure what I did, I connected to some other aps first, not sure if that cleared out something or not.

tablatronix, 2018-03-05:
I added this to wifimanager

`// #define WM_ERASE_NVS // esp32 erase() function will erase NVS`

closed 2018-03-05 by tablatronix

stickbreaker, 2018-03-06:
@tablatronix I just encountered the same problem again, I have a MQTT sketch running, and I uploaded a new version, it failed to connect with error =1.
SHOW_nvs_keys:
```
// failed to reconnect with wl=1 after running mqtt.ino, then uploading new version

Partition nvs found, 20480 bytes
page[0].State=0xfffffffc INIT FULL  means FULL  Seqnr=0 
page[1].State=0xfffffffc INIT FULL  means FULL  Seqnr=1 
page[2].State=0xfffffffe INIT  means ACTIVE Sequence value InValid: page[2].Seqnr=845 
page[3].State=0xffffffff  means UNINITIALIZED  Page[3] not active, State = 0xffffffff Seqnr=4294967295 
page[4].State=0xfffffffc INIT FULL  means FULL Sequence value InValid: page[4].Seqnr=844 

Logical Page to Physical Sector
Page 0 -> 0
Page 1 -> 1
Page 2 -> 255
Page 3 -> 255
Page 4 -> 255
```
Erased NVS

Uploaded MQTT, now it connects first time.
```
Connecting to Belkin.464C
.....
WiFi connected
IP address: 
192.168.2.19
```
My MQTT sketch uses 
```
#include <WiFi.h>
#include <PubSubClient.h>
#include <OneWire.h> // my version
#include <SensorAverage.h> // my library
#include "mqtt.h"
#include <ESPmDNS.h>
#include <wd_c2401_i2c.h> // i2c my library 
```

One of these libraries is destroying NVS, my libraries don't access/change any nvs parameters,  I'll look through the rest.  I'm thinking ESPmDNS.h.  It is the last major library I have added That I don't  "own".

Chuck.

tablatronix, 2018-03-06:
I was wondering what this error was telling me
E (1122) wifi: wifi_init 1143 ret=4363.
Now i know

tablatronix, 2018-03-06:
I recently added mdns to wifimanager also, so perhaps that is worth investigating.