espressif/arduino-esp32 #1186: DNSServer / CaptivePortal Example not working

kabauterman, 2018-03-06:
Hello, I just tested my NodeMCU ESP32 and Wemos LoLin Boards with the DNSServer / CaptivePortal Example to see how it works and if I can use it in some of my projects.

I can't get it to respond you any DNS request.

Doesn't the DNS Server needs to by set via DHCP from the Station config first, I don't see any DNS Server Address given to the client.

Arduino IDE 1.8.10, todays latest arduino-esp32 code, tested with a OnePlus 5 Smartphone connected to the ESP32.

Hope some one can help.

Keep up with the great work, hope to get this working soon.

jwktje, 2018-03-06:
I've also been testing this use-case a bunch and running into weirdness. My progress is described here:
https://github.com/tzapu/WiFiManager/issues/545

jwktje, 2018-03-08:
That commit might be just around the time I started testing. So I'll pull to be sure

amrlsayed, 2019-08-15:
Hi,
I have the esp32 Arduino SDK (version 1.0.2)  downloaded from the Arduino IDE Board manager.
and have the problem that the "captive portal" example doesn't run on Windows 10 but works fine on Android. even using the browser.
I am not sure If I need to pull changes from Github ? or if its even possible as I got the SDK from the "Board Manager" this month. I assume downloading from Boardmanager is the same as cloning the master branch ? 
My main purpose is using the WiFiManager from @tzapu. I found it works fine elsewhere but windows 10.
Do you have any recommendations ?
Thanks a lot.

atanisoft, 2019-08-15:
@amrlsayed Windows requires slightly different checks than the Android devices do for CaptivePortals. I'd suggest using the WiFiManager from @tzapu instead of the example here. If the CaptivePortal checks inside the WiFiManager code fail I'd suggest open an issue on the issue tracker for that library.

tablatronix, 2019-08-15:
I have seen windows 10 firewall block captive portals.

amrlsayed, 2019-08-15:
@atanisoft Thank you, yes came back to "captive portal" example as @tzapu wifimanager library didn't work with me for windows 10. But works fine on other devices . ok i would open an issue there. I though it does make more sense to comment here when the example does not work in the first place.
@tablatronix Ok I will try to disable the firewall.
Thanks Guys for the fast response.

tablatronix, 2019-08-15:
Yes stick with the minimal sketch to reproduce, makes sense

amrlsayed, 2019-08-15:
@tablatronix  just for feedback I disabled the firewall and the problem still exists. I will open another issue in the WiFiManager Repo.
Thanks a lot

tablatronix, 2019-08-15:
Sure, I have not seen windows 10 issues reported in the other cp issues here so we can gather info there

closed 2019-10-28 by stale[bot]