espressif/arduino-esp32 #1189: No data in the 3rd UART buffer if you use all 3 the same time.

JacoFourie, 2018-03-06:
### Description:
When I use all 3 UARTS they will all 3 send data. But I can not receive any data on the 3rd UART.  I can read in serial0 and serial 1 but not on serial 2

### Sketch:

This is just an code snippet.

````

              String payload = "{";
              payload += "\"temperature\":"; payload += desp_temp; payload += ",";
              payload += "\"target\":"; payload += target_temp; payload += ",";
              payload += "\"wifi\":"; payload += wifi_quality; payload += ",";
              payload += "\"actuator\":"; payload += actuator_feedback;
              payload += "}";
                                                                                                
              // -------------------   Testing Code
              ModbusSerial.flush();
              ModbusSerial.println(payload);

              delay(500);
              
              //Read the data on the port
              int datalen = ModbusSerial.available();
              Serial.println(datalen);
              String SerialData;
              for (int z = 0; z < datalen; z ++) {
                  SerialData += (char)ModbusSerial.read();
              }

              Serial.print("Data from the Modbus Serial Port : ");
              Serial.println(SerialData);

              //--------------------------

````

Here is the paython code that is on the other side. Just for a test

````
# -*- coding: utf-8 -*-

import serial
import traceback
import time

# Define the serial port
ser = serial.Serial('COM5', 9600 , timeout = 15)
ser.parity = serial.PARITY_EVEN    
ser.flushInput()

#Set the run flag
run = True

#Run while the flag is set
while run:

    try:
        serial_line = ser.readline()
        ser.flushInput()
        print('Serial Data : ' + serial_line)
        print('Sending data to the ESP32')
        ser.write('Hallo Jaco !! ' + '\r\n')
        time.sleep(0.5)
		
    except KeyboardInterrupt:
        print "End"
        run = False        

    except:
        traceback.print_exc()    
	
````

### Output on the serial monitor
````
0
Data from the Modbus Serial Port : 
0
Data from the Modbus Serial Port :
0
Data from the Modbus Serial Port :
````

### Output on the Python Side

````
Serial Data : {"temperature":28.44,"target":0.00,"wifi":58,"actuator":30}

Sending data to the ESP32
Serial Data : {"temperature":28.44,"target":0.00,"wifi":50,"actuator":30}

Sending data to the ESP32
Serial Data : {"temperature":28.44,"target":0.00,"wifi":62,"actuator":30}
````

negativekelvin, 2018-03-07:
https://github.com/espressif/arduino-esp32/issues/1145

JacoFourie, 2018-03-07:
@negativekelvin I do not close the port. I also did comment on that issue.

negativekelvin, 2018-03-07:
I don't think it matters, begin calls flush and triggers a hardware bug.  Needs workaround.

PhilColbert, 2018-03-07:
Have you updated to the latest arduino ? Think that should fix data coming in.

However main issue is the data corruption on Serial2....  Anyone any ideas on how to make Serial 2 reliable ? 

Thanks

JacoFourie, 2018-03-07:
@PhilColbert  how do I check the version of the Core. I did do the Git pull and ran get.exe again. But I am not sure if it is the latest version.

JacoFourie, 2018-03-07:
I deleted all the zip files and then ran get.exe again.

this was the output.

````
System: Windows, Info: Windows-10-10.0.16299
Platform: i686-mingw32
Downloading xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
Done
Extracting xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip
Downloading esptool-4dab24e-windows.zip
Done
Extracting esptool-4dab24e-windows.zip
Downloading mkspiffs-0.2.1-windows.zip
Done
Extracting mkspiffs-0.2.1-windows.zip
Renaming mkspiffs-0.2.1-windows/ to mkspiffs
Done
````

PhilColbert, 2018-03-07:
Try running ESP.getSdkVersion()

Mine is v3.1-dev-439-g37765d00-dirty

I am running arduino as a component by the way.

JacoFourie, 2018-03-07:
@PhilColbert thanks for the reply. do I run that on the ESP32. what do you mean by "I am running arduino as a component by the way." ?

PhilColbert, 2018-03-07:
Installing ESPIDF and then adding the arduino software afterwards.

https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

JacoFourie, 2018-03-07:
Aaa OK so you are doing it the other way around. I am using the Arduino IDE.

This is my SDK version.  v3.1-dev-239-g1c3dd23f-dirty

It seems I am running and OLD SDK. How do I update it as I did follow the steps as stated here.

https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

PhilColbert, 2018-03-07:
I dont think you can do ?

It needs the guys running the arduino core to update?

I maybe wrong however !

Running with the component and then using Eclipse to code is better ... well it is for me :) plus you get more control over whats going on :)

JacoFourie, 2018-03-07:
Can I use all the Arduino Libaries on the ESPIDF side ? I have done a lot of work on the Arduino side that I would love to bring over if it will work.

PhilColbert, 2018-03-07:
Sure, but you can still use all the espidf stuff now in the arduino code.

Adding it as a component means you can do all this still.

JacoFourie, 2018-03-07:
Here is a little test app. This does not work.

````
#include <HardwareSerial.h>

HardwareSerial MySerial(1);
HardwareSerial ModbusSerial(2);

void setup() {
  Serial.begin(115200);
  MySerial.begin(115200, SERIAL_8N1, 16, 17);
  ModbusSerial.begin(9600 ,SERIAL_8E1, 4, 15);
  
}

void loop() {
  
  // -------------------   Testing Code
  ModbusSerial.flush();
  ModbusSerial.println("This is a test from the ESP32");
               
  //Read the data on the port
  int datalen = ModbusSerial.available();
  Serial.println(datalen);
  String SerialData;
  for (int z = 0; z < datalen; z ++) {
      SerialData += (char)ModbusSerial.read();
  }   
  Serial.print("Data from the Modbus Serial Port : ");
  Serial.println(SerialData);

  //--------------------------

  delay(2000);

}
````

Python Code
````
# -*- coding: utf-8 -*-

import serial
import traceback
import time

# Define the serial port
ser = serial.Serial('COM5', 9600 , timeout = 15)
ser.parity = serial.PARITY_EVEN    
ser.flushInput()

#Set the run flag
run = True

#Run while the flag is set
while run:

    try:
        serial_line = ser.readline()
        ser.flushInput()
        print('Serial Data : ' + serial_line)
        print('Sending data to the ESP32')
        ser.write('Hallo Jaco !! ' + '\r\n')
        ser.write('Hallo Jaco !! ' + '\r\n')
        ser.write('Hallo Jaco !! ' + '\r\n')
        #ser.write('')      
        time.sleep(0.5)
		
    except KeyboardInterrupt:
        print "End"
        run = False        

    except:
        traceback.print_exc()
````

This is the output on the ESP32 side
````
Data from the Modbus Serial Port : 
48
Data from the Modbus Serial Port : Hallo Jaco !! 
(o[2'R⸮⸮⸮$7{⸮ćh
⸮⸮ݨ⸮⸮W⸮⸮
0
Data from the Modbus Serial Port : 
0
Data from the Modbus Serial Port : 
0
Data from the Modbus Serial Port : 
48
Data from the Modbus Serial Port : M⸮⸮⸮⸮v⸮⸮⸮N6⸮J⸮l⸮-⸮qY3#曑n
````

and this is the Python side.

````
ending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32

Sending data to the ESP32
Serial Data : +³¿äXË╗£O├ä╦n6h┼Úö│úîbvFÏõ[╗§ÛÚ Ü▀¹Õ)ÝÇ@ß°y~y~y< ù5ûnpêsÞçñ¹"B¯õìıA░═jRÿ(³O╝{ÙqÙ}¿{$Ï▒÷This is a test from the ESP32
````

JacoFourie, 2018-03-07:
If I change the code to use UART 1 then the code works.  So ther eis a issue with UART 2 !!

This code will work
````

#include <HardwareSerial.h>

//HardwareSerial MySerial(1);
HardwareSerial ModbusSerial(1);

void setup() {
  Serial.begin(115200);
  //MySerial.begin(115200, SERIAL_8N1, 16, 17);
  ModbusSerial.begin(9600 ,SERIAL_8E1, 4, 15);
  
}

void loop() {
  
  // -------------------   Testing Code
  ModbusSerial.flush();
  ModbusSerial.println("This is a test from the ESP32");
               
  //Read the data on the port
  int datalen = ModbusSerial.available();
  Serial.println(datalen);
  String SerialData;
  for (int z = 0; z < datalen; z ++) {
      SerialData += (char)ModbusSerial.read();
  }   
  Serial.print("Data from the Modbus Serial Port : ");
  Serial.println(SerialData);

  //--------------------------

  delay(2000);

}
````

Here is the output on the ESP32 side.

````
48
Data from the Modbus Serial Port : Hallo Jaco !! 
Hallo Jaco !! 
Hallo Jaco !! 

48
Data from the Modbus Serial Port : Hallo Jaco !! 
Hallo Jaco !! 
Hallo Jaco !! 

48
Data from the Modbus Serial Port : Hallo Jaco !! 
Hallo Jaco !! 
Hallo Jaco !! 

48
Data from the Modbus Serial Port : Hallo Jaco !! 
Hallo Jaco !! 
Hallo Jaco !! 

48
Data from the Modbus Serial Port : Hallo Jaco !! 
Hallo Jaco !! 
Hallo Jaco !! 

48
Data from the Modbus Serial Port : Hallo Jaco !! 
Hallo Jaco !! 
Hallo Jaco !! 

48
Data from the Modbus Serial Port : Hallo Jaco !! 
Hallo Jaco !! 
Hallo Jaco !! 

````

and the Python side.

````
ending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32

Sending data to the ESP32
Serial Data : This is a test from the ESP32
````

negativekelvin, 2018-03-07:
Would suggest you try commenting out these lines: 
https://gist.github.com/negativekelvin/0ce857285fdd039b780d9ce7e606a58b/revisions

In arduino flush() is only supposed to empty the TX buffer so if you want to add another function to empty the RX buffer like flushRX() you will have to do it the way esp-idf does it 
https://github.com/espressif/esp-idf/blob/c97b8756f740297e23f3011603724005db22c557/components/driver/uart.c#L280-L291

@me-no-dev

PhilColbert, 2018-03-07:
Is it this issue ?

Note:
UART2 doesn’t have any register to reset Tx_FIFO or Rx_FIFO, and the UART1_TXFIFO_RST and UART1_RXFIFO_RST in UART1 may impact the functioning of UART2. Therefore, these 2 registers in UART1 should only be used when the Tx_FIFO and Rx_FIFO in UART2 do not have any data.

Problem is I have a GPS on Serial2 which is always sending in data..... Serial is always sending data out and Serial1 can have data in it most of the time....

PhilColbert, 2018-03-08:
Ok, changed the way flush works to the espidf way.....  so it empties the rx and tx buffer and doesnt do the reset.

This is ok for me for testing.

Removed the lines Kevin mentioned and added :+1: 

while(uart->dev->status.rxfifo_cnt != 0 || (uart->dev->mem_rx_status.wr_addr != uart->dev->mem_rx_status.rd_addr))
	{
        READ_PERI_REG(UART_FIFO_REG(uart->num));
    }
No corruption yet :) , still testing

JacoFourie, 2018-03-08:
I also commented out those 2 lines and the serial part seems to work now. I could get my Modbus sample program to read as well. But when I move it to the bigger program I dont get reads again. But in the bigger program I use threads on the 2 CPU's so I am not sure if that is a problem. I will take the sample modbus program and change it to use taks/threads as well and see if it still works.

@PhilColbert did you add that code to a function and call it before each send or how do you use it ?

PhilColbert, 2018-03-08:
Changes the way flush works to clear all RX out of the buffer, this is ok for me.

void uartFlush(uart_t* uart)
{

    if(uart == NULL) {
        return;
    }

    UART_MUTEX_LOCK();
    
     //uart->dev->conf0.txfifo_rst = 1;
     //  uart->dev->conf0.txfifo_rst = 0;
    
     //   uart->dev->conf0.rxfifo_rst = 1;
     //  uart->dev->conf0.rxfifo_rst = 0;
  
  
	while(uart->dev->status.rxfifo_cnt != 0 || (uart->dev->mem_rx_status.wr_addr != uart->dev->mem_rx_status.rd_addr))
	{
           READ_PERI_REG(UART_FIFO_REG(uart->num));
        }

    UART_MUTEX_UNLOCK();
}

Mine issue I think is that, on software restart, as we have removed the reset now, there was characters left in the buffer, so it gets confused and then characters come out randomly.

This clears it down completely and all serials can start with no data, as far as I can see, this works.

I  have done several software restarts and hard restarts and no issues yet.

However be aware that flush now clears RX :)

JacoFourie, 2018-03-08:
Mmmmm so all is working for me in the sample app without tasks. But as soon as I split it and run it in threads then UART2 wont read the data again. Aaaa this is so frustrating. Why is it such an issue just to get serial ports working.

PhilColbert, 2018-03-08:
To be honest I find it quite bizarre this issue hasnt come up before or sorted ?

Useful to be able to use all 3 serials and software restart.

Mine still running 100%, seems fixed for what I need, would like to see something added to the codebase to make this work ok for other people.

JacoFourie, 2018-03-09:
OK so I found what is cuasing the last UART not to have data in the buffer. But I dont know who to fix it.
In the main program use a OLED screen. I use the Adafruit driver.  as soon as I enable this driver using.
````
display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
````
then the I do not get any data in the read buffer from the last UART.
I think this screen uses I2C so maybe the UART and I2C is not working well together. 

@PhilColbert do you have a OLED secreen to test this with also ?

JacoFourie, 2018-03-09:
AAAAAAaaaaaaaaaaaaaaa  I am so sorry I found the issue. The screen OLED reset pin was set to pin 4 and that is the same pin as one of the UART pins that the last UART was set-up to use.
I am testing it now.

PhilColbert, 2018-03-09:
Ha :) that would do it ! 

Sounds like you have solved the issue, my setup is 100% now since changing the flush, all 3 UARTS work perfectly and software reset works.

Glad thats sorted, was getting worried at one point!

JacoFourie, 2018-03-09:
All seems to be working now 100%. Thanks all for the help.

closed 2018-03-09 by JacoFourie

PhilColbert, 2018-03-09:
Well its not really a closed issue? Its still broken and needs a proper fix coding and releasing doesnt it ?

JacoFourie, 2018-03-12:
@PhilColbert should we not open a new issue just for the UART flush issue?

PhilColbert, 2018-03-12:
think that would be a good idea , it does need fixing for other people
doesnt it ?

rrobinet, 2018-03-13:
Hi,
I would like to test this patch however it looks that something is missing. 
While compiling I have the following error:
```
... /hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c: In function 'uartFlush':
... /hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c:327:70: error: 'volatile union <anonymous>' has no member named 'wr_addr'
  while(uart->dev->status.rxfifo_cnt != 0 || (uart->dev->mem_rx_status.wr_addr != uart->dev->mem_rx_status.rd_addr))
                                                                      ^
.../hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c:327:106: error: 'volatile union <anonymous>' has no member named 'rd_addr'
  while(uart->dev->status.rxfifo_cnt != 0 || (uart->dev->mem_rx_status.wr_addr != uart->dev->mem_rx_status.rd_addr))
``` 
Any idea where `wr_addr `and `rx.addr`  should be defined  (in a newer version of Arduino/esspresif maybe ?

Thanks
Robert

negativekelvin, 2018-03-13:
>in a newer version of Arduino/esspresif maybe ?

Yes the Jan 17 update

rrobinet, 2018-03-13:
OK  I think this is the one I am using, however  it is the patched version stickbreaker to cope with the I2C (Wire) issue.
I will try to merge both versions and applied the UART patch again to see if it solves my UART2 issue.
Thanks for the tip but wait for the confirmation
Robert

rrobinet, 2018-03-13:
Great!!
I have installed the last version of Arduino-esp + the UART patch above and NO PROBLEM with UART2 anymore after ESP.restart() .
Thanks again
@PhilColbert  @me-no-dev 
NB: Can this patch becomes official?

PhilColbert, 2018-03-14:
From my point of view, it can and it cant, it does change the way flush works, to keep compatibility with arduino which flush means clear tx and not RX out then this change cant be added.

It needs more work to change the way serial starts to use this method instead of calling flush and then this is then left as it was ?

Someone with more knowledge of the full startup and flush of serial needs to sort :)

I have had my serial 2 running and rebooting for 75 hours now with no issues :)

JacoFourie, 2018-03-20:
Maybe a new function should be added to flush RX and leave to original flush ?

zamorae, 2018-04-16:
@JacoFourie @me-no-dev 
Hi guys,
I posted issue #1314 on the UART2 subject but it seems you have advanced it a lot on this thread. I see that there is some sort of workaround that consists on flushing or clearing the RX buffer  from UART2 by commenting some lines in a file but I really do not understand where should that happen. Is that commenting suppose to happen in the core files of ESP32 library or where? Can anybody point me out to the definition or to a small step by step guide on where to modify/comment the lines to get Serial2 working properly?

Thanks in advance
-EZ

zamorae, 2018-04-16:
Ok, I have modified the esp32-hal-uart.c in the cores library, but now I have the same error that @rrobinet talks about in a previous post, except that is marking 'rxififo'. 

This is the error at compile time:

``
C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-uart.c: In function 'uartFlush':

C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-uart.c:324:25: error: 'volatile union <anonymous>' has no member named 'rxfifo'

  while(uart->dev->status.rxfifo.cnt != 0 || (uart->dev->mem_rx_status.wr_addr != uart->dev->mem_rx_status.rd_addr))
``

So what is the correct way to implement this patch?

Thanks
-EZ

stickbreaker, 2018-04-16:
@zamorae the `uart` record is defined in [tools/sdk/include/soc/soc/uart_struct.h](https://github.com/espressif/arduino-esp32/blob/1da3138ed6bf4675836f6ebaa15bb5db280a6e36/tools/sdk/include/soc/soc/uart_struct.h#L146) 

instead of `uart->dev->status.rxfifo.cnt` try `uart->dev->status.rxfifo_cnt`

Chuck.

zamorae, 2018-04-16:
Hi Chuck @stickbreaker,
It seems that I have the definition of the command incorrect, thanks for pointing it out. I'll make the modification and report back.

Regards
-EZ

zamorae, 2018-04-17:
@me-no-dev  @JacoFourie  @PhilColbert @rrobinet  @stickbreaker 

Gents,
Help! I have implemented the UART modification in the uartFlush method of esp32-hal-uart.c as @stickbreaker recommended and is compiling correctly,  but I still have the issue. I have 3 chips interconnected (2 ESP32's and 1 ESP8266MOD) using Serials. 2 of them connected to the ESP32 central chip that must provide communications via 2 serials (UART1/UART2) to get and receive messages from the other 2 chips. In all cases:

a) I can get messages from the client chips sent to the central/master chip but the central cannot reply back
b) If I use the central chip as passthrough (getting data from Serial1 and pushing forward using serial2, I just can get one chip to receive data correctly and the other receives it with every odd letter missing
c) If I test the communications separately (only one chip and either Serial1 or Seria2 connected, not both) I can get messages back and forth corrrectly.

@PhilColbert @rrobinet @JacoFourie you state above that the patch is working correctly for you and that you have 100% comms on all 3 uarts and sfw reset.

Can you provide a laundry list of things that need to be considered in order to get all the patch/modification correctly? I just wonder if I could be missing some of the steps. Check #1314 if you want to see an example of the code I'm using. 
1) I have Arduino IDE 1.8.5 and I just recently installed (less than a month) the ESP32 version from github.
2)I am using ESP-WROOM-32 and communicating to it using ESP32 Dev Module. All seems to be working fine.

Let me know your thoughts.

Thanks in advance!
-EZ

negativekelvin, 2018-04-17:
You should not expect serial.read to return the same data when called twice.  You should not use flush with the patch to clear the RX buffer because you may discard unread data.

JacoFourie, 2018-04-17:
Have you tested each UART on its own one by one using a test app?
I made a little echo server using python that will send a string back when it gets data on the port.
Remember to do a flush before you send.

JacoFourie, 2018-04-17:
Here you can see my controller that I am porting to an ESP32.

https://www.youtube.com/watch?v=jYnhPLxoM1k

PhilColbert, 2018-04-17:
Yeah still working great for me, have a ereader on serial0, a custom built RF board on serial1 and a gps on serial2, all sending and receiving with no corruption, thankfully ! :)

zamorae, 2018-04-18:
Guys, thanks for the support!

I followed your recommendations, but I still can't understand why it's not working. I will share another sketch I defined that is not communicating correctly either, please point out any wrongdoing you find. I removed the flush() and the repeat Serial.write as mentioned by @negativekelvin. I am trying to achieve a echo test as proposed by @JacoFourie using only the ESP8266MOD and the central ESP32. using Serial 1 at (4,2) on the ESP32 and TX/RX as Serial1 on the ESP8266MOD, it seems that the ESP32 is echoing itself, even though I instructed to issue a different comment once the data is received from ESP32. I will share the sketch shortly.

zamorae, 2018-04-21:
Hi everyone,
I have concluded my testing using several approaches to the UART connection problem, and I think I found the issue: a loose ground cable that was putting noise into UART line 2. The strange thing is I returned the uart.flush() method located on the esp32-hal-uart.c file as it originally was and is still working correctly. I re-did the protoype using the devices I need and is working fine now. Thanks all for your support.

Regards
-EZ

vanvince, 2018-07-06:
Hello guys,
I have the same problem on UART2. I'm trying to connect an esp32 WROOM32 to an Arduino UNO WI-FI just for testing, I'm aiming to communicate with another board in future but I want to be sure about the code.
I've followed every step that you all had shown (comment the line in esp-hal-uart.c etc) but nothing is usefull. If I use UART0 everything is fine, I can send message to arduino and I can receive data from arduino. 
To be more clear here is the code:
esp32 wroom32 side
#include <HardwareSerial.h>

#define RX_PIN  16
#define TX_PIN  17

HardwareSerial cu(2);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  //cu.begin(115200,SERIAL_8N1,RX_PIN,TX_PIN);
  cu.begin(115200);
  
  delay(500);
  Serial.println("setup done!");
  Serial.print(ESP.getSdkVersion());

  cu.flush();
  
}

void loop(){
  // put your main code here, to run repeatedly:
  delay(100);
  
  if(cu.available()){
    Serial.print((char)cu.read());
    }
  
  if(Serial.available()){
    //uart mod
    cu.print(Serial.readString());
  }
  
}

arduino side
#include <SoftwareSerial.h>

SoftwareSerial cu(7,8);

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  cu.begin(115200);

  Serial.println("done!");
}

//UART0
void loop(){
  // put your main code here, to run repeatedly:
  if(Serial.available())
    cu.print(Serial.readString());

  if(cu.available())
    Serial.print(cu.readString());
  
  delay(100);

  
  }

I want to use UART2 for a simple reason, UART0 doesn't work with the gsp module that I aim to use, i don't know why. 

Regards,
Vince

JacoFourie, 2018-07-08:
All 2 UARTs work if you use the methods we showed above.

vanvince, 2018-07-12:
@JacoFourie  yeah, you're right. UART2 is working fine! Now I can send and receive data to and from esp32 with an Arduino UNO Wifi. 
However I'm trying to connect mine esp32 to a GPS/GSM module, in particula Ai-Thinker A7, by UART 2. I can read data from the GPS but I don't receive any data if I send an AT-command. 
For example I send through UART 2 just an "AT" (without the quotes) and wainting for the respons, in this cas an "OK", but it never comes. 
To be more clear here are the result over the serial port
to CU
from CU
at

The result "at" under "from CU" is an echo, I suppose. I wait for seconds but no one "OK" appear on the serial port. 
I tried also with an Arduino Mega2560, which has more hardware serial port as esp32, and it works good. 
The code is the same I posted some day ago.

Regards,
Vince

JacoFourie, 2018-07-12:
Break it down in smaller chunks. If you have it working on an Arduino then use just that code on a serial port that you know works on the ESP32 side. So get all the bits working on their own. Then you start to bring it together one by one as sometimes one thing will stop another from working. So break it down and the once you have it working them move it over. I also had an issue then I discovered I used the same pin of the screen reset as the data pin.

rrobinet, 2018-07-22:
@PhilColbert , @me-no-dev 
With a previous version of the espressif arduino-esp32  patched the with the PhilCobert (esp32-hal-uart.c) patch above, everything  was working find.
As I understand this was just a workaround expecting that this issue was fixed with the new set of libraries, so I decided to load the last version, which of course is generating several  new issues while compiling my sketches.
The easiest one (I expect!) is the fact that there is no way to compile a simple script that includes the HardwareSerial library anymore .
Typically
```
#include <HardwareSerial.h>
HardwareSerial Serial2(2);
void setup() {
}
void loop() {
}
```
or 
```
HardwareSerial Serial2(2);
void setup() {
}
void loop() {
}
```
generates

```
> /var/folders/bq/1_7b3x851wj3d2wxhqqntcnr0000gp/T/arduino_build_921324/arduino.ar(HardwareSerial.cpp.o):(.bss.Serial2+0x0): multiple definition of `Serial2'
> /var/folders/bq/1_7b3x851wj3d2wxhqqntcnr0000gp/T/arduino_build_921324/sketch/sketch_jul22a.ino.cpp.o:(.bss.Serial2+0x0): first defined here
> collect2: error: ld returned 1 exit status
> exit status 1
> Error compiling for board WEMOS LOLIN32.
```
In summary; what should I do to have a Serial 2 (Pins 16 & 17) working with the new release.
Thanks 
Robert

negativekelvin, 2018-07-22:
Serial1 and Serial2 are already defined so just skip to Serial2.begin

rrobinet, 2018-07-22:
Ok
Thanks I have also discovered that the hardwareserial definition was not necessary anymore
Robert