espressif/arduino-esp32 #1190: TTGOV2

choopk, 2018-03-07:
Has anyone manage to get the SD card library working with this (https://github.com/LilyGO/TTGO-LORA32-V2.0) board. Or is there any other library that you guys can recommend.

JamesGKent, 2018-03-08:
looking at the pinout for that board:
https://github.com/LilyGO/TTGO-LORA32-V2.0

and comparing to an example sketch:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino

it looks like you need to use the SD_MMC library, give that a try?

choopk, 2018-03-09:
Nope does not work

JamesGKent, 2018-03-09:
when you say doesn't work, what does happen? there is more than one way that sketch might "not work"
we would need to know more about what you are using and what it does. what type of SD card are you using? SDHC? SDXC? how is it formatted etc?

choopk, 2018-03-09:
It failed to mount. Im using a micro sd. SDHC formatted to fat32

JamesGKent, 2018-03-09:
are you able to turn on extra logging?
the SD_MMC library doe have log messages in it to help us work out why it failed if you can turn them on.

choopk, 2018-03-09:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
E (69) sdmmc_req: handle_idle_state_events unhandled: 00000004 00000000
E (70) sdmmc_cmd: sdmmc_card_init: send_op_cond (1) returned 0x107
Card Mount Failed

JamesGKent, 2018-03-09:
i was looking more for error messages like this one:
["Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true."](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/src/SD_MMC.cpp#L62)
or this one:
["Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place."](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/src/SD_MMC.cpp#L68)

those error messages seem to come from the ESP IDF, and not in the arduino libraries.

choopk, 2018-03-09:
I have not turn on extra logging. How am i suppose to enable it?

JamesGKent, 2018-03-09:
it looks like that menu option is only available on the "ESP Dev Module" board option, but that board can be configured to match the options of other varients.
[image]

but if you set the flash mode, size and frequency to match your board, set the logging to a higher level such as debug or verbose and then recompile/upload and post the new output.

choopk, 2018-03-09:
Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.
Card Mount Failed

JamesGKent, 2018-03-09:
sorry, not sure i can help you any further.
according to [schematics](http://www.ttgobbs.com/attachment.php?aid=MTJ8ODM2ZmNjODB8MTUyMDU4NzcyMHxkNjc5SGd1cWVVMlZDbkR2UHcvSEROUEdrdEdCZWcweUw4aXovdHFLNU1WSlBIQQ%3D%3D) there are already 10k pullups fitted to all the SD lines.

I don't know if 10k is the correct value though.

only thing i can suggest trying would be commenting out this [https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/src/SD_MMC.cpp#L46](line) to try running at a slower speed?

choopk, 2018-03-09:
Did that still giving the same error message. Thanks for the help though

JamesGKent, 2018-03-09:
what about if you change it to a much lower speed?
eg SDMMC_FREQ_PROBING      (400 instead of 20000 or 40000)

JamesGKent, 2018-03-09:
alternatively try changing your begin call to:
```begin("/sdcard", true);```

choopk, 2018-03-09:
still getting the same error.

JamesGKent, 2018-03-09:
what about using the normal SD lib, but set up the SPI before passing it in. so this [example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino)

with the setup changed from:
```
void setup(){
    Serial.begin(115200);
    if(!SD.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
```

to:
```
void setup(){
    Serial.begin(115200);
    SPI.begin(14, 2, 15);
    if(!SD.begin(13)){
        Serial.println("Card Mount Failed");
        return;
    }
```

choopk, 2018-03-09:
@JamesGKent  yup it worked flawlessly!! Thanks :D!

closed 2018-03-09 by choopk

JamesGKent, 2018-03-09:
wow, that was a total shot in the dark...
btw just to clarify from my reading [here](http://elm-chan.org/docs/mmc/mmc_e.html) SD cards can be used in 3 modes, one of which is SPI and the other two are SD specific. on AVRs the SPI mode is normally used because there is hardware built in for it, so it is faster that software emulating either of the other two modes. on the ESP it looks like we have the MMC hardware to use the other two modes natively.
I believe those other two modes (particularly the 4 bit mode) should be faster than SPI, however for you it seems that this didn't work. 
when using the SD_MMC lib the 4 bit is the default unless the board definition says only 1 bit mode is available, so the default didn't work, my suggestion of using ```begin("/sdcard", true);``` should have activated 1 bit mode, which also didn't work. so my last suggestion using the normal SD lib is using SPI, but configuring the pins of the SPI interface onto the pins normally used by MMC.

choopk, 2018-03-09:
Thanks a bunch for the help and the great follow up explanation.

beegee-tokyo, 2018-03-10:
@JamesGKent I shared your finding in the [Wiki](http://desire.giesecke.tk/index.php/2018/03/10/ttgov2-sd-card-support/). Thank you.

JamesGKent, 2018-03-12:
@beegee-tokyo would it not be worth trying to add some explanation as to what the setup is doing?
personally i still think it would be worth trying to investigate why the MMC library doesn't work, but as i don't have one of these boards i am not able to do so.

beegee-tokyo, 2018-03-12:
@JamesGKent You are right, will add some more info to it. Don't have such a board either, so I cannot test anything.

cyberman54, 2018-03-18:
If i insert a FAT formatted SD card in the card reader slot of my TTGOv2 board, and restart it, it seems trying to boot from SD card:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
```

Any hint how to avoid this and boot from internal flash?

ricardo-samaniego, 2018-08-12:
For the record: this exact solution works also for the LilyGO ESP32 TTGO T4 V1.1 board.

You can see this board and the pins used for SD card connection here: [https://github.com/LilyGO/ESP32-TTGO-T4](https://github.com/LilyGO/ESP32-TTGO-T4)

nailsonme, 2018-08-20:
@JamesGKent Thanks a lot!

mhaberler, 2019-03-07:
Have a look at https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card#note-about-gpio12

After having similar problems I followed the espefuse.py set_flash_voltage 3.3V suggestion and all problems went away.

I think this is a irreversible operation.

akaminetzkyp, 2019-05-14:
@JamesGKent thank you very much! I'm now able to use an SD card on a TTGO LoRa32 V2.1 board. Cheers!

jshank, 2020-06-02:
If anyone else runs into this, the problem is that the pins_arduino.h header file has the wrong SPI pins for the TTGOV2 board (it has the V1 pins). You can either manually update your pins_arduion.h file or follow @JamesGKent recommendation in https://github.com/espressif/arduino-esp32/issues/1190#issuecomment-371777387 and manually initialize the SPI library with the correct pins. By default, the SD library will automatically call the SPI library with the pins (incorrectly) defined by the boards header file.