espressif/arduino-esp32 #1191: Multiple BLEClient

fabiencomte, 2018-03-07:
Hello,
I tried to work with multiple clients but it looks limited in BLEDevice.

I did the attached modifications but it crash if i try MAX_CLIENTS_COUNT > 1

[BLEDevice.zip](https://github.com/espressif/arduino-esp32/files/1789613/BLEDevice.zip)

beegee-tokyo, 2018-03-08:
I had the same problem, check out this issue: https://github.com/nkolban/esp32-snippets/issues/307
It seems only 1 client is supported in the current library. You can try the changes I did and it will work.

fabiencomte, 2018-03-23:
Thank you, i will read that asap

fabiencomte, 2018-03-29:
Hello,
I looked to the code and this is not my case. Your exemple is a server that can accept multiple clients and for me, the ESP32 acts with multiple links to N servers.

Do you have an idea please ?

Thank you

nsamala, 2018-05-24:
@fabiencomte I just created a PR to address this issue. In my code, you can connect to ESP32 to N servers:

https://github.com/nkolban/ESP32_BLE_Arduino/pull/13

sizen801, 2018-10-03:
I am trying to calculate the throughput of ESP32 BLE in star connection. I was able to connect more than one client to a serer. To calculate throughput i am simultaneously sending characters from one client to another through server, and calculating the time difference between the arrival time of two characters. The problem that i am facing is that readValue() function in client side is not able to read characters fast enough. It is producing some delay before it receives second character. Due to this i am not getting correct receiving time.
So i tried using notify(). I am receiving the characters right after one another ans is thus able to calculate the correct time . This gives me a correct throughput. But the problem is i am able to notify only one client (the latest one which was connected to network). How can i notify two different clients using same server? I tried the instructions given above but it is giving me compling error in setValue() function as stated above.
Can anyone please help me?

closed 2019-08-15 by stale[bot]