espressif/arduino-esp32 #1192: help with exception

tablatronix, 2018-03-07:
### Hardware:
Board:							doit esp32 dev
Core Installation/update date:			today
IDE name:							platformio

### Description:
No idea, this issue goes away as I change parts of code.
There is no code executed between crashes, not even delays()
Is this some kind of compiler bug ?

### Sketch:
I do not have a sketch yet, its rather convoluted atm.

### Debug Messages:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Register dump:
PC      : 0xffffffff  PS      : 0x00060130  A0      : 0x800827d7  A1      : 0x3ffe3be0  
A2      : 0x3f420008  A3      : 0x00031580  A4      : 0x3ff000cc  A5      : 0x3ff00020  
A6      : 0xffffffff  A7      : 0xfffffff0  A8      : 0x800d41ca  A9      : 0x3ffe3bc0  
A10     : 0x3f4129c4  A11     : 0x3ffc3864  A12     : 0x00001200  A13     : 0x00000000  
A14     : 0x00000016  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000014  
EXCVADDR: 0xfffffffc  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x7fffffff:0x3ffe3be0 0x400827d4:0x3ffe3c00 0x400829c0:0x3ffe3c30 0x4007887a:0x3ffe3c50 0x400788e1:0x3ffe3c80 0x40078a21:0x3ffe3cb0 0x40078b4a:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
```

```
Exception Cause: Not found

0x40078000: ?? ??:0
0x40078000: ?? ??:0
0x40078a28: ?? ??:0
0x4000c46c: ?? ??:0
0x4000c477: ?? ??:0
0x400827d4: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c:328
0x400829c0: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c:203
0x4007887a: ?? ??:0
0x400788e1: ?? ??:0
0x40078a21: ?? ??:0
0x40078b4a: ?? ??:0
0x40007c31: ?? ??:0
0x4000073d: ?? ??:0
```

CelliesProjects, 2018-03-31:
Maybe join us at #1025?

tablatronix, 2018-03-31:
Will give it a try and report back

a-c-sreedhar-reddy, 2018-12-04:
Hey @tablatronix  I am facing the same issue my cores restart reasons are 14,16. Do you know how to solve this?

closed 2019-08-15 by stale[bot]