espressif/arduino-esp32 #1194: Increasing BLE sketch size of ~156k after adding BluetoothSerial lib

reaper7, 2018-03-08:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			Branch_b4b9a79 05/03/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
after adding the BluetoothSerial library (https://github.com/espressif/arduino-esp32/commit/b4b9a79eead317d899fe4f199b4d536b99c25112), sketches which use BLE library (https://github.com/nkolban/ESP32_BLE_Arduino/tree/6bad7b42a96f0aa493323ef4821a8efb0e8815f2) increase finally size by **~156 kbytes!**

Lets try for example library: BLE -> example: BLE_uart:

- branch with previous commit (https://github.com/espressif/arduino-esp32/commit/835268c326e0a6c95a36e66f5b6a6d24c2c51f8f) finally sketch size: 931869 bytes (71%)

- branch with BluetoothSerial library (https://github.com/espressif/arduino-esp32/commit/b4b9a79eead317d899fe4f199b4d536b99c25112) finally sketch size: 1088621 bytes (83%)

For this reason my sketches with BLE and several additions do not fit esp32 flash any more,
even though I do not use new BluetoothSerial library in these cases (increase size from 1265006 to **1422022**)

What is interesting, I did not notice the increase final sketch size when BLE library is not used.

In the current situation, creation a little more advanced programs with BLE and for e.g. display, becomes impossible due to lack of space.

### Sketch:
library: BLE -> example: BLE_uart

beegee-tokyo, 2018-03-08:
The Bluetooth stack is very large. You need to change you partition sizes as shown in the [Wiki](http://desire.giesecke.tk/index.php/2018/01/30/change-partition-size/).

reaper7, 2018-03-08:
@beegee-tokyo - tnx! this is a option but...  
...it's looks like these two libraries use two separate bluetooth stacks?

beegee-tokyo, 2018-03-08:
I do not know if there are two different stacks for BLE and Bluetooth, because that is somewhere down below in the ESP-IDF. But I started with using BLE only first and added Bluetooth Serial later (check my [blog](http://desire.giesecke.tk/index.php/2018/02/27/esp32-bluetooth-serial-and-ble-server-running-together/)) the program size didn't double, so I guess there is only one stack.

reaper7, 2018-03-08:
In that case, if only one stack, who eats ~157k ? new libbt.a from sdk?

I check this on raw example from library BLE->BLE_Server as-is (without any addition like SerialBT)
current master branch and earlier branch from 5 march 2018 "**Add BluetoothSerial library**" https://github.com/espressif/arduino-esp32/commit/b4b9a79eead317d899fe4f199b4d536b99c25112
generate output files with ~1085890bytes
while the previous branches (until 4 march 2018) generate output files with ~929130bytes

in my opinion it is a very large increase

jjoe64, 2018-06-08:
Hi all, I faced the same issue and did not found a easy solution, so I finally forked the library and create a "light version" of it.
Only the BLE server will work.  This is a fork of the official esp32 ble library and is stripped to decrease the size. It is only possible to use the ble server features.

I saved about ~30KB program size with this version.

[https://github.com/jjoe64/esp32_ble_light](https://github.com/jjoe64/esp32_ble_light)

aravindara, 2018-11-03:
when i do that my program is compiling with errors

closed 2019-08-15 by stale[bot]