espressif/arduino-esp32 #1195: Multiple Timers on ESP32

kszere, 2018-03-08:
### Hardware:
**Board:**							TTGO LORA SX1278 ESP32 0.96 OLED 16 Mt bytes (128 Mt bit) 433Mhz for arduino
**Core Installation/update date:**			07/03/2018
IDE name:							Atom + PlatformIO
Upload Speed:						115200

### Description:
#1 This code crash few sec after run. How use multi Timer on ESP32?
#2 How make more than 4 Timers? Example: Use 1 Timer with programming 5x Timer with deferrent times ex. 1,5,30,60,300 sec?

### Sketch:
https://github.com/pcbreflux/espressif/blob/master/esp32/arduino/sketchbook/ESP32_multitimer/ESP32_multitimer.ino

```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU1)
Register dump:
PC      : 0x400852cd  PS      : 0x00060034  A0      : 0x800858bb  A1      : 0x3ffc0590
A2      : 0x3ffc10f8  A3      : 0x00000000  A4      : 0x00000005  A5      : 0x00000020
A6      : 0x00000020  A7      : 0x00000000  A8      : 0x0000cdcd  A9      : 0x0000cdcd
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x00060021  A13     : 0x00000001
A14     : 0x00060021  A15     : 0x3ffc6c78  SAR     : 0x0000001d  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x400852cd:0x3ffc0590 0x400858b8:0x3ffc05b0 0x4008673f:0x3ffc05d0 0x40086491:0x3ffc05f0 0x4008167a:0x3ffc0600 0x400d59f7:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078a28
```

beegee-tokyo, 2018-03-08:
Try the new Ticker library (nearly the same as on ESP8266). [Ticker](https://github.com/espressif/arduino-esp32/tree/master/libraries/Ticker). Works fine for me with more than 4 timers/tickers.

atanisoft, 2018-03-08:
I also use all four hardware timers without issues. If you need more timers the Ticker library referenced above is good for that. For my use case I don't know that Ticker is suitable, as part of each invocation of the timer the period changes based on external factors.

kszere, 2018-03-08:
@beegee-tokyo Thanks for "Ticker".
How use 4 HW Timers without crash?
In my code, functions take something 2 sec and also something 10 sec. It crash my ESP.

sansillusion, 2018-03-08:
From my understanding (I might be wrong), you should use the timers to set variables and act upon it in the main loop so the delay will be in main loop not timers functions.

atanisoft, 2018-03-08:
With hardware timers you must do whatever tasks you want to do quickly, the shorter amount of time in the ISR the better. As @sansillusion suggests, setting variables or toggling a GPIO pin is about the max you should really be doing. In my case I am triggering the various hardware timers at ~58uS, ~116uS, ~98uS or 196uS. The time between ISR invocations depends on the data being converted to a digital signal on a pair of GPIO pins.

beegee-tokyo, 2018-03-09:
@kszere what I do is
 - have threads that do different tasks, e.g. read a temperature sensor, another one read a light sensor, another one sends data to MQTT broker, ... These threads go to sleep immediately after being started
 - have several tickers running that call a routine every 10 sec, every 30 sec, and so on
 - these routines only wake-up a thread then they return
 - the woken thread does the task, then goes back to sleep.

I made a kind of tutorial [Using the multitasking capabilities of the ESP32 / FreeRTOS](http://desire.giesecke.tk/index.php/2018/02/07/using-the-multitasking-capabilities-of-the-esp32-freertos/)

closed 2019-08-15 by stale[bot]