espressif/arduino-esp32 #1196: Check_config errors during compilation of wifi and http libraries

ifrew, 2018-03-08:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Node 32S
Core Installation/update date:			7/Mar/2018
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					80Mhz
Upload Speed:						?115200?

### Description:
Pulled down the esp32 ardunio core from yesterday and recompiled my app. I am using visual studio community with visual Micro on windows 10. Started getting a lot of check config errors. I haven't changed anything in my code yet or my environment so has there been some new config setup I need to do to get rid of these errors?

Using library WiFiClientSecure version 1.0 in folder file:///D:/Documents/Arduino/hardware/espressif/esp32/libraries/WiFiClientSecure
"D:\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DHAVE_CONFIG_H "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_Node32s -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"Node32s\" -DARDUINO_VARIANT=\"node32s\" -DESP32 -DCORE_DEBUG_LEVEL=3 -I"D:\Documents\Arduino\hardware\espressif\esp32\cores\esp32" -I"D:\Documents\Arduino\hardware\espressif\esp32\variants\node32s" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\HTTPClient\src" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\WiFiClientSecure\src" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\SPI\src" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\Update\src" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries\Preferences\src" -I"C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\Extensions\mjlfxd1j.24d\Micro Platforms\visualmicro\ide\libraries" -I"D:\Documents\Arduino\hardware\espressif\esp32\libraries" -I"C:\Program Files (x86)\Microsoft Visual Studio 14.0\Common7\IDE\Extensions\mjlfxd1j.24d\Micro Platforms\default\debuggers" -I"D:\Documents\Arduino\libraries" "D:\Documents\Arduino\hardware\espressif\esp32\libraries\WiFiClientSecure\src\ssl_client.cpp" -o "C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhere_flarm_vario\espressif_node32s\Release\WiFiClientSecure\ssl_client.cpp.o"
 
esp_config.h:2665: In file included from
x509_crt.h:29: from
ssl_client.h:14: from
ssl_client.cpp:15: from
 
check_config.h: 406:2: error: #error "MBEDTLS_PLATFORM_STD_CALLOC defined, but not all prerequisites"
   #error "MBEDTLS_PLATFORM_STD_CALLOC defined, but not all prerequisites"
 
check_config.h: 410:2: error: #error "MBEDTLS_PLATFORM_STD_CALLOC defined, but not all prerequisites"
   #error "MBEDTLS_PLATFORM_STD_CALLOC defined, but not all prerequisites"
 
check_config.h: 414:2: error: #error "MBEDTLS_PLATFORM_STD_FREE defined, but not all prerequisites"
   #error "MBEDTLS_PLATFORM_STD_FREE defined, but not all prerequisites"
 
check_config.h: 419:2: error: #error "MBEDTLS_PLATFORM_STD_EXIT defined, but not all prerequisites"
   #error "MBEDTLS_PLATFORM_STD_EXIT defined, but not all prerequisites"
"D:\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DHAVE_CONFIG_H "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-ID:\Documents\Arduino\hardware\espressif\esp32/tools/sdk/ [... cut ...]
check_config.h: 659:2: error: #error "MBEDTLS_HAVE_INT32\MBEDTLS_HAVE_INT64 and MBEDTLS_HAVE_ASM cannot be defined simultaneously"
   #error "MBEDTLS_HAVE_INT32\MBEDTLS_HAVE_INT64 and MBEDTLS_HAVE_ASM cannot be defined simultaneously"

Using library Wire version 1.0 in folder file:///D:/Documents/Arduino/hardware/espressif/esp32/libraries/Wire
  Using previously compiled file: C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhere_flarm_vario\espressif_node32s\Release\Wire\Wire.cpp.o
Error compiling libraries

beegee-tokyo, 2018-03-09:
`Using previously compiled file: C:\Users\Ifrew\AppData\Local\Temp\VMBuilds\airwhere_flarm_vario\espressif_node32s\Release\Wire\Wire.cpp.o` 
might be the problem. Did you try to clean the project and then compile it again?

ifrew, 2018-03-09:
It did this after I brought down the core, so it compiled everything first time through. However, I just cleaned the entire solution and rebuilt. Same issue. I guess since nobody else has this issue its a problem with my environment or they aren't using the libraries I'm using or the code that is being used now needs something other folks aren't using. I'll dig into it more.

ifrew, 2018-03-11:
Seemed to be something wrong with the IDE environment. Downloaded the core again from scratch and still got the same issue. However After restarting the IDE it went away. No idea what was causing it. Clsoing the issue.

closed 2018-03-11 by ifrew

arnold-b, 2019-03-06:
Same Problem