espressif/arduino-esp32 #1197: wakeup esp32 from deepsleep using mqtt (AWS IOT SDK)

Abishek05, 2018-03-09:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?09/Mar/2018?
IDE name:							?Arduino IDE? 
Flash Frequency:					?80Mhz?
Upload Speed:						?115200?

### Description:
Is there a way to wakeup esp32 from deep sleep by issuing a command to a topic subscribed by the device?. I looked into qos1 feature of mqtt which is a queing system, and when i wakeup the device i should be able to process the messages which were sent even when the device is asleep. Is there any examples to use qos1 in the application layer? So with this i can timely wakeup and process the messages.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

stickbreaker, 2018-03-10:
@Abishek05 Deep sleep has the WiFi radio powered off, so no data is being received, The only events that can wake it from deep sleep are hardware issues.
* Power cycling
* pin interrupt
* timer expiration

To achieve what you want I would suggest you set up a re-occurring timer. Set the timer for 10 minutes, wake up, connect to WiFi, Connect to the AWS server, read the queued messaged, process them, reset the timer, go back to sleep.

Chuck.

Abishek05, 2018-03-12:
Thanks for your response, how do I enable qos1 in the AWS sdk so that I can queue the messages.

lbernstone, 2018-03-16:
You set the qos when you publish or subscribe to a topic.  You will need to look at the documentation for the library you use for the exact syntax.

closed 2019-08-15 by stale[bot]