espressif/arduino-esp32 #1198: Path to xtensa-esp32-elf-g++

hoernche, 2018-03-09:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			latest - where can I find the version? 
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200
PC-OS:                                                     Windows 10

### Description:

Hey guys, 

finally I received my ESP32 Module and I try since a while to get it running with the Arduino IDE. 
I followed this tutorial step-by-step: 
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

It didn't work after various trys so I uninstalled the IDE, installed it freshly and copied the esp32 directory manually. I also run the get.exe as administrator - still same messages. 

What concerns me, are the double backslashes in the path and then the change to slashes for the esp32 directory. I also fiddled with the platform.txt file and tried to replace all "/" by "\" but it didn't resolve the problem, I just got a different error message, still with "\\" all over the path (see Debug Message 2). 

Am I on the right track, do I have to modify the platform.txt somehow? Any other recommendations? 
I'm on this for many hours, would be thankful for any help. 

Best Regards, 

Lukas

### Sketch: Example Sketch WiFiScan

### Debug Message 1:
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 115200, None"

exec: "C:\\Users\\lukas\\Documents\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++": file does not exist
Fehler beim Kompilieren für das Board ESP32 Dev Module.
Ungültige Bibliothek C:\Users\lukas\Documents\Arduino\hardware\espressif\esp32\libraries\BLE in C:\Users\lukas\Documents\Arduino\hardware\espressif\esp32\libraries\BLE gefunden
Ungültige Bibliothek C:\Users\lukas\Documents\Arduino\hardware\espressif\esp32\libraries\BLE in C:\Users\lukas\Documents\Arduino\hardware\espressif\esp32\libraries\BLE gefunden

## Debug Message 2:
exec: "C:\\Users\\lukas\\Documents\\Arduino\\hardware\\espressif\\esp32tools\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-g++": file does not exist
Fehler beim Kompilieren für das Board ESP32 Dev Module.

JamesGKent, 2018-03-13:
have you actually looked in that folder to see if there are any files there?
and what did running get.exe actually output?

klaus-liebler, 2018-11-05:
Hi Lukas,

I am facing the same issue. @Lukas, have you solved it. If yes, how?

Thank you, regards

Klaus

lbernstone, 2018-11-06:
https://desire.giesecke.tk/index.php/2018/04/02/get-exe-get-py-fails-to-download-required-files/

closed 2019-08-15 by stale[bot]