espressif/arduino-esp32 #1202: WifiMulti does not support Static IP

williamesp2015, 2018-03-11:
I am using ESP32 Arduino websocket using WifiMulti. There is example using Static IP ((https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientStaticIP/WiFiClientStaticIP.ino) I copied below:
#include <WiFi.h>

const char* ssid     = "your_network_name";
const char* password = "your_network_password";
const char* host     = "example.com";
const char* url      = "/index.html";

IPAddress local_IP(192, 168, 31, 115);
IPAddress gateway(192, 168, 31, 1);
IPAddress subnet(255, 255, 0, 0);
IPAddress primaryDNS(8, 8, 8, 8); //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional

void setup()
{
  Serial.begin(115200);

  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.println("STA Failed to configure");
  }

  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected!");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("ESP Mac Address: ");
  Serial.println(WiFi.macAddress());
  Serial.print("Subnet Mask: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("Gateway IP: ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("DNS: ");
  Serial.println(WiFi.dnsIP());
}
/******************************/
My problem is with my project  using WiFiMulti which is not supporting Static IP
 WiFiMulti.addAP(SSID, pass);
Serial.print("Wait for WiFi... ");
while(WiFiMulti.run() != WL_CONNECTED) {
      Serial.print(". ");
        delay(500);
     }

josepedrodiaz, 2018-05-01:
Same problem here

closed 2019-08-15 by stale[bot]

peter1a, 2021-01-20:
Is now any support for static IP?

chupocro, 2021-03-01:
`WiFi.config(ip, gateway_router, subnet);`

should be **after**

`WiFi.begin(ssid, password);`

or **after**

`wifiMulti.addAP(ssid, password);`

jjavieralv, 2023-05-02:
Example of working with WifiMulti and staticIP
`
#include <WifiMulti.h>
//insert here your values
#define WIFI_SSID ""
#define WIFI_PASSWORD ""
IPAddress local_IP(192, 168, 1, 40);
IPAddress gateway(192, 168, 1, 1);
IPAddress subnet(255, 255, 255, 0);
void setup() { 
 //config wifi and check
  if (!WiFi.config(local_IP, gateway, subnet)) {
    Serial.println("STA Failed to configure");
  }
  WiFi.mode(WIFI_STA);
  wifiMulti.addAP(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to wifi");
  while (wifiMulti.run() != WL_CONNECTED) {
    Serial.print(".");
    delay(100);
  }
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println();
}
`

abhishekabhi789, 2023-10-05:
If gateway varies among different wifi APs then how can a device use static IP? I tested with 4 hotspot and two of them have same address `192.168.43.1` so I can use any adress like `192.168.43.###` However, the other two hostpots have different address, resulted in showing null as IP adress in hostpot config page, at the same time prints the static ip in `WiFi.localIP()`.
Now Im using mDNS, but it's not available on all android versions.

Sojourneer, 2025-02-25:
I have the same issue.  This issue should not have been closed.