espressif/arduino-esp32 #1204: digitalWrite(pin, HIGH) with PIN 4 and 12 doesnt work

nickoloss, 2018-03-11:
### Hardware:
Board:							        ESP32 PICO Dev Module
Core Installation/update date:			11/March/2018
IDE name:							Arduino IDE 1.85
Flash Frequency:					        40Mhz
Upload Speed:						921600

### Description:
Describe your problem here

### Sketch:
----------------- ElectricEngine.hpp ---------------------

#ifndef ElectricEngine_hpp
#define ElectricEngine_hpp

#include "Arduino.h"

#define clockwise_pin 4
#define counterclockwise_pin 12
#define pwm_pin 24

class ElectricEngine {
    
    int oldDirection = 0;
    int pwm_pin_channel = 1;
    void setClockwiseDirection();
    void setCounterclockwiseDirection();
    
public:
    ElectricEngine();
    void run(int speed, int direction);
    void stop();
};

#endif /* ElectricEngine_hpp */

----------------- ElectricEngine.cpp ---------------------

#include "ElectricEngine.hpp"

ElectricEngine::ElectricEngine() {
    
    
    pinMode(pwm_pin, OUTPUT);
    pinMode(clockwise_pin, OUTPUT);
    pinMode(counterclockwise_pin, OUTPUT);
    // Initialize channels
    // channels 0-15, resolution 1-16 bits, freq limits depend on resolution
    // ledcSetup(uint8_t channel, uint32_t freq, uint8_t resolution_bits);
    ledcAttachPin(pwm_pin, pwm_pin_channel);
    ledcSetup(pwm_pin_channel, 500, 8);
    

}

void ElectricEngine::run(int speed, int direction) {
    
    if (direction == 1 ) {
        setClockwiseDirection();
        ledcWrite(pwm_pin_channel, speed);
        oldDirection = 1;
    } else if (direction == -1 ) {
        setCounterclockwiseDirection();
        ledcWrite(pwm_pin_channel, speed);
        oldDirection = -1;
    }
    

};

void ElectricEngine::stop() {
    ledcWrite(pwm_pin_channel, 0);
    digitalWrite(counterclockwise_pin, LOW);
    digitalWrite(clockwise_pin, LOW);
};

void ElectricEngine::setClockwiseDirection() {
    digitalWrite(counterclockwise_pin, LOW);
    digitalWrite(clockwise_pin, HIGH);
};

void ElectricEngine::setCounterclockwiseDirection() {
    digitalWrite(clockwise_pin, LOW);
    digitalWrite(counterclockwise_pin, HIGH);
};

---------------------- sketch.ino -------------------

include "ElectricEngine.hpp"

ElectricEngine engine;

setup(){

}

loop(){
engine.run(200, 1);
delay(1000);
engine.stop();
engine.run(200, -1);
delay(1000);

}

### Debug Messages:
Only PIN 12 goes HIGH as supposed

copercini, 2018-03-12:
Why should it go HIGH?

Your last command is `engine.run(200, -1);`
that will trigger
```cpp
    } else if (direction == -1 ) {
        setCounterclockwiseDirection();
```

and set it to LOW

```cpp
void ElectricEngine::setCounterclockwiseDirection() {
    digitalWrite(clockwise_pin, LOW);
    digitalWrite(counterclockwise_pin, HIGH);
};
```

nickoloss, 2018-03-12:
Each of theese Pins should go HIGH and LOW after 1 Second and each other. But only GPIO 12 is doing that.

```
loop(){
   engine.run(200, 1); //       PIN 4 HIGH  and  PIN 12 LOW.  
   delay(1000);
   engine.stop(); 
   engine.run(200, -1); //      PIN 4 LOW  and  PIN 12 HIGH
   delay(1000);

}

```

liebman, 2018-03-12:
This works for me.  I connected a red LED& 220 ohm resister from each pin to each pin to +3.3v.  The LEDs alternate being lit.

```c
#include "Arduino.h"

void setup()
{
    pinMode(4,  OUTPUT);
    pinMode(12, OUTPUT);
}

void loop()
{
    static bool state = false;
    digitalWrite(4,  state ? HIGH : LOW);
    digitalWrite(12, state ? LOW  : HIGH);
    state = !state;
    delay(1000);
}
```

nickoloss, 2018-03-13:
I can confirm that. Strange why my scetch is not working. Thankyou.

closed 2018-03-13 by nickoloss

rei-vilo, 2020-01-24:
Works on pin 4, but why not for the other pins?