espressif/arduino-esp32 #1206: More a question than an issue about the upgrade process.

JacoFourie, 2018-03-12:
Hi all. I am not sure if this has been answered somewhere else. If so please point me in that direction. How do I know I am running the latest version of the core wrapper ? I follow the steps in the doc where I run the git gui to pull the latest changes and then run get.exe. But how do I check that I am running the latest. I can see that it will pull some changes from time to time as git will show it. But when I run the get.exe it will just decompress the old downloaded zip files. 

Reason I am asking is I made some changes to the core files as we had some issues with the UARTS and was given a workaround. So I do not want to loose those changes/ fixes when I update the core. I have made a back-up of the files but I still just want to make sure I understand the process of the update. And also that any fixes will get applied on my end. 

Kind Regards.

Jaco.

sansillusion, 2018-03-12:
I too would be interested to find out more about the process.
I have had an issue where for a long while I was following update procedures and I tough I was up-to-date since Git GUI was showing updates from time to time. But it turns out it was not up-to-date, there where library missing and many files where not matching the repository. I ended up deleting and re installing and all started working as expected.

lbernstone, 2018-03-12:
This project is still in a pre-release state.  me-no-dev has indicated that a release will be coming soon.  It will be distributed as a json package that you can add in boards manager in the Arduino IDE.

JacoFourie, 2018-03-13:
OK so while we wait for the project what is the best way ? Do we delete everything and re-install every time ?

JacoFourie, 2018-03-13:
@JamesGKent I am running Windows. does the same commands apply ?

JacoFourie, 2018-03-13:
Also does the git pull do the same job as the get.exe or should you run get.exe after the git pull ?

JamesGKent, 2018-03-13:
as long as you installed esp32 using git then the same commands apply.
git pull will pull the latest code from this repository, get.exe will pull down the latest toolchain, you should run get.exe after the pull, if there is no newer toolchain then get.exe will just reextract the already downloaded latest copy you have.

JacoFourie, 2018-03-13:
Thanks a lot @JamesGKent. While we are on the topic. if somebody made a patch, then I assume you download those files and copy them into the folder where needed. But if you run a git pull it will over write those patch files and you have to copy them back again ?

JamesGKent, 2018-03-13:
@JacoFourie if you make any local changes at all, either downloaded from someones patch or something you've done manually then those git commands will remove any added files and any added changes, so anything you have done would need to be re done if indeed it is still needed.

e.g. the i2c patch which is currently not part of master would need to be reapplied after the git pull.
note that the better approach to making changes/patches like that would be to merge pull requests with your local copy. that way instead of swapping out whole files which may have other changes, only the functions that are changed in the patch will be altered in your local copy.

JacoFourie, 2018-03-13:
Aaa OK as you can see I have not used GIT that much. Other than downloading libs and asking questions. So I assume you can then config your own GIT copy to pull the patch from the patch location and merge it to your own copy? Is that what you are saying ? Seems I need to check out some GIT training lol.

JamesGKent, 2018-03-13:
ok as an example, pretending i have an old version installed to which i have the i2c rewrite:
with a git bash open in my esp32 folder i run:
```
git reset --hard @{u} # reset to upstream
git clean -df # clean untracked files
git pull # get latest code from repo
./tools/get.exe # get latest toolchain
git fetch origin /pull/840/localbranch # bring pull request 840 into branch "localbranch"
git checkout localbranch # switch to the modified "localbranch" branch
```

note that you could merge multiple PRs like this if needed.
reference from here:
https://help.github.com/articles/checking-out-pull-requests-locally/

closed 2018-03-22 by JacoFourie