espressif/arduino-esp32 #1207: BluetoothSerial not returning -1 when no bytes available to read

anteph, 2018-03-12:
### Hardware:
Board:							NodeMCU
Core Installation/update date:			~8/03/2018
IDE name:							Arduino IDE

### Description:
Hi! First of all, thank you very much for this awesome library :)

Not sure if this is an issue or a desired behavior, but when calling the read function of the BluetoothSerial Object, it is returning me 1073493200, whereas the regular serial function for the UART is returning -1 (as expected).

The weird part is that looking at the source code there is an explicit return of zero when the FreeRTOS queue is empty. Am I missing some cast or something? I think not because the return type is int and it is explicitly returning zero, which makes the behavior weird. 

Anyway, shouldn't it be returning -1, to be consistent with the Hardware Serial implementation?

Best regards,
Nuno Santos
### Sketch:
```
#include "BluetoothSerial.h"
 
BluetoothSerial SerialBT;
 
void setup() {
  Serial.begin(115200);
  
  if(!SerialBT.begin("ESP32")){
    Serial.println("An error occurred initializing Bluetooth");
  }
}
 
void loop() {
  Serial.println(SerialBT.read());
  Serial.println(Serial.read());
  delay(50);
}
```

copercini, 2018-03-13:
Obrigado por reportar Nuno!

This is because if `available()` returns 0 the `read()` function simply doesn't return anything causing this unexpected behavior

I've created a fix here: https://github.com/espressif/arduino-esp32/pull/1208

anteph, 2018-03-25:
De nada :)

Thanks for the quick resolution!

Best regards, 
Nuno Santos

closed 2018-03-25 by anteph