espressif/arduino-esp32 #1211: RC522 inconsistent with ESP32. Is SPI too fast ?

SrikanthBala, 2018-03-13:
Please fill the info fields, it helps to get you faster support

### Hardware:
Board:							ESP32 Dev Module?
Core Installation/update date:		Feb2018
IDE name:						Arduino 1.8.5
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
I tried my RC522 module using the Dump info and with appropriately set CS and RST as described in https://www.youtube.com/watch?v=waYM37_fUsg&t=133s 

My board is read 1 out of 10 attempts and card ID is read but not the rest of the 1K info
I believe it has something to do with speed problems. Has anyone got it working consistently  ? if so how ? Would appreciate some help. Thanks

SrikanthBala, 2018-03-13:
I forgot to mention I was using MFRC522 library version 1.3.6
On seeing some similar problems in this forum, I tried version 1.3.5

Now the Tag ID is consistently read but it is unable to dump info and shows a message 
PCD_Authenticate() failed: Timeout in communication.
Any suggestions ?

beegee-tokyo, 2018-03-13:
Just guessing, I don't have that module.
Could be that the library is interrupted while dumping the info by a task switch.
If possible try to disable task switching while you read from the MFRC522.
I had a timeout problem with the DHT11 sensor and could only fix it by disabling task switching.
Example how to do it is in the repo of my [DHTesp library](https://github.com/beegee-tokyo/DHTesp/blob/master/DHTesp.cpp) lines 199 to 203 and lines 215-217.

SrikanthBala, 2018-03-13:
Thanks beegee-tokyo for that insight. I tried it but didn't work. 
Then I remembered reading somewhere that the card might have been formatted in some other way, so only the TagID was being correctly read. So I tried another card and YIPEEE... it worked

I went back and installed the 1.3.6 again and it worked too, ... every time. I had foolishly tried reading the same card all the time and it was presumably formatted different. 

So ill close this now. Thanks beegee-tokyo for your help. I was about to sleep after a long day at work and saw your comment. So I got out to try it and now my problem is solved and I can look forward to progress tomorrow :-)

closed 2018-03-13 by SrikanthBala