espressif/arduino-esp32 #1212: unreliable connecting to wifi

JohnGEyles, 2018-03-13:
### Hardware:
Board:							HUZZAH32
Core Installation/update date:			8Jan2018
IDE name:							Arduino IDE w/ Esspressif extensions
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
I'm having very unreliable behavior connecting to my home wifi network.   It works sometimes and not other times, for unknown reasons.   I have tried reloading my sketch.   I have tried moving very close to my wifi router (to which other devices are successfully connecting).   I have repeatedly tried pushing the reset button.

I'd be thankful for any pointers as to how to debug this problem.   If I should post this elsewhere, please advise.

I have only included my setup routine, as that's where the failure occurs (as it hangs in the while loop).

### Sketch:
```cpp
#include <WiFi.h>
#include <SPI.h>

int i, j, num;

int enabs[] = {14, 15, 25, 27, 32, 33};

const char* ssid     = "ATT1530";
const char* password = "";

char numstr[20];

WiFiServer server(80);
void setup() {

  for (i=0; i<6; i++) {
      pinMode (enabs[i], OUTPUT);
      digitalWrite (enabs[i], HIGH);
   }
   
    Serial.begin(115200);
    pinMode(LED_BUILTIN, OUTPUT);      // set the LED pin mode

    delay(10);

    // blink LED, before trying wifi connect
   blinkx (3, 100, 500, 1000);
   
   // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    
    server.begin();

    SPI.begin();}

```

### Debug Messages:
Nothing, just a continuous series of dots as the while loop tries to connect.

lrmoreno007, 2018-03-13:
Add this before WiFi.begin();

``WiFi.enableSTA(true);``

lbernstone, 2018-03-13:
If you set your board to esp32 dev module and turn on debugging, it should give you a lot more info about why it cannot connect.

JohnGEyles, 2018-03-14:
WiFi.enableSTA(true) did not help.

JohnGEyles, 2018-03-14:
I set my board to ESP32 Dev Module, and Core Debug Level to "Verbose", in the Tools dropdown (if that's what you intended).

All I got was this:

```
Connecting to ATT1530
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
............................
```

JohnGEyles, 2018-03-14:
FWIW, I plugged in an old Airport Extreme I had lying around, and it seemed to connect:

```
Connecting to jge's Wi-Fi Network
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
......[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 7 - STA_GOT_IP
.
WiFi connected.
IP address: 

```10.0.1.3

But the thing isn't configured correctly (a whole 'nother issue) so I'm not sure if it's really a viable connection, but it appears to be.   

The wifi I'm attempting to connect to is a Netgear 7550 DSL gateway.   (Lord willing, ATT is going to seem like a bad dream within a week or two).

JohnGEyles, 2018-03-14:
Yep, I created another wifi network using the AIrport Extreme, and the ESP32 connects and the link functions properly.    ATT, 'nuff said ...

stickbreaker, 2018-03-15:
@JohnGEyles 
My ESP32, a WeMos Battery&Bluetooth was giving me fits.  Failing to connect, incomplete transmissions...

What I found was that my device is very orientation specific, my router is 20ft right of me:
* If I set the ESP32 to the left of me with it's antenna pointed horizontally behind me, the Signal strength was -81dBm.
* If I stood it with the antenna vertical, perpendicular to me, the strength increased to -78dBm.  
* If I stood it with the antenna vertical, inline with me, the signal strength increased to -66dBm.

Try reorienting your module.

```c++
    Serial.printf("New Client. RSSi: %ld dBm\n",WiFi.RSSI()); 
```

Chuck.

JohnGEyles, 2018-03-15:
That's interesting, Chuck.  I guess the ESP32 doesn't have an antenna pin.   I'll mess around with it with that WiFi call.  But I'm happy enough that it's robust with the Airport Extreme.

cjdell, 2018-05-21:
I am also having trouble connecting to WiFi. This applies to any of the Arduino samples that I've tried that utilise WiFi.

Symptoms are strange. In almost all circumstances, WiFi fails to connect on boot, and will never work even after many hard boots.

Very occasionally, after flashing a new program, the WiFi will connect instantly upon the first boot, and will ALWAYS connect on subsequent reboots and hard reboots. It will continue to work reliably until I flash a new program (even if it's the same one) and then never work again.

Summarised, if it works upon first boot upon flashing, it will always work upon reboots, otherwise it will never work.

Tried many different routers, and also tried erasing the flash with esptool.py.

Can anyone shed light on this?

I have two of these modules (same problem on both): https://www.banggood.com/Wemos-TTGO-WiFi-Bluetooth-Battery-ESP32-0_96-Inch-OLED-Development-Tool-p-1213497.html?rmmds=search&cur_warehouse=CN

Thanks.

rmick, 2018-05-26:
Yes Cjdell. 

You have just described the exact same behaviour i am seeing with a Heltec WiFiKit32 module. It is driving me insane !

drrehak, 2018-06-05:
hi, I just had troubling connecting my new Adafruit ESP32 Feather board with my Comcast router. but I got it to work when I tried the 2.4ghz channel SSID.  worth a try for anyone that came across this issue...

RobbertH, 2018-06-17:
Hi there, 
I'm having issues too. The first time I connected, everything went fine, but from the first reupload onwards, it kept doing this (verbose debug output):

Connecting to [...]
[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 13 - AP_START
.....[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:309] _eventCallback(): Reason: 201 - NO_AP_FOUND
..............................

RobbertH, 2018-06-18:
Strange. Connected to my android hotspot, then tried connecting to the normal router again. Everything works fine again.

rmick, 2018-06-20:
I had the exact same problem, and it drove me crazy for ages. It all started when I added a 2nd Wifi mode to the sketch on my Heltec WIFIkit32 ESP32 module.

This is the solution I found. My unit now works perfectly 99% of the time.

        //DO NOT TOUCH
        //  This is here to force the ESP32 to reset the WiFi and initialise correctly.
        Serial.print("WIFI status = ");
        Serial.println(WiFi.getMode());
        WiFi.disconnect(true);
        delay(1000);
        WiFi.mode(WIFI_STA);
        delay(1000);
        Serial.print("WIFI status = ");
        Serial.println(WiFi.getMode());
        // End silly stuff !!!

       // Connect to provided SSID and PSWD
        WiFi.begin(otaSSID, otaPSWD);

The problem seems to be that the ESP32 remembers the last WifiMode setting when it reboots, so you must interrogate it, disconnect it, and interrogate it again, to reset it - then you can initialise it. If you do not do the above little dance, then the Init fails.

dpondaco, 2018-08-11:
regarding rmick comments above.  I tried the code recommended, but got a bunch of compiler errors.  Do you have a list the the libraries the should be inlcuded in the ESP32 reset sketch.  I am having same problem as everyone else and not sure where to go from here.

Bodmer, 2018-11-24:
I am here because I have the same issue. With seemingly identical TTGO boards some connect quickly running identical code and others take minutes or never connect.

Changing one irrelevant line of code like a print statement suddenly seems to make it work. Then adding more code breaks it again. In the worst case it seems there is about a 1 in 4 chance that adding a few extra lines of code will make these boards work. Once they are persuaded to work they connect reliably in any power-up or reboot.

This suggest that something is not being initialized properly in the WiFi library but that most board happen to come up in an acceptably configured state. It is going to be difficult to track this down until one of the core developers gets hold of one of these rogue boards!

rodmcm, 2018-11-26:
I am also having the same problem, downloaded ESP32 board to Arduino last week and nothing but problems. Old programs that tests AP and STA WIFI connections do not work on AP mode.. Also I see lots of reboots when a PC or laptop attempts to connect... I am sure that the ESP32 core has issues...

crimeacs, 2018-12-07:
I also have the same problem with my ESP32-WROOM. I am able to create AP and connect to it from my phone or PC, but not the other way around. I have tested my ESP32 with several routers, different SSIDs etc, but it always says: Authentication failed.
Nor did it help to erase flash, of do WiFi.disconnect() trick, nor ESP.reboot() trick.

a-c-sreedhar-reddy, 2018-12-20:
Is there anyone who solved this ?  Please share the solution.

GithubMystic, 2019-01-03:
Hi Guys.
I have just posted an Issue (255) regarding ==> Unreliable behaviour on ESP32 but reliable on ESP8266 under MQTT https://github.com/knolleary/pubsubclient/issues/554
Reading the post you guys made, I get the feeling that these two posts might be related.
It would be interesting to see if anyone gets the same symptoms as me when testing the ESP32 as I have done in my post.
Would anyone be prepared to try this ? hopefully this would help to get to the bottom of these issues.

lbernstone, 2019-01-03:
Please do not hijack threads.  Your issue is clearly not the same.

a-c-sreedhar-reddy, 2019-01-03:
hey @lberstone did you find the solution for this problem

GithubMystic, 2019-01-03:
hi @lberston.
I do object somewhat to being accused of hijacking a thread.
In my opinion I can see similarities in the underlying issue of the unreliability of WiFi on the ESP32.
Just because they manifest themselves in the way I am using it as per my post does not mean that the underlying issue  causing the issues in my post are not the same.
The forums are for sharing and helping each other with issues and where someone sees an issue related to theirs, we should all be trying to help each other where possible.
To anyone else reading this post, please do consider that the underlying issues in this post and my post could very well be related, as they pertain to unreliablity of WiFi on the ESP32 that so far has not been identified.
Many thanks

eziosoma, 2019-01-13:
this other issue seems to fix the problem. https://github.com/espressif/arduino-esp32/issues/1147

eziosoma, 2019-01-13:
Tryed #1147, the problem still presents

Pythonaire, 2019-01-15:
Idea ... Me too, stumbling about unstable Wifi connections. I use the ESP32 for sensoring. Now I read in the ESP32 manufacturer description, the ADC2 Pins/GPIO/Interrupts are used by the Wifi driver. So, measuring with ADC2 Pins and Wifi at the same time produce incomprehensible errors. Same thing on other Pins blocked by the systems for internal use.

rodmcm, 2019-02-07:
This is a variation on an above post, but has cured my problems. Disconnect both AP and STA mode with a long delay

		WiFi.disconnect(true);                     // disconnects STA Mode
		delay(1000);
                WiFi.softAPdisconnect(true);           // disconnects AP Mode 
		delay(1000);

rodmcm, 2019-02-07:
I also agree that there seems to be no one taking control of this issue, Seems to me that we as users need to somehow bypass this Forum and send our concerns direct to Espressif.  They are the product manufacturer after all

Another option is to go with another supplier.. Anyone had success with anything else?

baaridunnasr, 2019-02-07:
@LuchoManto 

```
//--------------------------------------------------------------------------------------------------
// Function Details
//--------------------------------------------------------------------------------------------------
// Name : wifiClear
// Purpose : clears all wifi settings before connecting. must be run before every wifiConnect
// Argument(s) : void
// Returns : void

void ESP32Wifi::wifiClear()
{
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    wifiStatus = WiFi.status() == WL_CONNECTED;
    delay(100);
}
//--------------------------------------------------------------------------------------------------

//--------------------------------------------------------------------------------------------------
// Function Details
//--------------------------------------------------------------------------------------------------
// Name : wifiConnect
// Purpose : connects to given ssid using password for given number of tries. updates the wifiStatus.
// Argument(s) : char *SSID, char *PASS, unsigned short tries
// Returns : void

void ESP32Wifi::wifiConnect(char *SSID, char *PASS, unsigned short tries)
{
    _SSID = SSID;
    _PASS = PASS;
    wifiStatus = WiFi.status() == WL_CONNECTED;
    while ((tries > 0) && (!wifiStatus))
    {
        wifiClear();
        WiFi.begin(_SSID, _PASS);
        delay(5000);
        wifiStatus = WiFi.status() == WL_CONNECTED;
        tries--;
    }
}
//--------------------------------------------------------------------------------------------------

//--------------------------------------------------------------------------------------------------
// Function Details
//--------------------------------------------------------------------------------------------------
// Name : wifiDisconnect
// Purpose : disconnects from the network and switches off the radio. sets wifiStatus to false.
// Argument(s) : void
// Returns : void

void ESP32Wifi::wifiDisconnect()
{
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    wifiStatus = false;
}
//--------------------------------------------------------------------------------------------------

```

Code that I use for connecting and disconnecting WiFi. Works pretty good. Solid connects and it kills the  connection and reconnects when I want it to. You could try this and let me know if you have any issues.

GithubMystic, 2019-02-07:
RodMcm
As I mentioned in a previous post, I have given up on the ESP32 / WiFi / MQTT combination due to unreliablility. I have spent far too much time trying all the permutations from forums, etc,...
I have switched to Raspberry Pi zeroW for all my Wifi/Mqtt nodes. These are now rock solid and never miss a beat. As I already use Raspberry pi for my automation hub, it also simplifies my whole environment. Let me know if you need more details

bperkins011, 2019-02-11:
I've had this problem too..  Some times it works, sometimes it doesn't - I've been digging into it and I can't find a pattern..  Except add a line of code and do a complete rebuild - (if it works - it will work on every powerup).  Sometimes it will chuck along for a few minutes and just decide to work for no reason..  It then gets to an HTTP client part of my code where it may struggle along for a couple minutes - then do connect and send a request and get back a response....  i.e.  I've had it just decide to work either instantly - or after sitting there for 3-5 minutes - or not at all..  When it works, it works fine.

If I were to guess - I would say that there is a pointer messed up and on any given build - just enough of the wifi processing gets corrupted to break things..  

Here is some debug code - 

SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
mode : sta(cc:50:e3:27:1f:ae) + softAP(ce:50:e3:27:1f:ae)
add if0
del if0
mode : softAP(ce:50:e3:27:1f:ae)
bcn 0
del if1
usl
mode : null
wifi evt: 8
wifi evt: 8
wifi evt: 8
[WIFI][APlistAdd] add SSID: pontoon
[WIFI] delete old wifi config...
[WIFI] start scan
mode : sta(cc:50:e3:27:1f:ae)
add if0
wifi evt: 8
*
scandone
*
[WIFI] scan done
[WIFI] 4 networks found
       0: [1][18:1B:EB:BE:29:65] 5MZMM (-88) *
 --->  1: [3][24:F5:A2:8F:4C:27] pontoon (-38) *
       2: [6][88:AD:43:25:B0:48] HOME-830C-2.4 (-87) *
       3: [6][4C:8B:30:64:70:47] Actiontec-4961-2.4G (-79) *

[WIFI] Connecting BSSID: 24:F5:A2:8F:4C:27 SSID: pontoon Channel: 3 (-38)
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
wifi evt: 1
STA disconnect: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 4
cnt 

connected with pontoon, channel 3
dhcp client start...
wifi evt: 0
[WIFI] Connecting Failed (6).
*
[WIFI] delete old wifi config...
state: 5 -> 0 (0)
rm 0
[WIFI] start scan
wifi evt: 1
STA disconnect: 8
*
scandone
*
[WIFI] scan done
[WIFI] 4 networks found
[... 248 lines cut ...]
[WIFI] MAC: 24:F5:A2:8F:4C:27
[WIFI] Channel: 3

Connected, IP address: 192.168.1.5
Set my Time at:
0:00:41
processHTTPRequest
[hostByName] request IP for: api.worldweatheronline.com
[hostByName] Host: api.worldweatheronline.com IP: 94.237.49.36
[HTTP] GET... failed, error: read Timeout
processHTTPRequest
[hostByName] request IP for: api.worldweatheronline.com
[hostByName] Host: api.worldweatheronline.com IP: 94.237.49.36
[HTTP] GET... failed, error: read Timeout
processHTTPRequest
[hostByName] request IP for: api.worldweatheronline.com
[hostByName] Host: api.worldweatheronline.com IP: 94.237.49.36
[HTTP] GET... failed, error: read Timeout
processHTTPRequest
[hostByName] request IP for: api.worldweatheronline.com
[hostByName] Host: api.worldweatheronline.com IP: 94.237.49.36
[HTTP] GET... code: 200
{"data":{"request":[{"type":"Zipcode","query":"01821"}],"time_zone":[{"localtime":"2019-02-10 22:21","utcOffset":"-5.0","zone":"America/New_York","sunrise":"06:47 AM","sunset":"05:12 PM","moonrise":"10:12 AM","moonset":"11:53 PM","moon_phase":"First Quarter","moon_illumination":"41"}]}}
wifi evt: 7
wifi evt: 7

tavdog, 2019-03-23:
I am seeing this issue with the Heltec ESP32 Wifi Kit.  Has anyone figured out a fix for this ?

kenken64, 2019-03-30:
3:32:40.039 -> I (33) wifi: wifi driver task: 3ffb4e6c, prio:23, stack:3584, core=0
03:32:40.074 -> I (105) wifi: wifi firmware version: 6b44342
03:32:40.074 -> I (105) wifi: config NVS flash: enabled
03:32:40.074 -> I (107) wifi: config nano formating: disabled
03:32:40.074 -> I (127) wifi: Init dynamic tx buffer num: 32
03:32:40.074 -> I (127) wifi: Init data frame dynamic rx buffer num: 10
03:32:40.074 -> I (127) wifi: Init management frame dynamic rx buffer num: 10
03:32:40.108 -> I (130) wifi: Init static rx buffer size: 1600
03:32:40.108 -> I (134) wifi: Init static rx buffer num: 4
03:32:40.108 -> I (138) wifi: Init dynamic rx buffer num: 10
03:32:40.143 -> I (209) wifi: mode : sta (3c:71:bf:4d:6e:4c)
03:32:40.177 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
03:32:40.177 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
03:32:40.282 -> I (347) wifi: new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
03:32:41.047 -> I (1087) wifi: state: init -> auth (b0)
03:32:41.047 -> I (1089) wifi: state: auth -> assoc (0)
03:32:41.047 -> I (1092) wifi: state: assoc -> init (6c0)
03:32:41.047 -> I (1092) wifi: new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
03:32:41.047 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
03:32:41.047 -> [W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 203 - ASSOC_FAIL
03:32:41.081 -> I (1105) wifi: flush txq
03:32:41.081 -> I (1107) wifi: stop sw txq
03:32:41.081 -> I (1110) wifi: lmac stop hw txq
03:32:41.081 -> I (1120) wifi: mode : sta (3c:71:bf:4d:6e:4c)
03:32:41.081 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
03:32:41.081 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
03:32:41.081 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
03:32:41.185 -> I (1244) wifi: new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1
03:32:41.185 -> I (1244) wifi: state: init -> auth (b0)
03:32:41.220 -> I (1246) wifi: state: auth -> assoc (0)
03:32:41.220 -> I (1249) wifi: state: assoc -> run (10)
03:32:41.220 -> I (1267) wifi: connected with kenken64, channel 9, bssid = 16:0c:c3:e4:26:00
03:32:41.255 -> I (1306) wifi: pm start, type: 1
03:32:41.255 -> 
03:32:41.255 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
03:33:07.705 -> [D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
03:33:07.705 -> [D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.1.56, MASK: 255.255.255.0, GW: 192.168.1.254
03:33:07.705 -> [27686] 
03:33:07.705 ->     ___  __          __
03:33:07.705 ->    / _ )/ /_ _____  / /__
03:33:07.740 ->   / _  / / // / _ \/  '_/
03:33:07.740 ->  /____/_/\_, /_//_/_/\_\
03:33:07.740 ->         /___/ v0.6.1 on ESP32
03:33:07.740 -> 
03:33:07.740 -> [27699] Connecting to blynk-cloud.com:80
03:33:07.808 -> [27803] Ready (ping: 75ms).

yoursunny, 2019-06-05:
I have a unit that won't connect to WiFi. DEBUG logging shows warning `4WAY_HANDSHAKE_TIMEOUT`.
After `esptool.exe --port COM8 erase_flash`, the problem disappears.
You may give this a try.

ESP32 can store some WiFi parameters on the flash, so that it can start connecting to the access point before user sketch starts executing. This improves performance and saves energy when deep sleep is used. However, if the stored WiFi parameters are somehow corrupted, it could cause strange problems.
My sketch normally contains `WiFi.persistent(false)` so that the WiFi parameters aren't stored, but many examples out there do not have this line.

carlerickrowan, 2019-06-08:
> My sketch normally contains `WiFi.persistent(false)` so that the WiFi parameters aren't stored, but many examples out there do not have this line.

I think WiFi.perisistent() implementation is already changed and default to false that's why you don't see it on examples.

eos1d3, 2019-06-27:
Just to share similar thing and I hope that may help a little to find the problem.

I am using Asus AC-86U router (2.4GHz mode) with ESP-32 without problem. When I use Android hotspot to connect repeatedly, it will always fail after a successful connection. This means it will connect at 1, 3, 5 connections, and will always fail at 2, 4, 6....

Using standard code:
```
while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
```

There are two strange things. If I change 500 to 1000 or more, and add `WiFi.begin(ssid, password);` it will always connect. ( I tried 750 but it still fails.)

```
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(1000);
    Serial.print(".");
    
    WiFi.begin(ssid, password);
  }
```

And another strange thing is, Asus 86U works only with standard code. Adding  extra`WiFi.begin(ssid, password);` in while loop it will never work!

tavdog, 2019-06-27:
Why is the ESP32 so bad at this wifi stuff while the ESP8266 works great all the time ?

YHatahet, 2019-07-17:
> Why is the ESP32 so bad at this wifi stuff while the ESP8266 works great all the time ?

I'm guessing it has to do with the fact that the 8266 is cheaper, and hence more people end up using and supporting it. Also ESP32 runs on Free RTOS, and using Arduino IDE to program it is basically running a massive library (according to my understanding, please correct me if I'm wrong).

rumoldus, 2019-08-17:
There is definitely a relation with interrupts on specific pins.
According to documentation, ADC2 is not to be used combined with Wifi. I found on another forum a suspicion that interrupts with the same pins also cause trouble. I also experienced Wifi problems, but when I remove all AttachInterrupts from pins 25, 26 and 32, Wifi works like a charm.

rumoldus, 2019-08-17:
ADC2 uses GPIOs 0, 2, 4, 12 - 15 and 25 - 27. Praise to Pythonaire (see issue 1829).

Ajioz, 2019-09-26:
> @LuchoManto
>
> ```
> //--------------------------------------------------------------------------------------------------
> // Function Details [...]

I get this error while trying to compile it:

ESP32Wifi has not been declared

rebeliousconformist, 2019-11-27:
I am getting an errie behaviour with the ESP32. The ESP32 connects to the wifi network. 
I have a rabbitmq broker working on the server side. 

The ESP32 wifi network is unstable and after reconnection it resend the MQTT message

11. I (951215) MQTT: Other event id:7
I (951215) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
E (951235) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
E (951235) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
5. I (951245) MQTT: MQTT_EVENT_DISCONNECTED
I (951245) MQTT: MQTT_EVENT_CONNECTED
1. I (951255) MQTT: sent publish successful, msg_id=25007
2. I (951255) MQTT: sent subscribe successful, msg_id=4577
3. I (951265) MQTT: sent subscribe successful, msg_id=32327
4. I (951265) MQTT: sent unsubscribe successful, msg_id=45803
8. I (951275) MQTT: MQTT_EVENT_PUBLISHED, msg_id=25007
6. I (951285) MQTT: MQTT_EVENT_SUBSCRIBED, msg_id=4577
I (951285) MQTT: sent publish successful, msg_id=0
6. I (951295) MQTT: MQTT_EVENT_SUBSCRIBED, msg_id=32327
I (951295) MQTT: sent publish successful, msg_id=0
7. I (951305) MQTT: MQTT_EVENT_UNSUBSCRIBED, msg_id=45803
9. I (951315) MQTT: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
9. I (951315) MQTT: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
11. I (966245) MQTT: Other event id:7
I (966245) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
E (966285) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
E (966285) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
5. I (966285) MQTT: MQTT_EVENT_DISCONNECTED
I (966295) MQTT: MQTT_EVENT_CONNECTED
1. I (966295) MQTT: sent publish successful, msg_id=18243
2. I (966305) MQTT: sent subscribe successful, msg_id=31514
3. I (966305) MQTT: sent subscribe successful, msg_id=22672
4. I (966315) MQTT: sent unsubscribe successful, msg_id=36679
8. I (966325) MQTT: MQTT_EVENT_PUBLISHED, msg_id=18243
6. I (966335) MQTT: MQTT_EVENT_SUBSCRIBED, msg_id=31514
I (966335) MQTT: sent publish successful, msg_id=0
6. I (966345) MQTT: MQTT_EVENT_SUBSCRIBED, msg_id=22672
I (966345) MQTT: sent publish successful, msg_id=0
7. I (966355) MQTT: MQTT_EVENT_UNSUBSCRIBED, msg_id=36679
9. I (966355) MQTT: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
9. I (966365) MQTT: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data

Any idea how to work around this ?

Im using ESP-IDF framework

kingyyp, 2020-01-29:
I am using a ESP32-cam for monitoring , however, the video stream is not showing regularly , mostly after 5-6 hours from power on.  i can see someone try raspberrypi as alternative, will try that as well. see if raspberrypi can be more stable.

ligantx, 2020-03-18:
Just some more information on this..
when i tried to connect to Android hotspot i had many connection issues even if i tried the cures above. Sometimes it is working sometimes is not..i haven't found any pattern for the successful connections. 

I changed to my home router (not hotspot) and everything works fine

MrPivato, 2020-04-23:
I have too a similar problem regarding WiFi connection via the ESP32 WROOM

**Using the code from example:**
```c
#include <WiFi.h>

const char* ssid     = "Vortyce";
const char* password = "*******";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

int value = 0;

void loop()
{
    delay(5000);
    ++value;

    Serial.print("connecting to ");
    Serial.println(host);

    // Use WiFiClient class to create TCP connections
    WiFiClient client;
    const int httpPort = 80;
    if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
    }

    // We now create a URI for the request
    String url = "/input/";
    url += streamId;
    url += "?private_key=";
    url += privateKey;
    url += "&value=";
    url += value;

    Serial.print("Requesting URL: ");
    Serial.println(url);

    // This will send the request to the server
    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            Serial.println(">>> Client Timeout !");
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server and print them to Serial
    while(client.available()) {
        String line = client.readStringUntil('\r');
        Serial.print(line);
    }

    Serial.println();
    Serial.println("closing connection");
}
```

**Gives me the following error:**
```
Connecting to Vortyce...............
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d4f1c: 3d196892 3dd03df0 81f03df0
Core 0 register dump:
PC      : 0x400d4f21  PS      : 0x00060633  A0      : 0x800d53c1  A1      : 0x3ffbbfa0  
A2      : 0x3ffbd2bc  A3      : 0x3ffbbfcf  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060620  A7      : 0x00000000  A8      : 0x3ff5f000  A9      : 0x00000000  
A10     : 0x3ffbc0e8  A11     : 0x00060623  A12     : 0x00060623  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d4f21:0x3ffbbfa0 0x400d53be:0x3ffbbfc0 0x400d53cf:0x3ffbbff0 0x400d6379:0x3ffbc010 0x4008a361:0x3ffbc030 0x40088b7d:0x3ffbc050

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
```

lbussy, 2020-04-29:
I used a modification of one of the above before connecting WiFi:

```
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);
```

So the theory that the radio stays connected somehow when flashing and resetting (I am using esptool) seems valid on the outside anyway.

closed 2020-07-12 by stale[bot]

MurtadhaS, 2020-12-16:
Just connect the ESP32 to the 5v instead of 3.3v.
That solved my wifi connection issues. :)

acicuc, 2022-10-18:
I've been testing a couple of boards. Both display this issue. Power supply is rock solid. It looks like it could be a timing problem but I haven't had any time to look very far into this. For me adding Wifi.scanNetworks() prior to WiFi.begin() helped a lot, connecting to the AP was fast and reliable after that.

Patronics, 2022-11-25:
Does anyone know whether newer ESP32 variants such as the ESP32-S3 have resolved these issues? Can I confidently get them for use with new projects, without having to worry about having the wifi difficulties that I and many others had in the past?

GerryBriggs, 2024-02-18:
My experience is that about a third of all wifi routers do not get along with ESP32.  The other2/3 are fine.