espressif/arduino-esp32 #1213: BluetoothSerial.cpp: In member function 'virtual int BluetoothSerial::read()':

adisik, 2018-03-14:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Wemos Lolil
Core Installation/update date:			LAst
IDE name:							Arduino IDE 1.6.12
Flash Frequency:					40Mhz
Upload Speed:						912000

### Description:
Dont compile example SerialToSerialBT. I get error - BluetoothSerial.cpp: In member function 'virtual int BluetoothSerial::read()':

Solution:
Add at the end function  return 0; 
int BluetoothSerial::read(void).

Sketch - Example from librarys - SerialToSerialBT.ino.

copercini, 2018-03-14:
I think it's the same as https://github.com/espressif/arduino-esp32/issues/1207

copercini, 2018-04-03:
Done https://github.com/espressif/arduino-esp32/commit/25dff4f044151f7f766c64b9d2ad90398472e6b3

closed 2018-04-03 by copercini