espressif/arduino-esp32 #1215: Feature Request:  availableForWrite()

CombiesGit, 2018-03-14:
### Hardware:
Board:							All ESP32
Core Installation/update date:			pre 14/march/2018

In HardwareSerial the  availableForWrite() method is missing.
[https://www.arduino.cc/reference/en/language/functions/communication/serial/availableforwrite/](url)

### Sketch:
```cpp
HardwareSerial Serial1(2); 

void setup() 
{
  Serial.begin(9600);
  Serial.println("");
  Serial.println("Start");

  Serial1.begin(9600);

}

void loop() 
{
  int tmp = Serial.availableForWrite();
  Serial.print("Serial.availableForWrite(): ");Serial.println(tmp);

  tmp = Serial1.availableForWrite();
  Serial.print("Serial1.availableForWrite(): ");Serial.println(tmp);
  delay(1000);
}
```

### Possible modification:

In HardwareSerial.h add prototype
```cpp
int availableForWrite(void);
```

In HardwareSerial.cpp add methode
```cpp
int HardwareSerial::availableForWrite(void)
{
      return uartAvailableForWrite(_uart);
}
```

In esp32-hal-uart.h add prototype
```cpp
uint32_t uartAvailableForWrite(uart_t* uart);
```

In esp32-hal-uart.c add function
```cpp
uint32_t uartAvailableForWrite(uart_t* uart)
{
    if(uart == NULL) {
        return 0;
    }
    return   0x7f - uart->dev->status.txfifo_cnt;
}
```

lbernstone, 2018-03-14:
If you have written and tested this, then please fork the project and add your code as a pull request.

JamesGKent, 2018-03-16:
one possible headache with this is that there isn't really a way to account for the multi threaded nature of the esp32, if for example something on core0 calls `availableForWrite` then something one core1 writes to the uart, it could invalidate the answer given to the thread on core0 before it has a chance to write to the uart.

i think it would still be worth adding this for api compatibility with arduino so that libraries can be used easily, but it may lead to strange hard to diagnose issues in some cases...

mtabu, 2018-03-16:
If you want to do non-blocking write, you always needs to have control of writers. You would probably not let several threads write to the uart.
Your concern is only valid if you make programs that does uart updates in different cores.
The "hard to diagnose issues" is already present in the available() function. If you call read() after a available() call, another thread can have read the data before you get to the read().
The issue for availableForWrite() is "only" that what is expected to be a non-blocking call to write is blocked.  

I have made a pull request (1226) with similar code as above. It works great. I have used this to update a Nextion display without blocking. What I do is to delay commands to a later loop() iteration, when the uart output buffer cannot receive the whole next command I want to send.

ttlappalainen, 2018-04-06:
Theoretically one can have problem with available()/availableForWrite() on different threads. But could some give me real example, why in earth I would have two threads reading same serial?

stickbreaker, 2018-04-06:
@ttlappalainen you are outputting data to `Serial()` between your `availableForWrite()` and `printf()` `esp_loge()` says bad stuff happened, or even inside the `printf()`, each parameter can generate its own call to `write()`.

I don't have a read example. 

There is another issue with the same subject, I recommended that the original poster (OP) just create an atomic write: `bool HardwareSerial::atomicWrite(uint8_t * buffer, uint8_t len)` this function would either post the entire buffer[] returning `true` or post nothing and return `false`.  Inside of `HardwareSerial()` you have access to the uart semaphore.  You can then acquire the semaphore, check buffer depth, post or return the error.

Chuck.

ttlappalainen, 2018-04-06:
Printf etc. does, what they do. But if I need non blocking system, I need to handle everything myself and just write bytes from some own buffer. Just now I can not imagine sample, where I would mix non blocking write and printf - it just does not sound good design.

Note also that on Arduinos availableForWrite is on Stream.

closed 2019-08-15 by stale[bot]