espressif/arduino-esp32 #1216: Unable to upload program to NodeMCU esp32s

godzsa, 2018-03-14:
### Hardware:
Board:							Node MCU ESP32S
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						921600

### Description:
I am trying to upload any sketch, but i get this error:

```
Sketch uses 167257 bytes (12%) of program storage space. Maximum is 1310720 bytes.
Global variables use 11212 bytes (3%) of dynamic memory, leaving 283700 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.1
Connecting........___
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 31328.7 kbit/s)...
Hash of data verified.
Compressed 14128 bytes to 9196...

Writing at 0x00001000... (100 %)
Wrote 14128 bytes (9196 compressed) at 0x00001000 in 0.1 seconds (effective 1066.2 kbit/s)...
File  md5: bd7911602e3b22bcb002bd514b2c87aa
Flash md5: 74b4e9a31a1d0c94fd2b544e978a4e67
MD5 of 0xFF is 3e505afdc356decdf9bc304ad6b9bfb3

A fatal error occurred: MD5 of file does not match data in flash!
A fatal error occurred: MD5 of file does not match data in flash!
```

It worked fine before, I do not know what happened. Later I think I messed something up because I pushed the EN and Boot buttons, after I could not upload the program, I am really new to arduino, so I have no clue on how to fix this issue. Could you help me please?

nickoloss, 2018-03-14:
I had the same problem with an esp32 devboard. The problem appeared while using wrong Flash mode. Check your Flash configuration.

godzsa, 2018-03-14:
@nickoloss I have tried all the flash modes, but still no progress. Could you describe what settings to use, or where can I find what is suitabe for NodeMCU esp32s?

godzsa, 2018-03-14:
Ok I wanted to build a circuit with a dht22 and when i flash with the dht in the circuit I get the erro, but without it it will flash. Does that mean, that my circuit is not correct?

stickbreaker, 2018-03-14:
Which pins are you using?

godzsa, 2018-03-14:
p12

godzsa, 2018-03-14:
i have tried with p16 and p2 as well but no luck, shall I upload a picture or flash settings or something?

kosso, 2018-03-16:
Try setting the baud rate to 115200.

godzsa, 2018-03-17:
Okay, so I solved it by using pin16 and no breadboard. I don't know what the problem is with breadboard, I have tried 2 of them, but with no luck. Without it it works fine. It also does not work with pin12. 
I used it like this: 
[image]

closed 2019-08-15 by stale[bot]