espressif/arduino-esp32 #1217: Expansion of NVS area and change of write start address

akaneko1019, 2018-03-15:
I am sorry in my unfamiliar English.

I am currently trying to expand the NVS domain. The reason is that I want to handle many setting values ​​in Preference.
The extension itself was very simple. Just to adapt the NVS domain extension on a custom partition.

You can also check to the operation.
But a problem occurred.
When programming using Arduino IDE, the writing start address is fixed at 0x10000.
I have adapted the following partitions.

-------------------------------------------------------------------- - -
# Name, Type, SubType, Offset, Size, Flags
nvs, data, nvs, 0x9000, 0x15000,
otadata, data, ota, 0x1e000, 0x2000,
app 0, app, ota - 0, 0 × 20000, 0 × 140000,
app 1, app, ota - 1, 0 × 160000, 0 × 140000,
eeprom, data, 0x99, 0x2A0000, 0x1000,
spiffs, data, spiffs, 0x2A1000, 0x15F000,

-------------------------------------------------------------------- - -

The above is an extension of NVS by 0x10000.
Therefore, the start address of the APP area will be 0x20000.
However, since writing start address is 0x10000 when writing with Arduino IDE, it was impossible to run the program even if it was written.
I tried writing the file compiled with Arduino IDE directly to 0x20000 using esptool, but it works.

How can I change the Arduino IDE's write start address?
Thank you.

lbernstone, 2018-03-15:
You can modify the partition table file at <arduino-esp32>/tools/partitions/default.csv

akaneko1019, 2018-03-15:
I already tried changing the partition table. Of course, if you write the compiled software directly with esptool, no problems will occur.

The problem is that app writing start address instructed to esptool when writing with Arduino IDE is fixed at 0x10000.
Please tell me how to change this address,

lbernstone, 2018-03-16:
Sorry, I didn't read your post carefully. Search in platform.txt for esptool.upload.pattern

akaneko1019, 2018-03-16:
I tried that method, but it was useless.
I do not know whether it is fixed in another place or bug.

akaneko1019, 2018-03-19:
I could not do it at all, but I could do it by restarting the PC.
Sorry for self-resolving.
Thank you very much.

Close.

closed 2018-03-19 by akaneko1019

beegee-tokyo, 2018-03-30:
Added to the Wiki: [Expansion of NVS partition requires adapt default app partition address in esptool](https://desire.giesecke.tk/index.php/2018/03/30/expansion-of-nvs-partition-requires-adapt-default-app-partition-address-in-esptool/)