espressif/arduino-esp32 #1218: error :  pbuf_free: p->ref > 0" failed

robademar, 2018-03-15:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
 
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						115200

### Description:
I have the following error I don't know how to analize.
The sketch has 3 tasks and it reads temperatures, gas sensors values and is using wifi to connect to blynk.
The error appeared twice in 3 hours.

assertion "pbuf_free: p->ref > 0" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/pbuf.c", line 694, function: pbuf_free
abort() was called at PC 0x400d825b on core 1

Backtrace: 0x400882f0:0x3ffd11c0 0x400883ef:0x3ffd11e0 0x400d825b:0x3ffd1200 0x400ee4fb:0x3ffd1230 0x400eb1ba:0x3ffd1250 0x400eba9d:0x3ffd12d0 0x400ebb0e:0x3ffd1300 0x400d2971:0x3ffd1320 0x401286d2:0x3ffd1340 0x400d4b6d:0x3ffd1370 0x400d4b99:0x3ffd1390 0x400d1dd6:0x3ffd13b0 0x400d21d3:0x3ffd1410 0x400d2633:0x3ffd1440 0x40111c64:0x3ffd1460

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

```

stickbreaker, 2018-03-16:
@robademar How are you serializing your access to individual resources?  You say you have three tasks that read gas sensors, and publish the results using WiFi?

You do realize you only have one WiFi radio? You are using and individual `WiFiClient()` for each task?
how are you communicating with your sensors?
  
I would say based on the file name where the error occurred, you are using a component that is not designed as multithread capable with multiple threads.

Chuck.

robademar, 2018-03-16:
Thanks Chuck for your message.I do not have direct access to wifi but I use Blynk platform for it . There is one task with the highest priority using cup 0 dedicated to wifi.
Inviato da Yahoo Mail su Android

closed 2019-08-15 by stale[bot]