espressif/arduino-esp32 #1219: Using 2 SPI ports on ESP32 ?

SrikanthBala, 2018-03-15:
### Hardware:
Board:							ESP32 Dev Kit
Core Installation/update date:	        Feb 2018
IDE name:					        Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200?

### Description:
I am using a SD card to store values from a RFID reader  RC522. 
Both use the SPI bus, and are known to have compatibility issues. I have connected them together and when I remove power of SD card RC522 works well and when I remove MOSI of RC522 SD card works

So i wanted to use the second SPI. I saw this issue #790 https://github.com/espressif/arduino-esp32/issues/790 but am still not sure how to use it. Could anyone help me ?
What pins do I use for HSPI ? or are they reconfigureable. A simple example would be great.

lbernstone, 2018-03-15:
I don't have a full example for you, but here's some tips. The pinmap is on the front page of the arduino-esp32 project.  You can remap SPI pins to (almost) whatever you want, but you might as well use the well tested defaults. Initialize the SPI with:
SPIClass spiRFID(HSPI);
spiRFID.begin(14,12,13,15); //CLK,MISO,MOIS,SS

Then you should be able to use spiRFID in your library instead of SPI.

SrikanthBala, 2018-03-15:
Got the RC522 to work on HSPI with a few mods to the library !! Now SD card and RC522 both working together. Thanks a million Ibernstone !

I'm not sure I did everything according to convention though. I defined SPIClass spiRFID(HSPI) in the MRFC522 library MRFC522.cpp file. I changed all 'SPI.xxx' references to 'spiRFID.xxx'. I then defined the same as 'extern SPIClass spiRFID ;' in my main sketch and put spiRFID(14,12,13,15); in the setup of the main Sketch. 
If I didn't define extern SPIClass spiRFID in my sketch spiRFID(14,12,13,15) would flag an error.

So was this how it was supposed to be done ?

Thanks

lbernstone, 2018-03-15:
If it works, then it works.  The correct way to do it is to make it so the library can accept an SPIClass object in its initialization which defaults to SPI.  Take a look at the SD library to see how they do it there.  That could then be given back to the library maintainer so the next guy doesn't have to muck about as much.

SrikanthBala, 2018-03-16:
Ok. Will do. Thanks again.

closed 2018-03-16 by SrikanthBala

fa1ke5, 2018-12-09:
This code work great, no modification SD or SPI lbr
#include "FS.h"
#include "SD.h"
#include "SPI.h"
SPIClass spiSD(HSPI);
#define SD_CS 13
#define SDSPEED 27000000
void setup(){
    Serial.begin(115200);
    spiSD.begin(14,2,15,13);//SCK,MISO,MOSI,ss // инициализируем HSPI1
    if(!SD.begin( SD_CS, spiSD, SDSPEED)){
   // if(!SD.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
}
void loop() {
}

erikhjertholm, 2019-05-31:
> This code work great, no modification SD or SPI lbr
> #include "FS.h"
> #include "SD.h"
> #include "SPI.h"
> SPIClass spiSD(HSPI);
> #define SD_CS 13
> #define SDSPEED 27000000
> void setup(){ [...]

Hi, I have tried this code, but do not have any success in writing to, or creating a file. The SD.begin does return true, so it seems to be mounted ok. Also, everything works fine when using the vspi pins..

Any idea on how to fix this?

mellowmonk, 2019-10-12:
@SrikanthBala 
Please share the sketch that worked too. Thanks.

froussel, 2020-01-01:
Hi, Happy new year
I want connect ggto-t-display (use 19, 18, 5, 16, 23, 4 for SPI display ST7789V)  and a SD card.
How ?
Thanks

fa1ke5, 2020-01-03:
connect sd card using display SPI and use another  CS pin, whats a problem?

froussel, 2020-01-03:
> connect sd card using display SPI and use another CS pin, whats a problem?

The pins of thedisplay are not accessible and even if it was the case it was impossible to make a link screen then sd then screen etc... without redoing a .begin each time or so I have to explain
Excuse me i am a beginner

fa1ke5, 2020-01-03:
As I know you need use class definition of you need using any function more than one.
See my code below.
SPIClass spiSD(HSPI);
and
spiSD.begin(14,2,15,13);
for one instance, 
SPIClass xxxxx(zzzz);
Xxxxx.begin(yyyy); 
For second instance....now problems.
Do xxxx.begin, yyy.begin in setup() and use both in loop() as xxx.write(), yyy.write()
You may use some spi for for any devices grow class definition and different CS pin for other device....

froussel, 2020-01-05:
Thank you
However, pin 14 is already used by the screen
Can we really put any pin like misi, mosi and clk in spiSD.begin(x, y, z, t) ?

fa1ke5, 2020-01-05:
Yes, you can use other pins, but forwarding through gpio matrix degrade you spi port sped, 26mhz maximum as I know

hern4ndes, 2020-03-01:
after many time i found  this library, worked for me http://www.iotsharing.com/2017/05/how-to-use-arduino-esp32-to-store-data-to-sdcard.html

SaeedNazarian, 2020-06-11:
The issue is probably the sd card module. the level shifter ic on the module does not let go of the miso pin

theMubashir919, 2020-11-02:
@SrikanthBala @lbernstone 
> I'm not sure I did everything according to convention though. I defined SPIClass spiRFID(HSPI) in the MRFC522 library MRFC522.cpp file. I changed all 'SPI.xxx' references to 'spiRFID.xxx'. I then [...]

Thank you very much. I was stuck on a similar issue for 2 weeks. 
I'll explain clearly how my issue was resolved (I used SPI2 instead of spiRFID) :

1. Define `SPIClass SPI2(HSPI);` in the beginning of MFRC522.cpp in Documents\Arduino\libraries\MFRC522-1.4.7\src\
2. Replace all 'SPI.xxx' references to 'SPI2.xxx' like `SPI.transfer(reg);` becomes `SPI2.transfer(reg);`
3. Define `extern SPIClass SPI2;` in your main sketch outside void setup
4. Put `SPI2.begin(14,12,13,15); //CLK, MISO, MOIS, SS` in void setup

If someone doesn't want to do step 1 and 2, I've already attached my modified MFRC522.cpp file here. Just extract and replace the original one in Documents\Arduino\libraries\MFRC522-1.4.7\src\ with this one. Remember to save a backup of the original just in case. Then follow the steps 3 and 4 in main sketch.

[modified_MFRC522.zip](https://github.com/espressif/arduino-esp32/files/5476254/modified_MFRC522.zip)

e-yazici, 2020-11-05:
@theMubashir919 ,
Thanks for sharing the modified .cpp file but when i tried this, I'm getting compiling error. Did you get that same thing or do you have any idea to solve this? I'm a newbie and couldn't figure it out.

> libraries\rfid-master\MFRC522.cpp.o:(.bss.SPI2+0x0): multiple definition of `SPI2'
sketch\NFC_ID-byte.ino.cpp.o:(.bss.SPI2+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

`#include <SPI.h>
#include <MFRC522.h>

SPIClass SPI2(HSPI);

//#define RST_PIN 4
//#define SS_PIN 15
String ID;

MFRC522 mfrc522(15, 4);   // Create MFRC522 instance.
  
void setup() {  
   Serial.begin(115200);
   SPI2.begin(14, 12, 13, 15);      // Initiate  SPI bus
   mfrc522.PCD_Init();   // Initiate MFRC522  
   Serial.println("Approximate your card to the reader...");   
}
void loop{}`

Actually RC522 works on both HSPI and VSPI but I need to add SD card too. It will log the uid to sd card.

theMubashir919, 2020-11-06:
@e-yazici 
> `#include <SPI.h>
> #include <MFRC522.h>
> 
> SPIClass SPI2(HSPI);
> 

Have you tried using `extern SPIClass SPI2;` here instead of `SPIClass SPI2(HSPI);` ?

e-yazici, 2020-11-07:
Yeap, but it didn't work. I solved the issue with SPI library. I added `SPIClass SPInfc(HSPI);` in SPI.cpp and `extern SPIClass SPInfc(HSPI);` in SPI.h (*`SPI2(HSPI)` didn't work). Then modified all of the `SPI2.xxx` to `SPInfc.xxx` in MFRC522.cpp. With this modification, there is no need to SPIClass in main sketch, just using `SPInfc.begin(CLK,MISO,MOSI,SS);` in void Setup.

Thanks again for your step by step explanation. It gave me a starting point.