espressif/arduino-esp32 #1221: Wear Leveling compilation error in Arduino Core.

PhilColbert, 2018-03-15:
Updated Arduino Core 7 days ago and latest IDF 

When setting the wear levelling library to 512 in menuconfig, the arduino core will not compile and fails with 

C:/msys32/esp/esp-idf/AirWhere_v3.test/components/arduino/libraries/SD/src/SD.cpp: In member function 'uint64_t fs:                                                     :SDFS::totalBytes()':
C:/msys32/esp/esp-idf/AirWhere_v3.test/components/arduino/libraries/SD/src/SD.cpp:85:19: error: 'struct FATFS' has                                                      no member named 'ssize'
         *(fsinfo->ssize);
                   ^
C:/msys32/esp/esp-idf/AirWhere_v3.test/components/arduino/libraries/SD/src/SD.cpp: In member function 'uint64_t fs:                                                     :SDFS::usedBytes()':
C:/msys32/esp/esp-idf/AirWhere_v3.test/components/arduino/libraries/SD/src/SD.cpp:99:19: error: 'struct FATFS' has                                                      no member named 'ssize'
         *(fsinfo->ssize);
                   ^
make[1]: *** [/esp/esp-idf/make/component_wrapper.mk:274: libraries/SD/src//SD.o] Error 1
make: *** [C:/msys32/esp/esp-idf/make/project.mk:450: component-arduino-build] Error 2

Any ideas?

Would like to use the smaller partition size and this makes the compliation fail.

lbernstone, 2018-03-15:
Try checking out commit 1c3dd23fe0c547c5cdcd6a01c111145c572dab91 from esp-idf.  This is the code that matches the last arduino update.

PhilColbert, 2018-03-16:
This commit ? /confused ?

https://github.com/espressif/esp-idf/commit/1c3dd23fe0c547c5cdcd6a01c111145c572dab91

lbernstone, 2018-03-16:
That was the point at which arduino-esp32 was last compiled, so everything should be matched up b/w the code bases.

closed 2019-08-15 by stale[bot]