espressif/arduino-esp32 #1224: Critical section issue

elotn, 2018-03-16:
### Hardware:
Board:							OLIMEX ESP32-EVB
Core Installation/update date:			esp32-idf / esp32-arduino 9.03.2018 latest master branch
IDE name:							Arduino 1.8.4
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
When I'm trying enter critical section during reading from SD card I get Core  0 panic'ed (Interrupt wdt timeout). I increased interrupt wdt timeout to CONFIG_INT_WDT_TIMEOUT_MS 5000 and problem still exists.  In final application I need critical section to avoid context switching while sending data to HTTP client. I was also trying vTaskSuspendAll and xTaskResumeAll without success.

I'm using [Ethernet.zip](https://github.com/espressif/arduino-esp32/files/1818477/Ethernet.zip) library because of (https://github.com/espressif/arduino-esp32/issues/501) for OLIMEX ESP32-EVB I wonder if this could cause the problem.

Thank you in advance for any help.

This problem can be easily replicated  - check loops :

### Sketch:
```cpp

#include <stdio.h>
#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_vfs_fat.h"
#include "driver/sdmmc_host.h"
#include "driver/sdmmc_defs.h"
#include "sdmmc_cmd.h"

static const char* TAG = "example";

portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;

FILE* f;

void setup() {
  Serial.begin(115200);
  ESP_LOGI(TAG, "Initializing SD card");

  sdmmc_host_t host = SDMMC_HOST_DEFAULT();

  // To use 1-line SD mode, uncomment the following line:
  host.flags = SDMMC_HOST_FLAG_1BIT;
  host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;

  // This initializes the slot without card detect (CD) and write protect (WP) signals.
  // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
  sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();

  // Options for mounting the filesystem.
  // If format_if_mount_failed is set to true, SD card will be partitioned and formatted
  // in case when mounting fails.
  esp_vfs_fat_sdmmc_mount_config_t mount_config = {
    .format_if_mount_failed = false,
    .max_files = 5
  };

  // Use settings defined above to initialize SD card and mount FAT filesystem.
  // Note: esp_vfs_fat_sdmmc_mount is an all-in-one convenience function.
  // Please check its source code and implement error recovery when developing
  // production applications.
  sdmmc_card_t* card;
  esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);
  if (ret != ESP_OK) {
    if (ret == ESP_FAIL) {
      ESP_LOGE(TAG, "Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.");
    } else {
      ESP_LOGE(TAG, "Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place.", ret);
    }
[... 132 lines cut ...]

**DEBUG FOR LOOP2**
[I][SD_Test.ino:99] setup(): Read from file: 'Hello SC16G!'
[I][SD_Test.ino:104] loop(): Reading file
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1445 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40085467 on core 1

Backtrace: 0x40087f64:0x3ffca210 0x40088063:0x3ffca230 0x40085467:0x3ffca250 0x400d6bff:0x3ffca290 0x400d4015:0x3ffca2b0 0x400d4c99:0x3ffca2d0 0x400d57de:0x3ffca2f0 0x400d71ff:0x3ffca330 0x400db91a:0x3ffca360 0x4000bdbb:0x3ffca380 0x40001125:0x3ffca3a0 0x400594e9:0x3ffca3c0 0x400dd0fb:0x3ffca3e0 0x400d1121:0x3ffca400 0x400ee661:0x3ffca420

x40085467: wDev_ProcessFiq at ?? line ?
0x40087f64: get_prev_free_block at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap.c line 172 (discriminator 1)
0x40088063: assert_valid_block at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap.c line 147
0x40085467: wDev_ProcessFiq at ?? line ?
0x400d6bff: read_encoded_value_with_base at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind-pe.h line 207
0x400d4015: ETHClass::setHostname(char const*) at C:\Users\Tomek\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src/ETH.cpp line 222
0x400d4c99: EthernetServer::begin() at C:\Users\Tomek\Documents\Arduino\hardware\espressif\esp32\libraries\Ethernet\src/EthernetServer.cpp line 117
:  (inlined by) EthernetServer::begin() at C:\Users\Tomek\Documents\Arduino\hardware\espressif\esp32\libraries\Ethernet\src/EthernetServer.cpp line 75
0x400d57de: Print::printf(char const*, ...) at C:\Users\Tomek\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Print.cpp line 227
0x400d71ff: _Unwind_Resume at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind.inc line 224
0x400db91a: vfs_fat_link at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/vfs_fat.c line 509
0x400dd0fb: configure_pin at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./sdmmc_host.c line 298
:  (inlined by) sdmmc_host_init_slot at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/driver/./sdmmc_host.c line 338
0x400d1121: _flash_cache_start at ?? line ?
0x400ee661: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c line 1411 (discriminator 105)
```

elotn, 2018-04-26:
I just updated esp32-idf and esp32-arduino and unfortunately I still have this problem. Without critical section while reading from sd card or sending data to Http client it's impossible to achieve good web server performance. Any suggestions would be appreciated.

stickbreaker, 2018-04-26:
@elotn Why do you  need a critical section? 
is your webserver code sending one character at a time? I wrote a webserver using MEGA2560, and found that the ethernet library worked best with 256byte blocks, I encapsulated `client.print()` calls with a block buffering system.  I would saw an improvement from 5kB/s to 50kB/s.  I had to add a `bufferFlush()` at the end of `client()` session. 

Now with the ESP32 I would expect much better throughput, I believe that blocking would still benefit, but, probably the block size would be different. (Bigger I hope!)  Context switching only matters if you have multiple tasks ready, or a higher priority task need service. How complex (number of tasks) is your system? 

Chuck.

elotn, 2018-04-27:
Unfortunately in my system i have multiple tasks, also tasks with high priorities. I wonder why critical section cause core panic in this simple example described above.

closed 2019-08-15 by stale[bot]