espressif/arduino-esp32 #1227: i2s get data from internal ADC

wizz777, 2018-03-17:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module?
Core Installation/update date:			11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Try to get data from internal ADC from i2c
but in RX data only zero
mic-  MAX9812
Pin36 -> Mic OUT
GDN -> Mic GRND
3V -> Mic VCC

### Sketch:
#include "driver/i2s.h"
#include "driver/adc.h"

int i2s_num = 0; 
i2s_config_t i2s_config;

char* buf1= (char*) malloc(32768*2);

void example_disp_buf(uint8_t* buf, int length)
{

    Serial.print("======\n");
    for (int i = 0; i < length; i++) {
        Serial.printf("%02x ", buf[i]);
        if ((i + 1) % 8 == 0) {
            Serial.print("\n");
        }
    }
    Serial.print("======\n");

}

void setup() {
Serial.begin(115200);

  //config i2s
i2s_config.mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN);
  i2s_config.sample_rate = 48000;
  i2s_config.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT;
  i2s_config.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT;
  i2s_config.communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_I2S |     I2S_COMM_FORMAT_I2S_MSB);
  i2s_config.dma_buf_count = 2;
  i2s_config.dma_buf_len = 512;
  i2s_config.intr_alloc_flags = ESP_INTR_FLAG_LEVEL1;

   //install and start i2s driver
  if(i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL)==ESP_OK) Serial.println("1 driver OK");
  if ( i2s_set_pin(I2S_NUM_0, NULL)==ESP_OK) Serial.println("2 pin OK");
  if (i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_0)==ESP_OK) Serial.println("3 mode OK");
  if (i2s_start(I2S_NUM_0)==ESP_OK) Serial.println("5 i2s OK");
}

void loop() {
int i= i2s_read_bytes(I2S_NUM_0, buf1, 32768*2, portMAX_DELAY);

Serial.println(i);

 example_disp_buf((uint8_t*) buf1, 164);
delay(550);
```

### Debug Messages:
Out in Serial
1 driver OK
2 pin OK
3 mode OK
5 i2s OK
65536
======
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 ======
65536
======
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 
00 00 00 00 ======
65536

hongthiet1994, 2018-09-28:
Did you slove this problem?

wizz777, 2018-09-28:
No

hongthiet1994, 2018-09-29:
> No

Do you have skype account, can we discuss?

Matheus-Garbelini, 2018-10-20:
Any progress?

lbernstone, 2018-10-20:
The [IDF example](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2s_adc_dac/main/app_main.c#L214) includes i2c_adc_enable(), which I'm guessing is the missing #4 in your list.

closed 2019-08-15 by stale[bot]