espressif/arduino-esp32 #1231: Wrong Git Bash session command in Windows installation instructions?

frownbreaker, 2018-03-17:
There is a mistake in the documentation for windows installation: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

Specifically, instructions state: "open a Git Bash session pointing to ARDUINO_SKETCHBOOK_DIR and execute git submodule update --init --recursive"

```
$ **git submodule update --init --recursive**
fatal: Not a git repository (or any of the parent directories): .git
```

The command **_should_** read "`git clone --recursive -j8 https://github.com/espressif/arduino-esp32.git`"

```
$ git clone --recursive -j8 https://github.com/espressif/arduino-esp32.git
Cloning into 'arduino-esp32'...
remote: Counting objects: 7326, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 7326 (delta 0), reused 0 (delta 0), pack-reused 7323
Receiving objects: 100% (7326/7326), 108.12 MiB | 3.56 MiB/s, done.
Resolving deltas: 100% (4196/4196), done.
Checking out files: 100% (1304/1304), done.
Submodule 'libraries/BLE' (https://github.com/nkolban/ESP32_BLE_Arduino.git) registered for path 'libraries/BLE'
Cloning into 'C:/Users/seth/Documents/Arduino/arduino-esp32/libraries/BLE'...
remote: Counting objects: 265, done.
remote: Total 265 (delta 0), reused 0 (delta 0), pack-reused 265
Receiving objects: 100% (265/265), 157.62 KiB | 653.00 KiB/s, done.
Resolving deltas: 100% (146/146), done.
Submodule path 'libraries/BLE': checked out '6bad7b42a96f0aa493323ef4821a8efb0e8                          815f2'
```

The solution from here https://github.com/espressif/arduino-esp32/issues/793 see "palomarcoder commented on Dec 27, 2017"

MarkyAD, 2018-03-17:
The command "git submodule update --init --recursive" is valid but the documentation doesn't make clear that it needs to be executed from within the directory you clone the repository to. If you clone into arduino-esp32 then you need to cd arduino-esp32 before updating submodules.

beegee-tokyo, 2018-03-17:
I wrote a post in the [Wiki](http://desire.giesecke.tk/index.php/2018/02/27/alternative-to-install-arduino-esp32-for-arduino-ide-windows/) that shows the installation steps.

closed 2019-08-15 by stale[bot]