espressif/arduino-esp32 #1233: esp32 WiFi.hostByName returns 0.0.0.0

rokrodic, 2018-03-18:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation/update date:			17/mar/2018
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					        80Mhz
Upload Speed:						921600

### Description:
I use NTP library which worked OK on ESP8266. For the "si.pool.ntp.org" I get returned "193.2.4.2" when using the sketch below. The ESP32 returns "0.0.0.0"

### Sketch:
```cpp
  char ntpServerName[] = "si.pool.ntp.org";
  IPAddress ntpServerIP;
  while (Udp.parsePacket() > 0) ;
  WiFi.hostByName(ntpServerName, ntpServerIP);
```

Solution? Or bug?

lbernstone, 2018-03-18:
Try adding this to see if you are getting a DNS server from your DHCP:
`  ip_addr_t dnssrv=dns_getserver(0);
  Serial.print("DNS server: ");
  Serial.println(inet_ntoa(dnssrv));
`
I just tested on a dev board and the address was resolved correctly.

rokrodic, 2018-03-19:
Hi. It is odd. When I added your two includes and your three lines of code, I get IP from hostByName?!?! Cannot explain it, but I haven't looked into dns.h and inet.h yet.
I checked my router and it has entered DNSes. Your code returns my router address - DNS server: 192.168.1.1 and NTP works.
Thank you very much for your effort!!!

lbernstone, 2018-03-19:
[image]

stickbreaker, 2018-03-19:
@lbernstone Dating your self.  I doubt most on this forum understand the reference!

Chuck.

lbernstone, 2018-03-19:
Ayyyy.  @rokrodic, Please close the issue.

closed 2018-03-19 by rokrodic

stefanache, 2019-02-28:
Hi you can put directly the ip for your url(si.pool.ntp.org) and in this case no dees dns.
To transform address into IP you can use this online service: https://whatismyipaddress.com/hostname-ip
My results was:
Results:
Lookup Hostname: si.pool.ntp.org
Lookup IPv4 Address: 193.2.78.2
Lookup IPv4 Address: 46.54.225.12
so you can try directly these two IPs(see above)
let me know if worked.

josefgull, 2021-03-12:
i still get 0.0.0.0 could someone post the hole example to get the ip from pc in the sam wifi network