espressif/arduino-esp32 #1234: Wifi connection issue

helmar74, 2018-03-18:
### Hardware:
Board:							ESP32 NodeMCU
Core Installation/update date:			17/Mar/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						?115200?

### Description:
I entered valid SSID an Password in user_config.h. But ESP doesn't connect to Wifi AP. Configuration works fine with a lot of other tasmota devices. I only changed STA_SSID1  and STA_PASS1  in user_config.h but doesn't work at all.

Serial Log Output:
8000,len:0
load:0x40078000,len:11392
entry 0x40078a24

00:00:00 APP: Set Baudrate to 115200

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_00000000, GroupTopic sonoffs) Version 5.12.0c-STAGED
00:00:00 WIF: Connecting to AP1  in mode 11
00:00:15 WIF: Connect failed with AP timeout
00:00:15 WIF: Connecting to AP1  in mode 11
00:00:30 WIF: WPSConfig failed to start
00:00:31 WIF: SmartConfigSSId1 , Password1 
00:00:33 APP: Restarting
ets Jun  8 2016 00:22:57

JacekDob, 2018-03-18:
Create issue here:
https://github.com/JacekDob/Sonoff-Tasmota/issues

This is arduino-esp32 github.

helmar74, 2018-03-18:
Wrong repo -> closed

closed 2018-03-18 by helmar74