espressif/arduino-esp32 #1237: Sending UDP message stops working after > 1 minute of inactivity

korstiaanS, 2018-03-19:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Thing Sparkfun
Core Installation/update date:			17/mar/2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Sending of UDP packages stops working after > 1 minute of inactivity

### Sketch:
```cpp

//Change the code below by your sketch
#include <WiFiUdp.h>
#include <WiFi.h>

bool udpSendMessage(IPAddress ipAddr, String udpMsg, int udpPort) {
  /** WiFiUDP class for creating UDP communication */
  WiFiUDP udpServer;
 
  // Start UDP client for sending packets
  int connOK = udpServer.begin(udpPort);
 
  if (connOK == 0) {
    Serial.println("UDP could not get socket");
    return false;
  }
 
  int beginOK = udpServer.beginPacket(ipAddr, udpPort);
 
  if (beginOK == 0) { // Problem occured!
    udpServer.stop();
    Serial.println("UDP connection failed");
    return false;
  }
 
  int bytesSent = udpServer.print(udpMsg);
  if (bytesSent == udpMsg.length()) {
    udpServer.endPacket();
    udpServer.stop();
    return true;
  } else {
    Serial.println("Failed to send " + udpMsg + ", sent " + String(bytesSent) + " of " + String(udpMsg.length()) + " bytes");
    udpServer.endPacket();
    udpServer.stop();
    return false;
  }
}

/** UDP port */
static const int udpBcPort = 9750;
/** WiFiUDP class for listening to UDP message */
WiFiUDP udpListener;
char esid[20] = "YourSSID";
char epass[20] = "YourPassword";
 
void setup(void) {
  // Some initialization stuff
  WiFi.begin(esid, epass);
  Serial.begin(115200);
  // Start UDP listener
  udpListener.begin(udpBcPort);
  int c = 0;
  while ( c < 30 ) {
      if (WiFi.status() == WL_CONNECTED) { 
        break;
      } 
      delay(500);
      c++;
  }
  Serial.println(WiFi.localIP());
  Serial.println("Started");
}
 
void loop(void) {
  // here is your code that is processed in a loop
 
  // Check if message arrived
  int udpMsgLength = udpListener.parsePacket(); 
  if (udpMsgLength != 0) {
    Serial.println("Receiving");
    byte udpPacket[udpMsgLength+1]; // reserve a local buffer
    IPAddress udpIP = udpListener.remoteIP();
 
    // Read the received payload into the local buffer
    udpListener.read(udpPacket, udpMsgLength);
    udpPacket[udpMsgLength] = 0;
 
    udpListener.flush(); // empty UDP buffer for next packet
 
    String debugMsg = "UDP package from ";
    debugMsg += "Sender IP: " + String(udpIP[0]) + "." + String(udpIP[1]) + "." + String(udpIP[2]) + "." + String(udpIP[3]);
    debugMsg += "\n Msg: " + String((char *)udpPacket);
    Serial.println(debugMsg);
   // send message "OK" back to Packet Sender, this stops working after > 1 minute of not receiving a package
    udpSendMessage(udpIP, "OK", 9750);
  }
}
```

### Debug Messages:
```

```

To test:

Fill in SSID and password.
After reboot send UDP packet to the correct IP of the ESP (can see this in Serial monitor) to port 9750.
Make sure that the UDP listener is active on the Packet Sender (port 9750).
Shortly after the reboot, the packet is received on the ESP and it sends an "OK" back to the Packet sender. This works every time. (Can be seen on Packet Sender)
Then wait 1 minute.
Try again, send a message to ESP.
Message is received on ESP.(check this via serial console).
NO MESSAGE arrives at Packet Sender. It only will work again after a reboot.

Conclusion?; UDP sending stops working after 1 minute (or more) of inactivity?

everslick, 2018-03-19:
try to make `WiFiUDP udpServer;` global (move it up 2 lines) and move

```
// Start UDP client for sending packets
  int connOK = udpServer.begin(udpPort);
 
  if (connOK == 0) {
    Serial.println("UDP could not get socket");
    return false;
  }
```
into setup() ...

does this change anything?

everslick, 2018-03-19:
ah and of course remove `udpServer.stop();`

korstiaanS, 2018-03-19:
No, that doesn't work. I don't get any message? Not even from the start?

everslick, 2018-03-19:
on further inspection I see that you create 2 UDP objects on the same port (9750). this does not make sense. get rid of the udpServer and just use the udpListener object to send.

korstiaanS, 2018-03-19:
Ok, I deleted the second one. I only have 1 now and it works UNTIL I wait some minutes and the sending fails. (The "OK" stops coming in Packet Sender).
I use same applications for months now on ESP8266 without any problems.

Global, .stop() or not, ... I tried everything. Nothing helps, it stops if I don't send  anything for several minutes. Strangly, receiving keeps working (can been seen on Serial Monitor).

lbernstone, 2018-03-19:
Sounds like it could be a memory leak, and your debugMsg stuff seems like a source of leaks.  Try this instead:
```
    Serial.printf("UDP package from Sender IP: %d.%d.%d.%d\nMsg: %s\n",
                   udpIP[0],udpIP[1],udpIP[2],udpIP[3],udpPacket);
```

korstiaanS, 2018-03-19:
Hi all, 

After further research I also found some strange UDP issues that was solved by changing the board from SparkFun to something else.
Luckily I also had another ESP32 board from DoIT and I put the SAME TEST program om it.
Result: THAT ONE WORKS and KEEPS WORKING!
Conclusion: Something different on SparkFun board? I will try to ask SparkFun or any idea? The SparkFun has better reception (10dB better).

stickbreaker, 2018-03-19:
@korstiaanS My esp32 is very orientation sensitive,  Try printing the RSSI value(`Wifi.RSSI()`) and changing orientation. With my esp32 a WeMos bluetooth & Battery it varies from -63 dBm to -85dBm just depending on it's orientation.  The worst is with the ESP32 antenna horizontal, Best is with Antenna Vertical inline with Router.

At -85 dBm my communications are unreliable.

Chuck.

korstiaanS, 2018-03-19:
I know, I already tested. But still, the ESP32 Thing has better reception.

korstiaanS, 2018-03-20:
On the DoIT on the company wifi I noticed also something; If I click a lot on the button in Packet Sender (sending a lot of data) suddenly the module stop responding with "OK" and I see at the same time that he does not respond to a ping anymore. Also, I still on the serial monitor that the udp string is still coming in? So, at the time the sending (and only the sending) stops the answer to a ping also stops. The receiving still works. That behaviour I didn't see at my home network.

korstiaanS, 2018-03-20:
Tried on a company Cisco network; After every reboot of the ESP the "pings" from a PC to the ESP works for 50 sec. After approx. 50 sec. the timeout on the pings start and does not come back until I reboot. I do nothing else. I already tried esp_wifi_set_ps(WIFI_PS_NONE) but no help. Is this a DTIM / beacon interval problem?

korstiaanS, 2018-03-27:
Hi all,

After further investigation we found the reason:

By further investigations and packet captures we have encountered the root cause of the connectivity issue.

In normal condition we see by packet capture that the ESP32 module is using the correct IP and correct MAC address of it’s default gateway.
So IP communication with devices outside it’s subnet is correctly send to the default gateway mac (example: 00:08:e3:aa:aa:aa)

Every 60 seconds the ESP32 module receives an IGMP general membership query from the IGMP querier in it’s subnet. (in our case this IGMP querier is our Cisco core switch).
Upon receival of the first IGMP query the ESP module changes the default gateway mac address (example: 00:08:e3:aa:aa:aa) with the source mac address of the igmp querier (example: 00:09:e3:bb:bb:bb)
As of now all IP packets towards it’s default gateway are send towards the incorrect mac address (00:09:e3:bb:bb:bb) and the ESP32 is not reachable anymore from outside it’s subnet.

In our specific setup we have disabled the IGMP querier functionality for the specific VLAN/IP subnet on the Cisco core switch to solve this encountered issue.

Why the ESP32 have problems and the ESP8266 not is for me not clear.

hrishikesh-thakre, 2019-05-11:
Have you found out any solution to this problem? I am having the same issue currently.

closed 2019-10-14 by stale[bot]

vmzhivetyev, 2020-01-30:
> Have you found out any solution to this problem? I am having the same issue currently.

Did you achieve any progress with the problem? I'm having same issue.

umairz1, 2022-08-10:
I am having the same issue with the esp32 not receiving udp message after some time. If anybody has the solution please reply.

goiw111, 2024-06-03:
guys i have the same issue but i fix it by adding a  delay(100); after the udp.endPacket();