espressif/arduino-esp32 #1238: EEPROM does not compile on the Arduino IDE 1.8.5

dirkx, 2018-03-19:
Below patch is needed in order to get the EEPROM code to compile on recent SDKs.

Otherwise it moans/misses the call by reference populating the value and hence errors-out on the seemingly uninitialized value passed to readAll().

Dw.

commit 9889158c741360e86618a5a16e2fff2c876ae16e
Author: Dirk-Willem van Gulik <dirkx@webweaving.org>
Date:   Mon Mar 19 17:59:01 2018 +0100

    Quell compiler warnings in Arduino IDE 1.8.5

diff --git a/libraries/EEPROM/EEPROM.cpp b/libraries/EEPROM/EEPROM.cpp
index 62e15af..d0ab410 100644
--- a/libraries/EEPROM/EEPROM.cpp
+++ b/libraries/EEPROM/EEPROM.cpp
@@ -178,85 +178,85 @@ uint16_t EEPROMClass::length ()
 */
 uint8_t EEPROMClass::readByte (int address)
 {
-  uint8_t value;
+  uint8_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 int8_t EEPROMClass::readChar (int address)
 {
-  int8_t value;
+  int8_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 uint8_t EEPROMClass::readUChar (int address)
 {
-  uint8_t value;
+  uint8_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 int16_t EEPROMClass::readShort (int address)
 {
-  int16_t value;
+  int16_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 uint16_t EEPROMClass::readUShort (int address)
 {
-  uint16_t value;
+  uint16_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 int32_t EEPROMClass::readInt (int address)
 {
-  int32_t value;
+  int32_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 uint32_t EEPROMClass::readUInt (int address)
 {
-  uint32_t value;
+  uint32_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 int32_t EEPROMClass::readLong (int address)
 {
-  int32_t value;
+  int32_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 uint32_t EEPROMClass::readULong (int address)
 {
-  uint32_t value;
+  uint32_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 int64_t EEPROMClass::readLong64 (int address)
 {
-  int64_t value;
+  int64_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 uint64_t EEPROMClass::readULong64 (int address)
 {
-  uint64_t value;
+  uint64_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 float_t EEPROMClass::readFloat (int address)
 {
-  float_t value;
+  float_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 double_t EEPROMClass::readDouble (int address)
 {
-  double_t value;
+  double_t value = 0;
   return EEPROMClass::readAll (address, value);
 }
 
 bool EEPROMClass::readBool (int address)
 {
-  int8_t value;
+  int8_t value = false;
   return EEPROMClass::readAll (address, value) ? 1 : 0;
 }

bulentperktas, 2018-03-25:
is this patch where it will be used?

Thx,

derwentx, 2018-04-14:
Also getting the same compiler warnings on Arduino 1.8.4, ESP32 core version "0.0.1" according to platform.txt
(ignore the unused variable warnings)

```
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/derwent/Library/Arduino15/packages -hardware /Users/derwent/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/derwent/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/derwent/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10804 -build-path /var/folders/sx/43gc_nmj43dcwbw15n3pwm440000gn/T/arduino_build_422908 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /Users/derwent/Documents/GitHub/TeleCortex/server/server.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/derwent/Library/Arduino15/packages -hardware /Users/derwent/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/derwent/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/derwent/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10804 -build-path /var/folders/sx/43gc_nmj43dcwbw15n3pwm440000gn/T/arduino_build_422908 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /Users/derwent/Documents/GitHub/TeleCortex/server/server.ino
Using board 'esp32' from platform in folder: /Users/derwent/Documents/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /Users/derwent/Documents/Arduino/hardware/espressif/esp32
Detecting libraries used...
"/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/config" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bluedroid" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_trace" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/app_update" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bootloader_support" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/bt" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/driver" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp32" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/esp_adc_cal" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ethernet" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/fatfs" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/freertos" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/heap" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/jsmn" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/log" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mdns" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/mbedtls_port" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nvs_flash" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/openssl" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/spi_flash" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/sdmmc" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/spiffs" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/tcpip_adapter" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/ulp" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/vfs" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/wear_levelling" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/xtensa-debug-module" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/coap" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/console" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/expat" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/json" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/lwip" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/newlib" "-I/Users/derwent/Documents/Arduino/hardware/espressif/esp32/tools/sdk/include/nghttp" "-I/Users/d [... cut ...]
Using library EEPROM at version 1.0 in folder: /Users/derwent/Documents/Arduino/hardware/espressif/esp32/libraries/EEPROM 
exit status 1
Error compiling for board ESP32 Dev Module.

```

And there's definitely something up.
When I check out EEPROM.cpp:179, it looks like this
```
/*
   Read 'value' from 'address'
*/
uint8_t EEPROMClass::readByte (int address)
{
  uint8_t value;
  return EEPROMClass::readAll (address, value);
}
```
clearly value isn't being initialized. Is it meant to be `return EEPROMClass::readAll (address, &value);` instead?

There must be some kind of bug here.

derwentx, 2018-04-14:
Issue goes away when I initialise all the values before they are given to `EEPROMClass:::readAll`

Here's what my EEPROM.cpp looks like, should i submit a pull request?

```
/*
  EEPROM.h -ported by Paolo Becchi to Esp32 from esp8266 EEPROM
           -Modified by Elochukwu Ifediora <ifedioraelochukwuc@gmail.com>

  Uses a one sector flash partition defined in partition table
  OR
  Multiple sector flash partitions defined by the name column in the partition table

  Copyright (c) 2014 Ivan Grokhotkov. All rights reserved.
  This file is part of the esp8266 core for Arduino environment.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#include "Arduino.h"
#include "EEPROM.h"

#include <esp_log.h>

static const char* TAG = "eeprom";

EEPROMClass::EEPROMClass(uint32_t sector)
  : _sector(sector)
  , _data(0)
  , _size(0)
  , _dirty(false)
  , _name("eeprom")
{
}

EEPROMClass::EEPROMClass(const char* name, uint32_t user_defined_size)
  : _sector(0)
  , _data(0)
  , _size(0)
  , _dirty(false)
  , _name(name)
  , _user_defined_size(user_defined_size)
{
}

EEPROMClass::EEPROMClass(void)
  : _sector(0)// (((uint32_t)&_SPIFFS_end - 0x40200000) / SPI_FLASH_SEC_SIZE))
  , _data(0)
  , _size(0)
  , _dirty(false)
  , _name("eeprom")
{
}

EEPROMClass::~EEPROMClass() {
  // end();
}

[... 360 lines cut ...]
    return 0;

  if (address < 0 || address + len > _size)
    return 0;

  memcpy(_data + address, (const void*) value, len);
  _dirty = true;
  return len;
}

template <class T> T EEPROMClass::writeAll (int address, const T &value)
{
  if (address < 0 || address + sizeof(T) > _size)
    return value;

  memcpy(_data + address, (const uint8_t*) &value, sizeof(T));
  _dirty = true;

  return sizeof (value);
}

#if !defined(NO_GLOBAL_INSTANCES) && !defined(NO_GLOBAL_EEPROM)
EEPROMClass EEPROM;
#endif
```

closed 2019-08-15 by stale[bot]