espressif/arduino-esp32 #1240: Receive binary data with a BLE characteristic

centrolutions, 2018-03-20:
### Hardware:
Board:							Adafruit ESP32 Huzzah
Core Installation/update date:			15/mar/2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Our project requires the capability of receiving binary data over BLE. As far as I can find, the only way to receive data is with a string.
We're trying to transfer a .bin file, save it to SPIFFS and then perform an update (OTA). Everything seems to work except the file that gets created contains text instead of the actual bytes, so we get:

> Trying to start update...
> Written : 5340 successfully
> E (125836) esp_image: invalid segment length 0x40138809
> Error Occurred. Error #: 9
> 

Since there doesn't seem to be a way to do this with the current BLE library, please consider this a feature request.

SeanStayn, 2018-03-21:
We need this capability too!

jrickard, 2018-04-06:
This is a bit astounding but yes, I've run into this too.  BLE is commonly used to transfer a wide array of data types, and it seems we are stuck with strings only.  

I've been able to hack up BLECharacterstics.cpp to allow to at least send various data types which works well with the phones.  But so far haven't figured out how to getValue from a characteristic that isn't a string.

We need this too....

robbie489, 2018-08-30:
@centrolutions did you find an answer to your problem?

centrolutions, 2018-08-30:
@robbie489 no, we ended up using WiFi to transfer the file.

lbernstone, 2018-08-30:
The typical method to move binary as strings is to base64 encode and then decode on the receiver side.  There is a base64 library in the arduino-esp32 core (although I seem to remember preferring the one in mbedtls).

kbickham, 2019-08-01:
I too am needing something along these lines. I'm working with dev kit c and an avr (atmega 1280). I need to pass the entire state over between controllers. There as good library EasyTransfer, however the hardware UART protocols don't allow this as I have been able to implement. I'm not trying to dismiss the above, but I'm only able to pass 16 and 32 bit datatypes.

closed 2019-10-14 by stale[bot]

WayneKeenan, 2019-12-09:
I would like to be able to receive binary data.

`BLECharacteristic` needs to provide a `getDataLength()` method to get the length from the `BLEValue m_value` member variable. 

Also `BLEValue` can only support binary data by moving away from only using the `std::string` type, as this prevents binary data from being represented internally.