espressif/arduino-esp32 #1244: Invalid Library Found

strachaldrat, 2018-03-21:
### Hardware:
Board:							ESP32 WROOM
Core Installation/update date:			21/03/2018
IDE name:						Arduino IDE 1.8.5 
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
When attempting to add the Arduino-esp32 library to the Arduino libraries folder and selecting the board in the board's manager. I receive this error multiple times:

Invalid library found in C:\Users\asus\Documents\Arduino\libraries\hardware: C:\Users\asus\Documents\Arduino\libraries\hardware

I have so far re-installed Arduino, reinstalled an earlier version of Arduino. I have followed the guidelines suggested in issues #766, #793, #770. I used this guide to install the library https://learn.sparkfun.com/tutorials/esp32-thing-hookup-guide . Using commit 3a4ec66d41615cbb1c3e830cb6e761cdc4cca9d3

Thanks in advance for any help.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2018-03-21:
arduino-esp32 is not a library.  It goes directly in the <arduino>/hardware directory (which should already exist).

strachaldrat, 2018-03-21:
This is the file structure that I have put the esp32 folder in
[image]
Are you talking Arduino folder in program files (x86)?

stickbreaker, 2018-03-21:
@strachaldrat no, with your example
>This Pc > Documents > Arduino > hardware > espressif

Chuck.

strachaldrat, 2018-03-21:
Ok thanks

closed 2018-03-21 by strachaldrat