espressif/arduino-esp32 #1245: Pinging an ESP32 stops on certain Wifi networks (ESP8266 doesn't)

korstiaanS, 2018-03-21:
### Hardware:
Board:							DoIT DEVKIT
Core Installation/update date:			21/mar/2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz or 80MHz
Upload Speed:						115200

### Description:
**Pinging** an ESP32 using a simple sketch (just connect to an AP, with or without DHCP), no other communication is active, times out forever always after 5-120sec after a reboot. 
Strangely enough it only happens on certain networks (Cisco, Meraki) and on other it keeps on working and there is no problem (D-Link, Asus). 
And also strange; if I do the same tests with an ESP8266 it always works, on ALL networks (all brands).
What can cause this problem on the ESP32?

### Sketch:
```cpp
#include <WiFi.h>
#include <WiFiUdp.h>

int status = WL_IDLE_STATUS;
char ssid[] = "yourssid";      //  your network SSID (name)
char pass[] = "yourpassword";  // your network password (use for WPA, or use as key for WEP)

char packetBuffer[21];      //buffer to hold incoming packet
char ReplyBuffer[] = "OK";  // a string to send back

uint8_t ip1=10,ip2=0,ip3=91,ip4=100;
uint8_t gw1=10,gw2=0,gw3=88,gw4=1;
uint8_t sn1=255,sn2=255,sn3=252,sn4=0;
uint8_t dns1_1=10,dns1_2=0,dns1_3=0,dns1_4=101;
uint8_t dns2_1=10,dns2_2=0,dns2_3=0,dns2_4=102;

IPAddress ip = IPAddress(ip1,ip2,ip3,ip4);
IPAddress subnet = IPAddress(sn1,sn2,sn3,sn4);
IPAddress gateway = IPAddress(gw1,gw2,gw3,gw4);
IPAddress dns1 = IPAddress(dns1_1,dns1_2,dns1_3,dns1_4);
IPAddress dns2 = IPAddress(dns2_1,dns2_2,dns2_3,dns2_4);

WiFiUDP Udp;

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  WiFi.config(ip, gateway, subnet, dns1, dns2); // do not use for DHCP!
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");  
  printWifiStatus();
}

void loop() {
  delay(10);
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your WiFi shield's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

```

lbernstone, 2018-03-21:
Can you set up a simple web server (just a hello world) to see if tcp is still functioning.  That may help pin down whether it is a udp problem or something larger.

korstiaanS, 2018-03-21:
@lbernstone : Ok, I will do so. As soon I have the result I will give you the answer.

korstiaanS, 2018-03-22:
@lbernstone : I tested it and when the pinging stops working the website is also NOT reachable.

korstiaanS, 2018-03-22:
@lbernstone  : Also tried via platformio and same bad result.

korstiaanS, 2018-03-23:
Another test;  A complete program (with ArduinoOTA.h) on Arduino IDE 1.8.5 keeps working (pinging) but compiled and uploaded on PlatformIO stops also after a short time? This was with an ESP8266 but it has the same effect as described above on the ESP32.

korstiaanS, 2018-03-26:
Same problem if I use esp-idf instead of Arduino

korstiaanS, 2018-03-27:
Hi all,

After further investigation we found the reason:

By further investigations and packet captures we have encountered the root cause of the connectivity issue.

In normal condition we see by packet capture that the ESP32 module is using the correct IP and correct MAC address of it’s default gateway.
So IP communication with devices outside it’s subnet is correctly send to the default gateway mac (example: 00:08:e3:aa:aa:aa)

Every 60 seconds the ESP32 module receives an IGMP general membership query from the IGMP querier in it’s subnet. (in our case this IGMP querier is our Cisco core switch).
Upon receival of the first IGMP query the ESP module changes the default gateway mac address (example: 00:08:e3:aa:aa:aa) with the source mac address of the igmp querier (example: 00:09:e3:bb:bb:bb)
As of now all IP packets towards it’s default gateway are send towards the incorrect mac address (00:09:e3:bb:bb:bb) and the ESP32 is not reachable anymore from outside it’s subnet.

In our specific setup we have disabled the IGMP querier functionality for the specific VLAN/IP subnet on the Cisco core switch to solve this encountered issue.

Why the ESP32 have problems and the ESP8266 not is for me not clear.

hostmit, 2018-09-17:
I'm having similiar issue on ESP32 on mikrotik netowrk. ESP8266 works perfect in same net. I dont have any IGMP querier.... Where do I go to solve this?

me-no-dev, 2018-09-18:
Please post serious WiFi issues like this to esp-idf :) since Arduino just wraps on top of it, they will be able to actually help (and have engineers looking at the issues)

sachintulla, 2019-07-03:
any solution ? thanks in advance.

closed 2019-09-15 by stale[bot]