espressif/arduino-esp32 #1247: saving cert file on spiffs

NareshDealer, 2018-03-21:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:

I m trying to send a cert file to my board via webserver. but when i m making the request, the file is written properly but getting load cert error.
when i read the file from spiffs...
Read from file: 
-----BEGIN%20CERTIFICATE-----%0AMIIE0zCCA7ugAwIBAgIQGNrRniZ96LtKIVjNzGs7SjANBgkqhkiG9w0BAQUFADCB%0AyjELMAkGA1UEBhMCVVMxFzAVBgNVBAoTDlZlcmlTaWduLCBJbmMuMR8wHQYDVQQL%0AExZWZXJpU2lnbiBUcnVzdCBOZXR3b3JrMTowOAYDVQQLEzEoYykgMjAwNiBWZXJp%0AU2lnbiwgSW5jLiAtIEZvciBhdXRob3JpemVkIHVzZSBvbmx5MUUwQwYDVQQDEzxW%0AZXJpU2lnbiBDbGFzcyAzIFB1YmxpYyBQcmltYXJ5IENlcnRpZmljYXRpb24gQXV0%0AaG9yaXR5IC0gRzUwHhcNMDYxMTA4MDAwMDAwWhcNMzYwNzE2MjM1OTU5WjCByjEL%0AMAkGA1UEBhMCVVMxFzAVBgNVBAoTDlZlcmlTaWduLCBJbmMuMR8wHQYDVQQLExZW%0AZXJpU2lnbiBUcnVzdCBOZXR3b3JrMTowOAYDVQQLEzEoYykgMjAwNiBWZXJpU2ln%0AbiwgSW5jLiAtIEZvciBhdXRob3JpemVkIHVzZSBvbmx5MUUwQwYDVQQDEzxWZXJp%0AU2lnbiBDbGFzcyAzIFB1YmxpYyBQcmltYXJ5IENlcnRpZmljYXRpb24gQXV0aG9y%0AaXR5IC0gRzUwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCvJAgIKXo1%0AnmAMqudLO07cfLw8RRy7K+D+KQL5VwijZIUVJ/XxrcgxiV0i6CqqpkKzj/i5Vbex%0At0uz/o9+B1fs70PbZmIVYc9gDaTY3vjgw2IIPVQT60nKWVSFJuUrjxuf6/WhkcIz%0ASdhDY2pSS9KP6HBRTdGJaXvHcPaz3BJ023tdS1bTlr8Vd6Gw9KIl8q8ckmcY5fQG%0ABO+QueQA5N06tRn/Arr0PO7gi+s3i+z016zy9vA9r911kTMZHRxAy3QkGSGT2RT+%0ArCpSx4/VBEnkjWNHiDxpg8v+R70rfk/Fla4OndTRQ8Bnc+MUCH7lP59zuDMKz10/%0ANIeWiu5T6CUVAgMBAAGjgbIwga8wDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8E%0ABAMCAQYwbQYIKwYBBQUHAQwEYTBfoV2gWzBZMFcwVRYJaW1hZ2UvZ2lmMCEwHzAH%0ABgUrDgMCGgQUj+XTGoasjY5rw8+AatRIGCx7GS4wJRYjaHR0cDovL2xvZ28udmVy%0AaXNpZ24uY29tL3ZzbG9nby5naWYwHQYDVR0OBBYEFH/TZafC3ey78DAJ80M5+gKv%0AMzEzMA0GCSqGSIb3DQEBBQUAA4IBAQCTJEowX2LP2BqYLz3q3JktvXf2pXkiOOzE%0Ap6B4Eq1iDkVwZMXnl2YtmAl+X6/WzChl8gGqCBpH3vn5fJJaCGkgDdk+bW48DW7Y%0A5gaRQBi5+MHt39tBquCWIMnNZBU4gcmU7qKEKQsTb47bDN0lAtukixlE0kF6BWlK%0AWE9gyn6CagsCqiUXObXbf+eEZSqVir2G3l6BFoMtEMze/aiCKm0oHw0LxOXnGiYZ%0A4fQRbxC1lfznQgUy286dUV4otp6F01vvpX1FQHKOtw5rDgb7MzVIcbidJ4vEZV8N%0AhnacRHr2lVz2XTIIM6RUthg/aFzyQkqFOFSDX9HoLPKsEdao7WNq%0A-----END%20CERTIFICATE-----

so its failing to load this file information as CERT.

### Sketch:
String FileNames[2]={"/text.txt","/test2.txt","/test3.txt"};

if (currentLine.startsWith("GET /a?roca=") )
        {
         ClientMessage = CurrentLine.substring(12, CurrentLine.lastIndexOf(' '));
         WriteSPIFFS(FileNames[0].c_str(), &ClientMessage[0]);
       }

### Debug Messages:
[E][ssl_client.cpp:28] handle_error(): X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected
[E][ssl_client.cpp:30] handle_error(): MbedTLS message code: -8576

Any help is appreciated.. Thank you.

lbernstone, 2018-03-21:
This doesn't write to SPIFFS, but pretty easy to change that. https://github.com/lbernstone/setCerts/blob/master/setCerts.ino

NareshDealer, 2018-03-21:
@lbernstone Thank you.. This is exactly what i want.. 
The only question i have is here you are writing to NVS and i m thinking of writing to SPIFFS. 
For this usecase ( certs ) which is more reliable, long term solution. SPIFFS or NVS.?

lbernstone, 2018-03-21:
Ultimately, they are the same thing.  They are both writing to the flash.  If you don't feel comfortable working with the nvs, then use spiffs.  I come from a microservices devops background, so I think keys should be kept in a keystore.

NareshDealer, 2018-03-21:
@lbernstone Thank you for your response.. I tried to upload your code on my ESP32 and seems to have alot of errors .. most of them are complaining about incompatible with esp32 as they all based on esp8266.
errors are quite a few so i attached all of them in a file.
[errors.txt](https://github.com/espressif/arduino-esp32/files/1835418/errors.txt)

can you please give more details on keystore.?

thanks.

lbernstone, 2018-03-21:
You need specific versions of the included libraries.  Follow the link in the README. You need to pull the esp32 branch (`git pull origin esp32`) from bbx10's github.  Hopefully this stuff will get mainlined soon.  I have no compelling reason for you to use nvs instead of spiffs.  6 or a half dozen, it is just what makes more sense to you.

NareshDealer, 2018-03-22:
@lbernstone Thank you for pointing me in this direction... I will download the latest repo and try again.. But i m still interested to find out a way to simply pass the same using html form/code or something which is not involving this heavy lib.

skyladle, 2018-04-11:
@lbernstone Thanks you code , do you try AWS IoT symtec root CA upload through HTTPs ,
there is wrong while upload this CA, I do not know where wrong I made. hope you can give some solution or suggestion.

debug info as below.
PostArg FileName: symtec_G5_CA.pem
PostArg Name: root_cert
PostArg Type: application/octet-stream
Start File: symtec_G5_CA.pem Type: application/octet-stream
File Upload: root_cert
Certificate file too large
Certificate file too large
End File: symtec_G5_CA.pem Type: application/octet-stream Size: 3165
PostArg FileName:--------------private.pem.key

I download the CA from AWS symantec as below

[image]

lbernstone, 2018-04-11:
If you are trying to use my code, there is a limit on the file size of 1984 bytes (the largest a key can be for nvs).  That would be why is telling you certificate file too large.

skyladle, 2018-04-13:
thanks @lbernstone , your code suppose is ok. I try the another FSBrowser this http upload get the same fail result, the root CA can not upload successful. deep dig till the class WebServer::_parseForm(), the parse
can not get the end boundary from browser result in the code can not reach phase of UPLOAD_FILE_END.
hope some body  can show me how to the solve this issue. is that  root CA can not upload through http ?

some debug information 
content of endBuf is weird.

Quit FILE START phase
UPLOAD_FILE_WRITE PHASE
total size =1731
content of endBuf=---END CERTIFICATE-----
------WebKitF⸮?L2⸮?d2⸮?
content of boundary=----WebKitFormBoundaryCSnuDkbcULLGld08
boundary length=38

closed 2019-08-15 by stale[bot]