espressif/arduino-esp32 #1248: connecting to an adhoc wifi network on mac (sierra)

Isiee, 2018-03-21:
Hi, i'm new here. My Esp32 Huzzah feather cannot connect to the ad hoc network created on my mac (with or without password). I run the simple test script and manage to connect without any problems with routers/mobile phones etc. but with mac computers (i tried several) it just doesn't connect. thanks for helping. peace

lbernstone, 2018-03-21:
Ad hoc is not supported.  The ESP32 can run as an access point, connect to an access point, run ESP-now (https://www.espressif.com/en/products/software/esp-now/overview) and soon will be able to do mesh WiFi.

Isiee, 2018-03-21:
Thanks Ibernstone! So no chance at all to connect esp32 directly with my mac? i'm reading sensor data, then sending it with osc. it has to be fast. i hope there is a workaround for it.. thanks!

lbernstone, 2018-03-21:
Set the ESP in AP mode and connect your MAC to it.

Isiee, 2018-03-21:
Great idea! thanks, solved!

closed 2018-03-21 by Isiee